evcc-io/evcc #13547: OCPP: Car disconnect isn't always triggered after an Available status from the charger.

jduar, 2024-04-22:
### Describe the bug

I sometimes run into a situation where after disconnecting a vehicle the car isn't actually disconnected from the charger in EVCC - the car still appears as connected to the charger. The behaviour seems random.

Every time the car is disconnected the charger correctly sends the Available status, but sometimes EVCC seems to ignore it. I'll leave logs from different situations below.

### Steps to reproduce

1. Car is charging.
2. Disconnect the cable.
3. Pray that the car is marked as available.

### Configuration details

```yaml
network:
  schema: http
  host: evcc
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/en/docs/reference/configuration/interval/

log: debug
levels:
  site: debug
  lp-1: trace
  lp-2: trace
  cache: error
  db: error
  ocpp: trace
  flobz: trace

site:
  - title: Site
    meters:
      grid: grid
    residualPower: 0
    maxGridSupplyWhileBatteryCharging: 0

meters:
  - name: grid
    type: custom
    power:
      source: const
      value: 27600

chargers:
  - name: charger5
    type: template
    template: pulsarplus
    timeout: 2m
    stationid: charger5

  - name: charger6
    type: template
    template: pulsarplus
    timeout: 2m
    stationid: charger6

loadpoints:
  - title: Garage Charger 5
    charger: charger5
    mode: "now"

  - title: Garage Charger 6
    charger: charger6
    mode: "now"

vehicles:
  - name: bbb
    type: template
    template: flobz
    title: bbb
    url: http://psacc_bbb:5000
    vin: <bbb_VIN>

  - name: fff
    type: template
    template: flobz
    title: fff
    url: http://psacc_fff:5000
    vin: <fff_VIN>
```

### Log details

```text
First log - car in charger5 (lp-1) was charging, was then disconnected, and though the "Available" status was sent by the charger, at no point was the charger made available. It stayed with status B until the next day: 

[ocpp  ] TRACE 2024/04/18 20:48:38 received JSON message from charger5: [2,"2011629336","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-04-18T18:48:37Z"}]
[ocpp  ] TRACE 2024/04/18 20:48:38 sent JSON message to charger5: [3,"2011629336",{}]
[ocpp  ] TRACE 2024/04/18 20:48:38 received JSON message from charger5: [2,"1302591646","StopTransaction",{"meterStop": 1613877,"timestamp": "2024-04-18T18:48:37Z","idTag": "","transactionData": [],"reason": "EVDisconnected","transactionId": 1}]
[ocpp  ] TRACE 2024/04/18 20:48:38 sent JSON message to charger5: [3,"1302591646",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/04/18 20:48:38 received JSON message from charger5: [2,"1453110360","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-04-18T18:48:37Z"}]
[ocpp  ] TRACE 2024/04/18 20:48:38 sent JSON message to charger5: [3,"1453110360",{}]
[site  ] DEBUG 2024/04/18 20:48:55 ----
[lp-1  ] DEBUG 2024/04/18 20:48:55 charge power: 0W
[lp-2  ] DEBUG 2024/04/18 20:48:55 charge power: 0W
[site  ] DEBUG 2024/04/18 20:48:55 grid meter: 27600W
[site  ] DEBUG 2024/04/18 20:48:55 pv power: 0W
[site  ] DEBUG 2024/04/18 20:48:55 site power: 27600W
[lp-2  ] DEBUG 2024/04/18 20:48:55 charger status: A
[ocpp  ] TRACE 2024/04/18 20:49:08 received JSON message from charger6: [2,"362535338","Heartbeat",{}]
[ocpp  ] TRACE 2024/04/18 20:49:08 sent JSON message to charger6: [3,"362535338",{"currentTime":"2024-04-18T18:49:08Z"}]
[ocpp  ] TRACE 2024/04/18 20:49:16 received JSON message from charger5: [2,"921829077","Heartbeat",{}]
[ocpp  ] TRACE 2024/04/18 20:49:16 sent JSON message to charger5: [3,"921829077",{"currentTime":"2024-04-18T18:49:16Z"}]
[site  ] DEBUG 2024/04/18 20:49:25 ----
[lp-1  ] DEBUG 2024/04/18 20:49:25 charge power: 0W
[lp-2  ] DEBUG 2024/04/18 20:49:25 charge power: 0W
[site  ] DEBUG 2024/04/18 20:49:25 grid meter: 27600W
[site  ] DEBUG 2024/04/18 20:49:25 pv power: 0W
[site  ] DEBUG 2024/04/18 20:49:25 site power: 27600W
[lp-1  ] DEBUG 2024/04/18 20:49:25 charger status: B

Second log is from the following day - a car was connected to that charger, and eventually disconnected. The charger sent the "Available" status again, and this time it was recognized and the car was disconnected:

[ocpp  ] TRACE 2024/04/19 19:47:23 received JSON message from charger5: [2,"1060236339","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-04-19T17:47:23Z"}]
[ocpp  ] TRACE 2024/04/19 19:47:23 sent JSON message to charger5: [3,"1060236339",{}]
[site  ] DEBUG 2024/04/19 19:47:25 ----
[lp-1  ] DEBUG 2024/04/19 19:47:25 charge power: 0W
[lp-2  ] DEBUG 2024/04/19 19:47:25 charge power: 0W
[site  ] DEBUG 2024/04/19 19:47:25 grid meter: 27600W
[site  ] DEBUG 2024/04/19 19:47:25 pv power: 0W
[site  ] DEBUG 2024/04/19 19:47:25 site power: 27600W
[lp-1  ] DEBUG 2024/04/19 19:47:25 charger status: A
[lp-1  ] INFO 2024/04/19 19:47:25 car disconnected

Third log - car is disconnected, charger status is changed to B and stays as B indefinitely despite there being no car connected:

[ocpp  ] TRACE 2024/04/22 14:11:58 received JSON message from charger5: [2,"247051122","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEV","timestamp": "2024-04-22T12:11:57Z"}]
[ocpp  ] TRACE 2024/04/22 14:11:58 sent JSON message to charger5: [3,"247051122",{}]
[ocpp  ] TRACE 2024/04/22 14:12:08 received JSON message from charger5: [2,"124455824","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-04-22T12:12:07Z"}]
[ocpp  ] TRACE 2024/04/22 14:12:08 sent JSON message to charger5: [3,"124455824",{}]
[ocpp  ] TRACE 2024/04/22 14:12:08 received JSON message from charger5: [2,"72585321","StopTransaction",{"meterStop": 1638960,"timestamp": "2024-04-22T12:12:07Z","idTag": "","transactionData": [],"reason": "EVDisconnected","transactionId": 2}]
[ocpp  ] TRACE 2024/04/22 14:12:08 sent JSON message to charger5: [3,"72585321",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/04/22 14:12:08 received JSON message from charger5: [2,"1843939918","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2024-04-22T12:12:07Z"}]
[ocpp  ] TRACE 2024/04/22 14:12:08 sent JSON message to charger5: [3,"1843939918",{}]
[site  ] DEBUG 2024/04/22 14:12:25 ----
[lp-1  ] DEBUG 2024/04/22 14:12:25 charge power: 7360W
[lp-2  ] DEBUG 2024/04/22 14:12:25 charge power: 0W
[site  ] DEBUG 2024/04/22 14:12:25 grid meter: 27600W
[site  ] DEBUG 2024/04/22 14:12:25 pv power: 0W
[site  ] DEBUG 2024/04/22 14:12:25 site power: 27600W
[lp-1  ] DEBUG 2024/04/22 14:12:25 charger status: B
[lp-1  ] INFO 2024/04/22 14:12:25 stop charging <-
[lp-1  ] DEBUG 2024/04/22 14:12:25 wake-up timer: start
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.125.0

andig, 2024-04-23:
Interesting. It looks as if the charger communicates but evcc is not processing the response.

andig, 2024-04-23:
Oh, I have a notion:

```go
// timestampValid returns false if status timestamps are outdated
func (conn *Connector) timestampValid(t time.Time) bool {
        ...

	// reject older values than we already have
	return t.After(conn.status.Timestamp.Time)
}
```

The last status updates all have the same timestamp. I guess the latter is ignored since if is not strictly `After`. Will change that to `!Before`.

andig, 2024-04-23:
Closed in https://github.com/evcc-io/evcc/commit/269bb76c8389d2e0189c56d11ef771f0402792c9

closed 2024-04-23 by andig