evcc-io/evcc #13548: Ladevorgang / Geladene Energie wird nicht angezeigt

julred, 2024-04-22:
### Describe the bug

Hallo zusammen,

heute habe ich während des Ladens festgestellt, dass die im Ladevorgang eingespeiste Energie ins Auto nicht angezeigt wird. Die aktuelle Ladeleistung wird korrekt dargestellt. Auch wird der Ladevorgang nicht in der Historie angezeigt.

### Steps to reproduce

Sporadisch, heute das erste mal gesehen
[image]

### Configuration details

```yaml
root@evcc:/etc# cat evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  charge_brain_solar: trace
  lp-1: trace
  sma: error
# unique installation id
plant: ***

interval: 10s # control cycle interval

sponsortoken: ***

telemetry: false

mqtt:
  broker: 192.168.178.15:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt
  password: ***

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.129
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.128  
  password: ***  
  name: pv3
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: battery  
  modbus: tcpip  
[... 468 lines cut ...]
Apr 22 18:44:40 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:40 charge total import: 583.690kWh
Apr 22 18:44:40 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:40 charger status: B
Apr 22 18:44:40 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:40 pv charge current: 5.31A = 0A + 5.31A (-2442W @ 2p)
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 ----
Apr 22 18:44:50 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:50 charge power: 0W
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 pv power: 2609W
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 battery soc: 99%
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 battery power: -2451W
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 grid meter: -22W
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 grid powers: [-42 -2 22]W
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 grid currents: [-0.405 -0.675 0.528]A
Apr 22 18:44:50 evcc evcc[30908]: [site  ] DEBUG 2024/04/22 18:44:50 site power: -2473W
Apr 22 18:44:50 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:50 charge total import: 583.690kWh
Apr 22 18:44:50 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:50 charger status: B
Apr 22 18:44:50 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:44:50 pv charge current: 5.38A = 0A + 5.38A (-2473W @ 2p)
root@evcc:~#
```

### What type of operating system are you running?

Linux

### Version

0.125.0

VolkerK62, 2024-04-22:
`Apr 22 18:43:40 evcc evcc[30908]: [lp-1  ] DEBUG 2024/04/22 18:43:40 final charge energy: -575kWh`
das ist seltsam

julred, 2024-04-22:
> `Apr 22 18:43:40 evcc evcc[30908]: [lp-1 ] DEBUG 2024/04/22 18:43:40 final charge energy: -575kWh` das ist seltsam

Momentan macht meine Installation mehrere Zicken iVm mit dem MID Zähler (siehe andere Issues von mir).
Ist das yaml File hinsichtlich des Zählers Debuggings korrekt?
```

levels:
  cache: error
  charge_brain_solar: trace
  lp-1: trace
  sma: error
```

VolkerK62, 2024-04-23:
>charge_brain_solar: trace

das wird vermutlich nicht funktionieren. 

`lp-1: trace` sollte reichen.
`cfos: trace` wird wahrscheinlich nur die Wallbox sein

julred, 2024-04-23:
> > charge_brain_solar: trace
> 
> das wird vermutlich nicht funktionieren.
> 
> `lp-1: trace` sollte reichen. `cfos: trace` wird wahrscheinlich nur die Wallbox sein

Beim heutigen Ladevorgang wurde die bereits geladene Energie korrekt angezeigt, trotz meiner alten / fehlerhaften yaml config.
Habe diese nun nach deinem Tipp abgeändert.

Das Problem mit der fehlerhaften Lademenge muss nun an einer anderen Stelle liegen. Gibts aus der log Datei andere Erkenntnisse?

andig, 2024-04-23:
Ist das ein temporäres Problem? Der Zählerstand taucht im Log auf. Jetzt bräuchte es mal einen Blick auf /api/state in dem Moment wo keine Ladeenergie mehr angezeigt wird.

julred, 2024-04-23:
> Ist das ein temporäres Problem? Der Zählerstand taucht im Log auf. Jetzt bräuchte es mal einen Blick auf /api/state in dem Moment wo keine Ladeenergie mehr angezeigt wird.

Habe das Phänomen bisher nur einmal erlebt. Wie kann ich den /api/state einsehen - fürs nächste mal sicherlich interessant

andig, 2024-04-23:
Im Browser als URL angeben!
...und bitte mal ein komplettes Log des *ganzen* Ladevorgangs, nicht nur Teile!

closed 2024-04-23 by andig

julred, 2024-04-23:
> Im Browser als URL angeben! ...und bitte mal ein komplettes Log des _ganzen_ Ladevorgangs, nicht nur Teile!

Ist ein bisschen was...

[evcc.log](https://github.com/evcc-io/evcc/files/15080871/evcc.log)

julred, 2024-04-26:
> Im Browser als URL angeben! ...und bitte mal ein komplettes Log des _ganzen_ Ladevorgangs, nicht nur Teile!

Beim aktuellen Ladevorgang wird wieder die bisher geladene Menge nicht dargestellt:

```

{
  "result": {
    "auth": {
      "vehicles": {

      }
    },
    "battery": [
      {
        "power": 160,
        "energy": 376.395,
        "soc": 99,
        "controllable": true
      }
    ],
    "batteryCapacity": 0,
    "batteryDischargeControl": true,
    "batteryEnergy": 376.395,
    "batteryMode": "normal",
    "batteryPower": 160,
    "batterySoc": 99,
    "bufferSoc": 100,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "fatal": null,
    "greenShareHome": 1,
    "greenShareLoadpoints": 1,
    "gridConfigured": true,
    "gridCurrents": [2.721, 2.463, -4.586],
    "gridEnergy": 14980.6722,
    "gridPower": -1.2,
    "gridPowers": [540.1, 520.5, -1061.8],
    "homePower": 251.8,
    "loadpoints": [
      {
        "chargeCurrent": 7.53282608695652,
        "chargeDuration": 3060000000000,
        "chargePower": 3504,
        "chargeTotalImport": 595.19,
        "chargedEnergy": 0,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": false,
        "chargerPhysicalPhases": null,
        "charging": true,
        "connected": true,
        "connectedDuration": 0,
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": "0001-01-01T00:00:00Z",
        "effectivePriority": 0,
        "enableThreshold": 0,
        "enabled": true,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 16,
        "minCurrent": 6,
        "mode": "pv",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
        "phasesActive": 2,
        "phasesConfigured": 3,
        "phasesEnabled": 3,
        "planEnergy": 0,
        "planOverrun": 0,
        "planProjectedStart": "0001-01-01T00:00:00Z",
        "planTime": "0001-01-01T00:00:00Z",
        "priority": 0,
        "pvAction": "inactive",
        "pvRemaining": 0,
        "sessionCo2PerKWh": null,
        "sessionEnergy": 0,
        "sessionPrice": null,
        "sessionPricePerKWh": null,
        "sessionSolarPercentage": 0,
        "smartCostActive": false,
        "title": "Garage eGolf",
        "vehicleClimaterActive": null,
        "vehicleDetectionActive": false,
        "vehicleLimitSoc": 0,
        "vehicleName": "eGolf",
        "vehicleOdometer": 0,
        "vehicleRange": 0,
        "vehicleSoc": 0
      }
    ],
    "prioritySoc": 10,
    "pv": [
      {
        "power": 1639,
        "energy": 2064.653
      },
      {
        "power": 1958,
        "energy": 17654.321
      }
    ],
    "pvEnergy": 19718.974,
    "pvPower": 3597,
    "residualPower": 0,
    "siteTitle": "Home",
    "smartCostType": "pricestatic",
    "sponsor": "Trial Sponsorship 🤠",
    "sponsorTokenExpires": 704046483363441,
    "statistics": {
      "30d": {
        "avgCo2": 0,
        "avgPrice": 0.104730060250028,
        "chargedKWh": 28.29,
        "solarPercentage": 89.6209770547819
      },
      "365d": {
        "avgCo2": 0,
        "avgPrice": 0.175984952525894,
        "chargedKWh": 820487156.15802,
        "solarPercentage": 46.7925594138948
      },
      "total": {
        "avgCo2": 0,
        "avgPrice": 0.175984952525894,
        "chargedKWh": 820487156.15802,
        "solarPercentage": 46.7925594138948
      }
    },
    "tariffFeedIn": 0.082,
    "tariffGrid": 0.301,
    "tariffPriceHome": 0.082,
    "tariffPriceLoadpoints": 0.082,
    "vehicles": {
      "eGolf": {
        "title": "eGolf",
        "icon": "car",
        "capacity": 32,
        "features": [
          "Offline"
        ]
      }
    },
    "version": "0.125.0"
  }
}
```

[image]

julred, 2024-04-26:
Liefere noch das passende log file dazu:
[evcc.log](https://github.com/evcc-io/evcc/files/15129591/evcc.log)

andig, 2024-07-06:
@julred besteht das Problem noch? Passiert das immer oder nur manchmal?

julred, 2024-07-08:
> @julred besteht das Problem noch? Passiert das immer oder nur manchmal?

Das Problem besteht noch weiterhin. Ich bin aktuell mit dem cfos Support am Diskutieren. Es sieht so aus, dass die cfos Box meist nach Beendigung des Ladevorgangs einmal ungewollt neu startet. Im Log der cfos kann man den Neustart mitprotokollieren. Seitens Support wird vermutet, dass der Neustart durch das Schalten des 3 Poligen Schützes vermehrt auftritt. Evcc triggert den Ladevorgang ja bis zu 5(?) mal an - um auch eingeschlafene Autos zu wecken. Wenn das Auto nun vollgeladen wurde, und die cfos den Ladevorgang beendet, triggert evcc den 3 poligen Schütz nochmal an. In dieser Konstellation - so scheint es - startet die Wallbox neu. Ich erhalte die nächsten Tage eine Austauschwallbox - in der Hoffnung dass der Fehler dann weg ist. 

@andig Kann man das Nachtriggern des Ladevorgangs testweise über einen Parameter für die cfos deaktivieren?

andig, 2024-07-08:
> Es sieht so aus, dass die cfos Box meist nach Beendigung des Ladevorgangs einmal ungewollt neu startet.

Mir wäre nicht klar, warum wir dabei Ladevorgänge verlieren sollten? evcc schaut bei der cfos ja nur auf die Zähelrstände, die müssten trotzdem da sein.

> Wenn das Auto nun vollgeladen wurde, und die cfos den Ladevorgang beendet, triggert evcc den 3 poligen Schütz nochmal an.

Dafür bräuchte ich erstmal ein Log.