evcc-io/evcc #13555: Durchschnittliche Ladegeschwindigkeit korrekt ermitteln

Neophyte85, 2024-04-23:
Eines vorweg, ihr macht einen super Job. Bin total begeistert von eurem Projekt.

Ich sehe es nicht wirklich als Fehler aber wollte es trotzdem einmal einbringen. 

[image]

Ich schließe mein Auto oft schon abends an. Manchmal saugt er dann noch bisschen Rest und beginnt dann am nächsten morgen die Ladung erneut (Nur PV Überschuss).

Ich persönlich finde schade, dass er die durchschnittliche Ladegeschwindigkeit über die gesamte Dauer, die er angeschlossen ist ermittelt und nicht über die Zeit in der der Ladevorgang wirklich läuft.

Wahrscheinlich kann man hierrüber streiten aber ich persönlich hätte da gerne die durchschnittliche Ladegeschwindigkeit des eigentlichen Ladevorganges. 

Falls relevant, RasberryPi, TWC3, Tesla Model 3 und 125.0

andig, 2024-04-23:
@naltatis dafür müssten wir die Dauer im Status C mit tracken. Machbar aber keine Prio?

naltatis, 2024-04-23:
Sieht aus wie was TWC spezifisches. Hier sollte eigentlich nur aktive Ladezeit (analog "Dauer" im Haupt UI. verwendet werden.

andig, 2024-04-23:
Was siehst Du hier TWC-spezifisch? Speichern wir die Ladedauer im Session Log? Woher kommt die?

naltatis, 2024-04-23:
@andig Die Ladedauer (effektiv) wird separat bei jedem Ladevorgang erfasst. Nicht nur die Differenz aus Ende und Start. Die Funktion gibts ja schon ewig bei uns. Der Soll-Zustand ist, dass die Ladedauer pausiert, wenn die Ladung pausiert. In der Main UI sieht man das dadurch, dass dann die Ladedauer-Anzeige auch stehen bleibt.

Da der TWC ja etwas speziell ist, kann ich mir vorstellen, dass das in diesem Fall irgendwie anders läuft.

naltatis, 2024-04-23:
[image]

andig, 2024-04-23:
Also: wir speichern `lp.chargingTime`. If falle des TWC kommt die aus 

```
// ChargingTime implements the api.ChargeTimer interface
func (v *Twc3) ChargingTime() (time.Duration, error) {
	res, err := v.vitalsG()
	return time.Duration(res.SessionS) * time.Second, err
}
```

Das liefert uns also der TWC. Wir haben bisher keine klare Erwartung kommuniziert, was das API konkret liefern soll- ob Verbindungsdauer oder tatsächlich Ladedauer. Wäre mal spannend, was die anderen Charger so machen um das zu präzisieren?

premultiply, 2024-04-23:
Geliefert werden _soll_ via `ChargingTime` die Netto-Ladedauer. Also die auffsummierte Zeit während einer Session in der sich der Ladepunkt im Status C befindet.

Ob das alle Charger-Implementierungen (schon) so ausnahmslos machen wenn dieses Interface implementiert ist darauf würde ich allerdings nicht wetten.
Oftmals weiß man es halt vorher nicht wenn man etwas halb blind implementiert.

andig, 2024-04-23:
Im Zweifel müssen wir das beim Twc ausbauen, dann greift das Fallback am Loadpoint.

naltatis, 2024-04-24:
> Die Daten in ChargingTime() kommen aus Modbus Registern des Bender Controllers.

Ja, dann hier das gleiche Problem.

andig, 2024-04-24:
Also raus damit? Die OpenWB Pro macht es richtig :)

mfuchs1984, 2024-04-24:
> > Die Daten in ChargingTime() kommen aus Modbus Registern des Bender Controllers.
> 
> Ja, dann hier das gleiche Problem.

Habe den Beitrag oben nochmal korrigiert, ChargeDuration(), nicht ChargingTime() habe ich mir angeschaut.

andig, 2024-04-24:
Das passt schon- hab ich vorhin erst umbenannt nachdem ich über dieses Issue auf die Inkonsistenz gestossen bin.

premultiply, 2024-04-24:
Mir stellt sich gerade die Frage welchen wirklichen Mehrwert diese Daten (und Interfaces) direkt von der Wallbox überhaupt für uns haben.
Neben ChargeDuration gibt es ja auch noch `ChargedEnergy` was ähnliche potenzielle Probleme hat.

Ausser bei einem Neustart von evcc während einer laufenden Session (Praxisrelevanz?) fällt mir dazu kein richtiger Mehrwert gegenüber interner Zeitnahme nach Ladezustand und Differenzbildung mit TotalEnergy ein.

Kann das vielleicht (beides) einfach weg?

andig, 2024-04-24:
Ich sehe genau das- die Restartfähigkeit.

closed 2024-04-28 by andig