evcc-io/evcc #13557: Anzeige: Erzeugung pro Inverter per Mouse-over/Tooltip erweitern

jori0001, 2024-04-23:
Wenn man mit der dem Mauszeiger auf die gerade erzeugten kW geht zeigt der Mous-over/Tooltip an, von welchem Inverter gerade wieviel erzeugt wird.

[image]

Wäre es möglich den Invertern einen Namen zu geben und dann im Tooltip anzuzeigen, welcher Wert von welchem Inverter kommt?
Also hier im Beispiel:

[image]

TobiasHuber1980, 2024-04-23:
guggst du: https://github.com/evcc-io/evcc/issues/6781#issuecomment-1475851804

jori0001, 2024-04-23:
Ah ok, das hatte ich nicht gefunden. Also kam das schon vor nem Jahr auf. Schade, dass da noch nichts implemtiert wurde. Selbst um Code fummeln will ich auf keinen Fall, sonst geht am Ende nichts mehr. XD

closed 2024-04-23 by andig