evcc-io/evcc #13570: Nach Wechsel auf 0.125.0 (be9e343a) Fehlermeldungen "Network Error. GET" im Frontend, jedoch nicht im Log

eldrik, 2024-04-24:
### Describe the bug

Seit dem Wechsel erhalte ich minütlich Meldungen des Typs jedoch ohne einen URL Anhang. 

"Network Error.
GET"

Schaue ich ins Debug Log werden keine GET Fehler angezeigt.

Eine Idee an welcher Stelle und wie ich der Meldung näher begegnen könnte?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7080

log: debug
levels:
  cache: error
#  proxy-5200: trace
# unique installation id
plant: 37116dbc07c1e508a3730dbd3a94e1d603da226e639d9daebc1832d03118c510

interval: 30s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NzI0MjEwNjEsImlhdCI6MTY3NzgxMzA2MSwiaXNzIjoiZXZjYy5pbyIsInN1YiI6ImVsZHJpayJ9.Lx_HvpLp_P2O70aJMXGWCGmAcNGPsYUczIp6-YgHPTY

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

#modbusproxy:
#   - port: 5200
#     uri: 10.0.81.40:502
mqtt:
  clientid: evcc
  broker: localhost:1883
  topic: evcc

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: 10.0.81.40  
  port: 502  
  usage: grid  
  storageunit: 1  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 2  
  host: 10.0.81.40  
  port: 502  
  usage: pv  
  storageunit: 1  
  modbus: tcpip  
  name: pv2
- type: template
[... 399 lines cut ...]
[lp-1 ] DEBUG 2024/04/24 15:13:16 charge power: 0W
[site ] DEBUG 2024/04/24 15:13:46 pv power: 9044W
[site ] DEBUG 2024/04/24 15:13:49 grid meter: -8222W
[site ] DEBUG 2024/04/24 15:14:00 grid currents: [-26.9 -17.7 -7.9]A
[site ] DEBUG 2024/04/24 15:14:03 site power: -8222W
[lp-1 ] DEBUG 2024/04/24 15:14:03 charger status: A
[site ] DEBUG 2024/04/24 15:14:03 ----
[lp-1 ] DEBUG 2024/04/24 15:14:03 charge power: 0W
[site ] DEBUG 2024/04/24 15:14:21 pv power: 18177W
[site ] DEBUG 2024/04/24 15:14:22 grid meter: -17050W
[site ] DEBUG 2024/04/24 15:14:26 grid currents: [-34.9 -23.4 -12.4]A
[site ] DEBUG 2024/04/24 15:14:27 site power: -17050W
[lp-1 ] DEBUG 2024/04/24 15:14:27 charger status: A
[site ] DEBUG 2024/04/24 15:14:27 ----
[lp-1 ] DEBUG 2024/04/24 15:14:27 charge power: 0W
[site ] DEBUG 2024/04/24 15:14:53 pv power: 19083W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.125.0 (be9e343a)

eldrik, 2024-04-24:
> Zeigt die Browser Console Fehler an?

tatsächlich! 

Scheinbar wird zum einen wegen der definierten battery3 etwas geworfen, ich war davon ausgegangen, dass es reicht die Verwendung unter site: auszukommentieren (siehe evcc.yaml).

Sollte ich die Definition besser in Gänze entfernen oder auskommentieren?

Die anderen Huawei Geräte meckert er auch an, hat sich an der Huawei Abfrage etwas geändert oder am logging? Den Intervall für den Dongle habe ich nicht verringert. (vorher Version 0.124.1)

`Config-CieCu7-0.js:1 Error fetching device values for meter battery3 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/battery3/status net::ERR_HTTP2_SERVER_REFUSED_STREAM
index-CI2EAdOo.js:87 no data received, assume we are offline
Config-CieCu7-0.js:1 Error fetching device values for meter grid1 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/grid1/status net::ERR_HTTP2_SERVER_REFUSED_STREAM
index-CI2EAdOo.js:87 no data received, assume we are offline
Config-CieCu7-0.js:1 Error fetching device values for meter pv5 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/pv5/status net::ERR_HTTP2_SERVER_REFUSED_STREAM
Config-CieCu7-0.js:1 Error fetching device values for meter pv2 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/pv2/status net::ERR_HTTP2_SERVER_REFUSED_STREAM
Config-CieCu7-0.js:1 Error fetching device values for meter pv6 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/pv6/status net::ERR_HTTP2_SERVER_REFUSED_STREAM
index-CI2EAdOo.js:87 no data received, assume we are offline
Config-CieCu7-0.js:1 Error fetching device values for meter battery3 
ke {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}
index-CI2EAdOo.js:23 
 GET https://lademanagement.eldrik.de/api/config/devices/meter/battery3/status net::ERR_HTTP2_SERVER_REFUSED_STREAM`

andig, 2024-04-24:
@eldrik schau mal bitte ins Network Tab, zur Not als Screenshot. Das ist unlesbar.

eldrik, 2024-04-24:
Ist es so etwas besser?

[Konsole.pdf](https://github.com/evcc-io/evcc/files/15097850/Konsole.pdf)

Kann den Network Tab gerade nicht liefern, habe die battery3 Definition auskommentiert und das interval von 30 auf 45 Sekunden hochgesetzt, derzeit kommen keine Meldungen mehr.

Wenn ich morgen wieder am Rechner sitze, stelle ich wieder auf den vorherigen Zustand her und ergänze die Ausgaben.

TobiasHuber1980, 2024-04-24:
https://evccgroup.slack.com/archives/C01321PUJAD/p1713716681434209

lt. @naltatis mit #13319 behoben.

andig, 2024-04-26:
@naltatis können wir das- wenn Fehler im Nightly- auch hier lösen?

naltatis, 2024-04-26:
Ja, gibt jetzt nen kleinen Backport 😀

PR #13596 (2024-04-26): UI: better handle api errors when offline
fixes #13570

- 🍋 don't refresh device status (config) when offline
- 🔺 don't show rest api errors when offline
- 🔑 don't show passwort prompt when endpoint couldn't be reached (added unknown state)

subset of https://github.com/evcc-io/evcc/pull/13319 which also includes some other major improvements when offline or while restarting.
closed 2024-04-26 by naltatis