evcc-io/evcc #13582: Shelly: keine negativen Ströme mehr

VolkerK62, 2024-04-25:
Ich habe einen Shelly 3EM als Grid Meter.
Da wurden die Ströme bisher sowohl negativ, als auch positiv angezeigt.
Der Shelly selbst misst das nicht. Siw wird das Vorzeichen im code anhand der Phasenleistungen angepasst.
 
Mit ist gerade aufgefallen, dass das seit 17.4. nicht mehr so ist/war
Ich bin vom aktuellen Nightly auf Release 0.125.0 zurück und alles ist wieder ok.
Keine Ahnung, mit welchem Nightly das reingekommen ist.

andig, 2024-04-25:
Mangels Logfile richtig/falsch müsste ich hier jetzt direkt zu machen. Mir ist nix bekannt.

VolkerK62, 2024-04-25:
ok. Log kommt .... muss mal schauen wie ich den direkten Vergleich hinbekomme.

VolkerK62, 2024-04-25:
Hier der direkte Vergleich. Dieselbe Config auf zwei verschiedenen Systemen.

aktuelles Nightly
```
[site  ] DEBUG 2024/04/25 13:47:18 grid powers: [-303 -1552 2398]W
[site  ] DEBUG 2024/04/25 13:47:18 grid currents: [1.9 6.64 11]A
```

Release V 0.125.0
```
Apr 25 13:47:15 raspberrypi evcc[23852]: [site  ] DEBUG 2024/04/25 13:47:15 grid powers: [-313 -1555 2398]W
Apr 25 13:47:15 raspberrypi evcc[23852]: [site  ] DEBUG 2024/04/25 13:47:15 grid currents: [-1.88 -6.67 11]A
```

andig, 2024-04-26:
Kannst Du mal bitte den Shelly auf trace setzen und das nochmal per 

    evcc meter 

machen?

VolkerK62, 2024-04-26:
es gab vor 2 Wochen eine Änderung in `core/site.go`
Es scheint so, als hatte das keine Wirkung mehr
https://github.com/evcc-io/evcc/blob/1d4e69591f3f09c7d237fc7685eb6580be370320/core/site.go#L568

andig, 2024-04-26:
Also: `evcc meter` zeigt das Phänomen nicht an weil es erst in der Nachbearbeitung passiert. Schonmal doof. Welche Änderung meinst Du?

VolkerK62, 2024-04-26:
Es trat auf, nachdem die Sachen mit dem UI Log raus war.
Ich habe dann geschaut, was hat sich in dem Zeitraum noch geändert.
https://github.com/evcc-io/evcc/commit/31f59c9eaf5651fd42640883b407baec97be648a

Keine Ahnung, was du mit `git bisect` meinst.

andig, 2024-04-26:
Der Commit sollte damit nihts zu tun haben. Hilft leider nur, das nightly oder den Commit zu finden, mit dem das Problem kam.

VolkerK62, 2024-04-26:
ok. kann ich gerne machen. kann ich gezielt ein spezielles Nightly installieren?
Wie ich ein bestimmtes Release installiere ist mir bekannt.

StefanSchoof, 2024-04-26:
`git bisect` ist ein Befehl um den ersten "bad" commit (oder last "good" commit) zu finden.

Es geht ungefähr so:
- `git bisect start`
- `git bisect good 0.125.0` (tag oder hash von der letzten Version die funktioniert).
- `git bisect bad 1d4e69591f3f09c7d237fc7685eb6580be370320` (hash oder tag von der Version die nicht mehr funktioniert)
- version bauen und testen
- Wenn Fehler auftritt `git bisect bad`  wenn nicht `git bisect good`
- git checkt eine neue Version aus und die wieder bauen und testen usw.

und irgendwann sagt er welche Version die letzte good ist und damit bekommt man dann raus welcher commit es war.

https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination

VolkerK62, 2024-04-26:
das scheitet bei mir spätestes hier `version bauen und testen`

VolkerK62, 2024-04-26:
ok, habs gefunden .... Dank @StefanSchoof   https://github.com/evcc-io/evcc/discussions/9501#discussioncomment-6795314

VolkerK62, 2024-04-26:
ok. Ich habs.   `0.125.0+1713320065`
Ist die erste Version, mit der es nicht mehr funktioniert.

StefanSchoof, 2024-04-26:
Hast du auch den commit hash von dem letzten funktionierenden Version?

VolkerK62, 2024-04-26:
klar  `evcc 0.125.0 (39db91c9)`

StefanSchoof, 2024-04-26:
Dann sind das die commits die in Frage kommen: https://github.com/evcc-io/evcc/compare/39db91c9...9d8e2b19

andig, 2024-04-26:
Dann kann es kur der sein den Volker schon raus gesucht hatte. Ich seh nur nicht warum???

VolkerK62, 2024-04-26:
Hmm, also doch https://github.com/evcc-io/evcc/commit/31f59c9eaf5651fd42640883b407baec97be648a ??

andig, 2024-04-27:
> Hmm, also doch https://github.com/evcc-io/evcc/commit/31f59c9eaf5651fd42640883b407baec97be648a ??

Bingo, ich sehe es. Da hat der Compiler leider nicht gewarnt :(

andig, 2024-04-27:
Closed in 652cab3192b57df3a98d5f13cfe787c5fd67e2f8

closed 2024-04-27 by andig

VolkerK62, 2024-04-27:
Danke