evcc-io/evcc #13590: Tinkerforge Warp3 kein Umschalten 1P/3P

StevieC121176, 2024-04-25:
### Describe the bug

Ich habe die neue Warp3 über das Warp3 Template in EVCC eingebunden.
Das manuelle und auch automatische Umschalten 1P/3P funktioniert nicht. Es kommt die Info das in x-Minuten umgeschaltet wird, zählt herunter, aber schaltet nicht zwischen 1P und 3P um
Ab und an kommt eine Meldung : `WARN 2024/04/25 17:31:25 charger logic error: phases mismatch (got 1, expected 3)`

Eine Auffälligkeit gibt es noch. Beim stoppen der Ladung geht die Ladeleitung in EVCC nie auf 0. Es wird immer eine Restleistung angezeigt am Ladepunkt der auch mit in IN&Out einfliesst, obwohl die Leistung am Charger 0W ist (sh. Screenshot Charger)

Phasenumschaltung ist nur über das Warp eigene Gui möglich.

### Steps to reproduce

1. Ladung startet
2. keine Umschalten 1P/3P möglich
3. beim stoppen der Ladung wird trotzdem noch eine Restleistung angezeigt obwohl z.Bsp kein Auto mehr angeschlossen ist. Diese fließt auch noch in die ganzen Berechnungen (In & Out) mit ein (sh. Screenshot und Log)
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 3s # control cycle interval

sponsortoken: xxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv  
  modbus: tcpip
  endianness: little
  minsoc: 20
  maxsoc: 100
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip
  endianness: little 
  capacity: 12.8 
  minsoc: 20 
  maxsoc: 100  
  watchdog: 60s 
  name: battery
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: fritzdect 
  usage: pv  
  uri: https://fritz.box  
  user: xxxxx  
  password: 'xxxxx'  
  ain: xxxxx  
  name: pv3
#- type: template
#  template: shelly-pro-3em
#  usage: pv
#  host: 192.168.178.135
#  name: pv5
#- type: template
#  template: homematic 
#  usage: pv  
#  host: 192.168.178.68  
#  device: 'xxxxx'  
#  user: xxxxx  
#  password: 'xxxxx'  
#  meterchannel: 6  
#  cache: 1s  
#  name: pv2

chargers:
- type: template
  template: tinkerforge-warp3
  host: 192.168.178.80
  port: 1883
  user: xxxxx
  password: xxxxx
  topic: warp3/2avk
  timeout: 30s
  name: wallbox1
#- type: template
#  template: tinkerforge-warp
#  host: 192.168.178.80
#  port: 1883
#  user: xxxxx
#  password: xxxxx
#  topic: warp3/2avk
#  timeout: 30s
#  energymanager: warp3/2avk
#  name: wallbox1
#- type: template
#  template: daheimladen-mb
#  host: 192.168.178.73
#  port: 502
#  name: wallbox6
- type: template
  template: homematic
  host: 192.168.178.68
  port: 2010
  device: '0xxxxx'
  standbypower: 15
  user: xxxxx
  password: 'xxxxx'
  meterchannel: 6
  switchchannel: 3
  integrateddevice: true
  icon: bike
  Heating: true
  name: wallbox10
#- type: template
#  template: fritzdect
#  uri: https://fritz.box
#  user: xxxxx
#  password: 'xxxxx'
#  ain: xxxxx
#  standbypower: 15
#  integrateddevice: true
#  icon: bike
#  name: wallbox1
#- type: template
#  template: shelly
#  host: 192.168.178.101
#  user: Sxxxxx
#  password: xxxxx
#  channel: 0
#  standbypower: 15
#  integrateddevice: false
#  icon: bike  
#  name: wallbox1

vehicles:
#- type: template
#  template: id 
#  title: Volkswagen ID5  
#  user: xxxxx  
#  password: 'xxxxx'  
#  vin: xxxxx  
#  capacity: 82  
#  phases: 3  
#  icon: car  
#  cache: 15m  
#  mode: off  
#  minSoc: 10  
 # targetSoc: 100  
 # minCurrent: 6  
 # maxCurrent: 16  
 # name: ev5
#- type: template
#  template: enyaq
#  title: Skoda Enyaq 80iV
#  user: xxxxx
#  password: 'xxxxx'
#  vin: xxxxx
#  capacity: 82
#  phases: 3
#  cache: 15
#  mode: off
#  minSoC: 10
#  targetSoC: 100
#  minCurrent: 6
#  maxCurrent: 16
  #identifiers:
  #- 60022376
#  name: ev6
#  icon: car
- type: template
  template: offline 
  title: E-Bike
  capacity: 0.75
  phases: 1  
  mode: off  
  minCurrent: 1
  maxCurrent: 6
  name: ev8
  icon: bike

loadpoints:
- title: Garage
  charger: wallbox1
  mode: off
#  phases: 3
#  mincurrent: 6
#  maxcurrent: 16
#resetOnDisconnect: true
#  Priority: 1
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Fahrrad
  charger: wallbox10
  vehicle: ev8
  mode: off
  phases: 1
#  mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: false
 # Priority: 2
#- title: Individual
#  charger: wallbox1
#  vehicle: ev8
#  mode: off
#  phases: 1
#  mincurrent: 6
#  maxcurrent: 16
#  resetOnDisconnect: false
 # Priority: 2

site:
  title: zu Hause
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    - pv3
 #   - pv5
    battery:
    - battery
  residualPower: 100

influx:
  url: http://192.168.178.80:8086
  database: evcc
  user: evcc
  password: xxxxx
#modbusproxy:
#  - port: 5200
#    uri: 192.168.178.74:502 # IP und Port des Gerätes, das abgefragt werden soll

mqtt:
  broker: 192.168.178.80:1883
  topic: evcc
 # clientid: foo
  user: xxxxx
  password: xxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.315 # [currency]/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.074 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: "xxxxx"
```

### Log details

```text
sh. unten
```

### What type of operating system are you running?

Linux

### Version

0.125.0+1714011352

StevieC121176, 2024-04-25:
[evcc-20240425-174545-debug.log](https://github.com/evcc-io/evcc/files/15122167/evcc-20240425-174545-debug.log)

StevieC121176, 2024-04-25:
[image]

[image]

StevieC121176, 2024-04-26:
Jetzt habe ich die Warp3 mal über das Warp template mit gleichem Topic für den WarpEnergyMeter, aber gleiches Ergebnis. 

[evcc-20240426-093209-info.log](https://github.com/evcc-io/evcc/files/15126938/evcc-20240426-093209-info.log)

andig, 2024-04-26:
Aussage bisher war, dass beim Warp3 das EM Topic jetzt mit dem WB Topic identisch sein. Dem scheint nicht so zu sein.

Dann hat sich anscheinend beim Warp3 die Umschaltung geändert? Gibts dafür ein neues Topic? Welches? Klappt die Umschaltung wenn ihr direkt über das Topic schaltet? Dann würde ich das so aufnehmen.

StevieC121176, 2024-04-26:
Ich habe mir das Topic mit dem MQTT Explorer mal angeschaut, leider weiß ich nicht welcher Punkt für die Phasenumschaltung zuständig ist.
Momentan ist es mir nur möglich die Phasen direkt über das warp ui umzuschalten.

rtrbt, 2024-04-26:
Zieh bitte mal ein Debug-Protokoll (unter System->Ereignis-Log) von der Wallbox, wenn die Phasenumschaltung passieren sollte.

StevieC121176, 2024-04-26:
> Zieh bitte mal ein Debug-Protokoll (unter System->Ereignis-Log) von der Wallbox, wenn die Phasenumschaltung passieren sollte.

Hier mal das log. Ich habe auf 3 phasig umgeschaltet. Ausgeführt wurde das allerdings erst nach Neustart des brokers. Ein erneutes umschalten auf 1 phasig wurde nicht angenommen.

[debug-report-warp3-2avk-2024-04-26T15-29-35-967.txt](https://github.com/evcc-io/evcc/files/15131227/debug-report-warp3-2avk-2024-04-26T15-29-35-967.txt)

rtrbt, 2024-04-26:
Ich sehe im Log
```
Topic power_manager/external_control_update is an action. Ignoring retained message (data_len=22).
```
und 
```
Received message on unknown topic 'power_manager' (data_len=17)
```
Hast du von Hand MQTT-Nachrichten geschickt?
@andig Schickt EVCC die external_control_update mit retain-Flag?

andig, 2024-04-26:
@rtrbt Soweit ich sehe schicken wir das _nicht_ retained! `power_manager` kommt aus der Warp2:

	wb.emStateG, err = to.JsonGetter(mq("%s/power_manager/state", emTopic))
	if err != nil {
		return nil, err
	}

	wb.phasesS, err = provider.NewMqtt(log, client,
		fmt.Sprintf("%s/power_manager/external_control_update", emTopic), 0).
		WithPayload(`{ "phases_wanted": ${phases} }`).
		IntSetter("phases")
	if err != nil {
		return nil, err
	}

Müssen wir das jetzt auf `evse` oder was Anderes umbiegen?

StevieC121176, 2024-04-26:
> Ich sehe im Log
> 
> ```
> Topic power_manager/external_control_update is an action. Ignoring retained message (data_len=22).
> ```
> 
> und
> 
> ```
> Received message on unknown topic 'power_manager' (data_len=17)
> ```
> 
> Hast du von Hand MQTT-Nachrichten geschickt? 

Nein, nur über die weboberfläche der warp3

andig, 2024-04-27:
Laut Doku sollte `power_manager` immer noch korrekt sein: https://warp-charger.com/documentation/api.html#features_phase_switch

StevieC121176, 2024-04-27:
ich habe folgendes probiert, in EVCC die Phasen manuell versucht umzuschalten in den Einstellungen für den Charger. Im MQTT Explorer unter power_manager/external_control_update sieht man auch das sich `phases_wanted` ändert.
Im Ereignis log der Wallbox erscheint dann allerdings immer diese Meldung und die Aktion wird nicht ausgeführt: `2024-04-27 11:51:34,401 | mqtt             | Topic power_manager/external_control_update is an action. Ignoring retained message (data_len=22).`

[image]
[image]
[image]

andig, 2024-04-27:
Schickst du die mal mit mosquitto um sicher zu gehen, dass sie wirklich nicht retained ist?

StevieC121176, 2024-04-27:
mit mosquitto funktioniert die Umschaltung problemlos. Auch ohne Fehlermeldung. Nun hab ich auch im iobroker mqtt Broker eine Einstellung gefunden, das ohne retained gesendet wird. Damit funktioniert dieser Broker nun auch.

closed 2024-04-27 by StevieC121176

andig, 2024-04-27:
Das ist ein bisschen verwirrend. Du hast ioBroker als Mqtt Server? Und der hat die Option, selbstständig Nachrichten in `retained` zu verwandeln?

StevieC121176, 2024-04-28:
> Das ist ein bisschen verwirrend. Du hast ioBroker als Mqtt Server? Und der hat die Option, selbstständig Nachrichten in `retained` zu verwandeln?

Ja, ich nutze den mqtt Adapter in iobroker. Dort war der Punkt `Nachrichten ohne retain flag senden` nicht ausgewählt. Sh. Foto. Nachdem ich den gesetzt hatte funktioniert die phasenumschaltung problemlos mit dem warp3 Template 

[image]

andig, 2024-04-28:
Verstehe ich tatsächlich immer noch nicht. „Senden“ tut ja evcc und entscheidet über das Flag. Der Mqtt Server hat ausschließlich die Aufgabe die Nachrichten weiter zu geben. Aber seis drum, Hauptsache es geht.

rtrbt, 2024-04-29:
Ich hätte *gleich* fragen sollen, ob du ioBroker als MQTT-Broker benutzt. Die Implementierung ist eine Katastrophe, siehe z.B. hier: https://www.tinkerunity.org/topic/11943-warp1-verliert-wlan-und-mqtt-einstellungen-214/ In unserem Forum ist das inzwischen die erste Frage, die ich immer bei MQTT-Problemen stelle und die Erfolgsquote ist hoch :D