evcc-io/evcc #13591: Min+PV with Tinkerforge Warp2

bbr111, 2024-04-26:
### Describe the bug

Beim Umschalten von Aus -> Min+PV oder von PV -> Min+PV steht das System sehr lange auf Ladebereit. Warte auf Fahrzeug.
Die Externe Freigabe der WARP2 wird aber nicht gesetzt.
Meine Erwartung wäre hier, dass das System mit 6A+Überschuss aus PV das auto lädt.
Es wird dann irgendwann:
Warp2: charger out of sync: expected enabled, got disabled angezeigt.
Die Netzwerk Konnektivität ist da.
Wenn man auf Schnell(Now) stellt und dann auf Min+PV zurück.
Lädt das System, regelt aber nicht mehr auf 6 A runter, und bleibt bei 16 bzw 32 A in der externen Freigabe stehen.

### Steps to reproduce

siehe beschreibung

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 20s # control cycle interval

log: trace
# levels:
#   cache: error

mqtt:
  broker: 192.168.111.x:1883
  user: xxxx
  password: xxxx

meters:

  - name: pvbalkon
    type: custom
    power:
      source: http
      uri: http://192.168.xxx.xxx/status
      jq: .meters | map(.power) | add

  - name: batsma
    type: template
    template: sma-hybrid
    usage: battery
    capacity: 7.7
    modbus: tcpip
    id: 3
    host: 192.168.xxx.xxx

  - name: pvsma
    type: template
    template: sma-hybrid
    usage: pv
    modbus: tcpip
    id: 3
    host: 192.168.xxx.xxx

  - name: hmsma
    type: sma
    uri: 192.168.xxx.xxx

chargers:
  - type: warp-fw2
    broker: 192.168.xxx.xxx3 # Die IP Adresse oder der Hostname des MQTT Brokers
    #port: 1883  # Der Port des MQTT Brokers
    topic: warp2/WDB
    timeout: 30s
[... 139 lines cut ...]
[warp  ] TRACE 2024/04/26 09:41:35 recv warp2/WDB/evse/low_level_state: '{"led_state":1,"cp_pwm_duty_cycle":1000,"adc_values":[3914,3456,134,137,1026,3809,134],"voltages":[12033,9080,-12336,-12310,826,12278,-12336],"resistances":[2597,219],"gpio":[true,false,false,false,true,true,false,true,false,true,false,false,false,false,true,true,true,false,false,true,false,false,false,false],"charging_time":0,"time_since_state_change":302341,"uptime":825100577,"time_since_dc_fault_check":79277840,"temperature":0,"phases_current":3,"phases_requested":3,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":0}'
[warp  ] TRACE 2024/04/26 09:41:35 recv warp2/WDB/meter/all_values: '[237.0488129,235.992981,235.4793854,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,-33.05612564,-15.92076492,63.3187027,236.2057648,0,0,0,0,0,1,0,49.96749496,4262.667969,14.84599972,46.51699829,230.4909973,4286.473145,18471.37891,null,null,null,null,null,null,409.4125977,407.8567505,408.8924866,408.7053223,0,3.094112873,1.766494155,2.479082584,0,0,0,4.042015076,0,null,null,null,null,null,null,0,0,0,0,4277.513672,277.0079956,3804.040771,455.8079834,2.818999767,0,0,14.84599972,3804.040771,455.8079834,17.66499901,46.51399994,0.003,0,173.4449921,50.61599731,6.429999828,219.9589996,50.61899948,6.429999828]'
[sma   ] TRACE 2024/04/26 09:41:35 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/04/26 09:41:36 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[warp  ] TRACE 2024/04/26 09:41:37 recv warp2/WDB/evse/low_level_state: '{"led_state":1,"cp_pwm_duty_cycle":1000,"adc_values":[3914,3457,134,137,1026,3811,133],"voltages":[12033,9086,-12336,-12310,826,12284,-12342],"resistances":[2604,219],"gpio":[true,false,false,false,true,true,false,true,false,true,false,false,false,false,false,true,true,false,false,true,false,false,false,false],"charging_time":0,"time_since_state_change":305411,"uptime":825103647,"time_since_dc_fault_check":79280910,"temperature":0,"phases_current":3,"phases_requested":3,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":0}'
[warp  ] TRACE 2024/04/26 09:41:37 recv warp2/WDB/meter/all_values: '[237.0670013,235.8958282,235.2525787,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,-28.57627487,42.23854828,80.31635284,235.9668427,0,0,0,0,0,1,0,49.94891739,4262.667969,14.84599972,46.51699829,230.4909973,4286.473145,18471.37891,null,null,null,null,null,null,409.4828491,408.1703491,409.244873,409.0532227,0,2.860119343,3.063493967,2.400130033,0,0,0,3.053434134,0,null,null,null,null,null,null,0,0,0,0,4277.513672,277.0079956,3804.040771,455.8079834,2.818999767,0,0,14.84599972,3804.040771,455.8079834,17.66499901,46.51399994,0.003,0,173.4449921,50.61599731,6.429999828,219.9589996,50.61899948,6.429999828]'
[sma   ] TRACE 2024/04/26 09:41:37 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/04/26 09:41:38 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/04/26 09:41:39 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[warp  ] TRACE 2024/04/26 09:41:40 recv warp2/WDB/evse/low_level_state: '{"led_state":1,"cp_pwm_duty_cycle":1000,"adc_values":[3915,3456,134,137,1026,3810,134],"voltages":[12039,9080,-12336,-12310,826,12281,-12336],"resistances":[2592,219],"gpio":[true,false,false,false,true,true,false,true,false,true,false,false,false,false,true,true,true,false,false,true,false,false,false,false],"charging_time":0,"time_since_state_change":308234,"uptime":825106470,"time_since_dc_fault_check":79283733,"temperature":0,"phases_current":3,"phases_requested":3,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":0}'
[warp  ] TRACE 2024/04/26 09:41:40 recv warp2/WDB/meter/all_values: '[237.2874146,236.1693726,235.6377258,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,-28.07168579,0,59.33294296,236.3614655,0,0,0,0,0,1,0,49.95677185,4262.667969,14.84599972,46.51699829,230.4909973,4286.473145,18471.37891,null,null,null,null,null,null,409.9074707,408.2958069,409.4621582,409.2029724,0,2.04230547,3.836602688,2.420581102,0,0,0,3.053434134,0,null,null,null,null,null,null,0,0,0,0,4277.513672,277.0079956,3804.040771,455.8079834,2.818999767,0,0,14.84599972,3804.040771,455.8079834,17.66499901,46.51399994,0.003,0,173.4449921,50.61599731,6.429999828,219.9589996,50.61899948,6.429999828]'
[sma   ] TRACE 2024/04/26 09:41:40 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/04/26 09:41:41 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/04/26 09:41:42 recv 192.168.111.85: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[warp  ] TRACE 2024/04/26 09:41:43 recv warp2/WDB/evse/low_level_state: '{"led_state":1,"cp_pwm_duty_cycle":1000,"adc_values":[3913,3457,134,137,1026,3809,134],"voltages":[12026,9086,-12336,-12310,826,12278,-12336],"resistances":[2611,219],"gpio":[true,false,false,false,true,true,false,true,false,true,false,false,false,false,false,true,true,false,false,true,false,false,false,false],"charging_time":0,"time_since_state_change":311305,"uptime":825109541,"time_since_dc_fault_check":79286804,"temperature":0,"phases_current":3,"phases_requested":3,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":0}'
[warp  ] TRACE 2024/04/26 09:41:43 recv warp2/WDB/meter/all_values: '[237.272583,236.0487366,235.7392578,0,0,0,0,0,0,0,0,0,0,0,0,1,1,1,-27.09719276,-49.6413269,82.67513275,236.4224701,0,0,0,0,0,1,0,49.9513588,4262.667969,14.84599972,46.51699829,230.4909973,4286.473145,18471.37891,null,null,null,null,null,null,409.9841309,408.6877441,409.7092285,409.5240479,0,2.04230547,3.836602688,2.420581102,0,0,0,4.122653008,0,null,null,null,null,null,null,0,0,0,0,4277.513672,277.0079956,3804.040771,455.8079834,2.818999767,0,0,14.84599972,3804.040771,455.8079834,17.66499901,46.51399994,0.003,0,173.4449921,50.61599731,6.429999828,219.9589996,50.61899948,6.429999828]'
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-26:
> Beim Umschalten von Aus -> Min+PV oder von PV -> Min+PV steht das System sehr lange auf Ladebereit. Warte auf Fahrzeug.

Wo sehe ich das im Logfile?

> Die Externe Freigabe der WARP2 wird aber nicht gesetzt.

Was heisst das?

callap, 2024-04-27:
> Was heisst das?
Die Warp wartet auf die externe Freigabe durch EVCC. Das ist im Log/UI der Wallbox ersichtlich. Ich hatte das Phänomen heute auch, konnte es hat nach einem Update aller Komponenten aber nicht mehr reproduzieren. 
@bbr111  Kurzfristig lässt sich die externe Steuerung an der Wallbox auch zurücksetzen. EVCC übernimmt anschließend wieder. Schau mal, ob's hilft 🍀

bbr111, 2024-04-27:
@callap 
Danke das hat geholfen.
Hab alles geupdatet, neugestartet, und die config in evcc auf die letzte Fassung in der Doku angepasst.
Jetzt geht's wieder.

closed 2024-04-27 by bbr111