evcc-io/evcc #13592: charger get phases: modbus: response data size '4' does not match request quantity '1'

snapscanii, 2024-04-26:
### Describe the bug

Dieser Fehler an einer Keba P30 X Series mit Phasenumschaltung kommt seit dem Firmwareupdate auf 0.1.17.2 mit der 0.1.17.1 lief alles tadellos. Der Fehler tritt bei mir im Büro (2 Wallboxen) und auch zu Hause auf. Bei alle drei Wallboxen wurde auf die 0.1.17.2 geupdatet. Ein downgrade der Firmware ist leider nicht möglich. Der Fehler führt zum Aufhängen der Wallboxsteuerung. Keine Regelung mehr möglich. Leider sehe ich in der debug Ausgabe unten keinen Fehler....

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2024/04/26 11:45:15 evcc 0.125.0 (1d4e6959)
[main  ] INFO 2024/04/26 11:45:15 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/26 11:45:15 using sqlite database: /home/rherre/.evcc/evcc.db
[main  ] ERROR 2024/04/26 11:45:48 creating vehicle my_car2porsche failed: cannot create vehicle type 'porsche': login failed: unexpected status 400
[lp-1  ] DEBUG 2024/04/26 11:45:50 charge total import: 2066.877kWh
[lp-2  ] DEBUG 2024/04/26 11:45:50 charge total import: 1842.958kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 5s # control cycle interval

sponsortoken: *****

meters:
- name: grid1
  type: template 
  template: solaredge-hybrid
  id: 1  
  host: 192.168.0.100  
  port: 1502
  usage: grid  
  modbus: tcpip   
  
#- name: my_grid #alte Konfiguration mit Tibber als grid
#  type: template
#  template: tibber-pulse
#  usage: grid
#  token: *****
#  homeid: ####### Optional 
 
- type: template
  template: solaredge-hybrid 
  id: 1  #bisher 1
  host: 192.168.0.100  
  port: 1502
  usage: pv  
  modbus: tcpip  
  name: pv1  
    
- type: template
  template: solaredge-hybrid 
  id: 2  #bisher 2
  host: 192.168.0.100  
  port: 1502
  usage: pv  
  modbus: tcpip  
[... 311 lines cut ...]
[lp-2  ] DEBUG 2024/04/26 11:41:09 pv timer inactive
[lp-2  ] DEBUG 2024/04/26 11:41:09 max charge current: 6.57A
[site  ] DEBUG 2024/04/26 11:41:09 ----
[lp-1  ] DEBUG 2024/04/26 11:41:09 charge power: 0W
[lp-2  ] DEBUG 2024/04/26 11:41:09 charge power: 1353W
[main  ] DEBUG 2024/04/26 11:41:09 lp Garage aussen at prio 3 gets additional 1353W from Garage innen at prio 2, total 1353W
[site  ] DEBUG 2024/04/26 11:41:11 pv power: 3283W
[site  ] DEBUG 2024/04/26 11:41:11 battery 1 power: -247W
[site  ] DEBUG 2024/04/26 11:41:11 battery 1 soc: 19%
[site  ] DEBUG 2024/04/26 11:41:12 battery 2 power: -334W
[site  ] DEBUG 2024/04/26 11:41:12 battery 2 soc: 20%
[site  ] DEBUG 2024/04/26 11:41:12 battery soc: 20%
[site  ] DEBUG 2024/04/26 11:41:12 battery power: -581W
[site  ] DEBUG 2024/04/26 11:41:12 grid meter: 62W
[site  ] DEBUG 2024/04/26 11:41:12 giving loadpoint priority for additional: 1353W
[site  ] DEBUG 2024/04/26 11:41:12 site power: -1772W
```

### What type of operating system are you running?

Linux

### Version

125.0

andig, 2024-04-26:
> Dieser Fehler an einer Keba P30 X Series mit Phasenumschaltung kommt seit dem Firmwareupdate auf 0.1.17.2 mit der 0.1.17.1 lief alles tadellos. 

Dann hat wohl KEBA etwas geändert- was? Gibts ein Ticket bei Keba? Kein evcc Thema...

andig, 2024-04-26:
> Der Fehler führt zum Aufhängen der Wallboxsteuerung. Keine Regelung mehr möglich. Leider sehe ich in der debug Ausgabe unten keinen Fehler....

Ich auch nicht- daher nicht nachvollziehbar. Was hängt sich da auf?

snapscanii, 2024-04-26:
Die Ansteuerung der Schalter Aus PV Min+PV bzw Schnell

Abhilfe schafft nur Evcc stoppen und neu starten..... aber dann geht es nach kurzer Zeit von vorne los. 

Ich bin noch weiter am ausprobieren.....verbringe schon Stunden damit, den Fehler einzugrenzen. Er tritt aber erst seit dem neuesten FW Update von KEBA auf.

Ich glaube insbesondere dann, wenn man von PV bzw Min+PV auf Schnell umschaltet. Dann hängt sich Evcc auf. Vielleicht reagiert der Phasenumschalter S10 auf die neue FW anders. Die Meldung mit dem Wert 4 verstehe ich auch nicht.

andig, 2024-04-26:
> Dann hat wohl KEBA etwas geändert- was? Gibts ein Ticket bei Keba? Kein evcc Thema...

Ich mache hier mal zu bis es dazu Klarheit gibt.

closed 2024-04-26 by andig

snapscanii, 2024-04-26:
[image]

Ojoachim71, 2024-04-26:
Mach doch mal bei Keba ein Ticket dafür auf. Ich hab das Problem auch.....

snapscanii, 2024-04-26:
Habe mit denen heute telefoniert.... bin mal gespannt, ob da was kommt. Mir würde schon helfen wenn jemand weiß wie man ein FW downgrade bei KEBA macht. Sobald ich was Neues weiß, gebe ich Bescheid.

Ojoachim71, 2024-04-26:
Ich bin gespannt....

snapscanii, 2024-04-27:
Fürs erste bleibt uns nur, die Steuerung der Wallbox über UDP zu regeln. Ich hoffe es kann uns noch jemand helfen. EVCC ist absolute Spitze 🤩🤩🤩