evcc-io/evcc #13600: Login Tapo P110 schlägt fehl 2.

dominikruffing, 2024-04-27:
### Describe the bug

siehe #13476

Hallo, seit dem aktuellsten Update bekommen ich nach einiger Zeitr die Fehlermeldung:
charge meter: request failed: expected 200 OK, got 403 Forbidden. Error message: für meine Steckdose Tapo P110.
Vor dem Update ging alles. Die Steckdose ist über die Tapo-App auch erreichbar und funktioniert, Zugangsdaten passen auch.

Ich konnte herausfinden, dass nach einem Neustart des Addons ca. 12h ohne Einschränkungen funktioniert, danach kommt die Fehlermeldung von oben. Starte ich das Addon neu, geht es von vorne los.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: tplink
    type: template
    template: tapo
    host: 
    user: 
    password: 
    standbypower: 15
    integrateddevice: true
```

### Log details

```text
charge meter: request failed: expected 200 OK, got 403 Forbidden. Error message:
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-27:
Ohne Logfile ist da nichts zu machen :(

closed 2024-04-27 by andig

thierolm, 2024-04-27:
Habe #13604 aufgemacht ...