evcc-io/evcc #13608: Tinkerforge WARP3 Loadpoint Panic Error

Logirius, 2024-04-27:
### Describe the bug

The Tinkerforge Warp3 Charger is successfully connected to the MQTT Broker. After it is configured in evcc.yaml as charger ant Tinkerforge-warp Template all is still ok. But after there is configured a loadpoint Carport with reference to the configured charger a fatal error appears and the homeassistant addon is restarting in a loop. 

### Steps to reproduce

1. Configure the charger in evcc.yaml
2. configure the loadpoint with these charger in evcc.yaml
3. start the homeassistant addon
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.XXX.XXX
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
[... 563 lines cut ...]
github.com/evcc-io/evcc/charger.(*Warp2).currents(0x2717ee0?)
	github.com/evcc-io/evcc/charger/warp2.go:274 +0x88
github.com/evcc-io/evcc/charger.(*decorateWarp2PhaseCurrentsImpl).Currents(0xc00117bb68?)
	github.com/evcc-io/evcc/charger/warp2_decorators.go:2728 +0x12
github.com/evcc-io/evcc/core.(*Loadpoint).updateChargeCurrents(0xc000cc8c08)
	github.com/evcc-io/evcc/core/loadpoint.go:1319 +0x92
github.com/evcc-io/evcc/core.(*Loadpoint).Update(0xc000cc8c08, 0x4099a00000000000, 0x0, 0x0, 0x0, 0x3ff0000000000000, 0xc001cf1df0, 0x0)
	github.com/evcc-io/evcc/core/loadpoint.go:1545 +0xff
github.com/evcc-io/evcc/core.(*Site).update(0xc0017deb60, {0x3808dd8, 0xc000cc8c08})
	github.com/evcc-io/evcc/core/site.go:790 +0x4e7
github.com/evcc-io/evcc/core.(*Site).Run(0xc0017deb60, 0xc0018c8900, 0x2540be400)
	github.com/evcc-io/evcc/core/site.go:910 +0x2f5
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:284 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:283 +0x1adf
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-04-28:
Closed in https://github.com/evcc-io/evcc/commit/07e7dedc009802f04e8f0faa37ec1f9f66046634

closed 2024-04-28 by andig

Logirius, 2024-05-04:
Hallo vielen herzlichen Dank für die schnelle Korrektur. 

Inzwischen wurde das neue Release in Homeassistant veröffentlicht und ich konnte es installieren. 

Leider kommt jetzt dieser Fehler 

[lp-1  ] ERROR 2024/05/04 09:16:27 charge currents: json: cannot unmarshal object into Go value of type []float64

Handelt es sich um einen Folgefehler oder ein neues Problem? In zweiterem Fall reporte ich einen neuen Issue. 

Viele Grüße

Jürgen

andig, 2024-05-04:
Meine Glaskugel ist zur Reparatur. Gibts ein Logfile?