evcc-io/evcc #13631: Meter reading for Innogy eBox (Professional?)

SamuelBrucksch, 2024-04-29:
In the Modbus documentation this was not documented, but according to https://www.goingelectric.de/forum/viewtopic.php?p=2215577#p2215577 it can be read:
[image]

Address is 307, value is float 2byte.

Is it possible to add that? That should make the charged Energy way more accurate, as the current consumption is calculated based on the current, which is a little bit off.

andig, 2024-04-29:
Did you try it? Are these readings available for all eBoxes?

SamuelBrucksch, 2024-04-29:
I think the eBox Professional can be ordered with and without energy meter. Not sure about the normal eBox. So my assumption is that only the one with energy meter provides any values here. I don't know what it returns if it doesn't.

I currently don't have a modbus connection to the wallbox, so I did not try it.

andig, 2024-04-30:
@SamuelBrucksch what exactly do you mean bei "current meter reading"? Power? Energy? Let's please be precise.

SamuelBrucksch, 2024-04-30:
The energy meter reading (charged kWhs) is precise, the power meter is a bit inaccurate. So the calculated consumption based on power is always a little off. The energy meter (register 307) could improve this.

andig, 2024-04-30:
Mal auf deutsch, vielleicht ist das klarer. Was *exakt* ist dieses Register: Zählerstand oder Sessionenergie?

SamuelBrucksch, 2024-04-30:
Aktueller Zählerstand.

andig, 2024-04-30:
Closed in https://github.com/evcc-io/evcc/commit/da4af01d736503b77901648d8a03f45516a99591

closed 2024-04-30 by andig

SamuelBrucksch, 2024-04-30:
Nice, thanks!

SamuelBrucksch, 2024-05-01:
@andig wie kann ich sehen, ob es funktioniert? Habe mir eben den nightly gezogen, aber in den denug logs finde ich nichts zum Zählerstand. Hier mal die Logs nach dem init:
```
[lp-1  ] DEBUG 2024/05/01 07:39:07 charge power: 0W
[site  ] DEBUG 2024/05/01 07:39:07 pv power: 2628W
[site  ] DEBUG 2024/05/01 07:39:07 battery soc: 79%
[site  ] DEBUG 2024/05/01 07:39:07 battery power: -2024W
[site  ] DEBUG 2024/05/01 07:39:07 grid meter: -17W
[site  ] DEBUG 2024/05/01 07:39:07 site power: -2041W
[lp-1  ] DEBUG 2024/05/01 07:39:07 charge currents: [0.15 0.19 0.15]A
[lp-1  ] DEBUG 2024/05/01 07:39:07 charger status: B
```

Oder wird das garnicht geloggt?

premultiply, 2024-05-01:
Erstmal alle Instanzen stoppen.

Dann einfach nur `evcc charger` aufrufen.

lukaslo, 2024-05-04:
Habe eben auf 0.125.0 downporten müssen. Mein Innogy-Charger verbindet sich nicht mehr, ich bekomme ein Connection Refused mit 0.126.0. Meine eBox ist mit einem Meter ausgestattet, Firmware: 1.3.55 :(
Mache ich etwas falsch? Muss man in der Config noch was ändern?

andig, 2024-05-04:
Verbindungsprobleme haben sicher nichts damit dazu tun.

> Mache ich etwas falsch? Muss man in der Config noch was ändern?

Meine Glaskugel ist zur Reparatur.

andig, 2024-05-06:
@lukaslo siehe https://github.com/evcc-io/evcc/issues/13771 und Asche auf mein Haupt. Es scheint als würde sich die Box nach dem falsch ausgelesenen Register komplett abzumelden. Bitte Entschuldige!

premultiply, 2024-05-28:
https://github.com/evcc-io/evcc/pull/14098