evcc-io/evcc #13633: Adding Support for AlphaESS SMILE-G3-EVCT11

DobbyNator94, 2024-04-29:
**Is your feature request related to a problem? Please describe.**
Would like to see support for wallboxes from AlphaESS

**Describe the solution you'd like**
Support for the AlphaESS SMILE-G3-EVCT11.

**Describe alternatives you've considered**
Nothing.

**Additional context**
Nothing.

Thank you for your great work! Keep it up!

premultiply, 2024-04-29:
Please provide all necessary interface specifications from manufacturer.

DobbyNator94, 2024-04-29:
Hey, as far as I know, the Wallbox is communicating over Modbus with the Alpha ESS Storion SMILE series.

OCPP 1.6 Json, Using WiFi AP Mode Settings
RS485

https://alphaess.de/Public/Uploads/uploadfile/files/20230926/AlphaESS-Handbuch_SMILE_ModBus_RTU_TCP_V21.pdf

premultiply, 2024-04-29:
OCPP is supported out of the box.
Just try it. Nothing to do here.

For Modbus we need the Modbus documentation of this charger, not the storage system.

closed 2024-04-30 by andig

MonsterEnergy-wtf, 2024-11-01:
So I found this thread which handles exactly the same problem for me.

My inverter is available in the confuration (alpha-ess-smile)
But how to I add my SMILE-G3-EVCT11 to EVCC? I'm more confused after reading this thread than before :)

So what to configure?
Has someone a working configuration or a setup which covers my need? Maybe I could adapt the configuration if someone is willing to help here.

Suggestions welcome.
Thanks,
Martin

MonsterEnergy-wtf, 2024-11-05:
All right folks.

I digged a bit deeper and got the following information.

1) Reboot the charger. It will spawn a network. Login to wifi and go to 192.168.4.1. 
2) Login with your PIN for your AlphaESS wallbox. In case you forgot it, get in touch with alphaESS ( in my case use mailto:info@alpha-ess.de). They are happy to help you with that.
3) Under "network" you can change the url to your EVCC instance. Like ws://<yourEVCCServer>:8887 and save it.
4) EVCC will now be able to communicate with the wall box.

My config for EVCC:
chargers:
  - name: wallbox
    type: template
    template: ocpp
    stationid: ALP-XXX # this is the serial of your charger

Unfortunately I get this:
Nov 05 21:59:39 evcc[3979]: [main  ] INFO 2024/11/05 21:59:39 evcc 0.131.4
Nov 05 21:59:39 evcc[3979]: [main  ] INFO 2024/11/05 21:59:39 using config file: /etc/evcc.yaml
Nov 05 21:59:39 evcc[3979]: [db    ] INFO 2024/11/05 21:59:39 using sqlite database: /var/lib/evcc/evcc.db
Nov 05 21:59:40 evcc[3979]: [main  ] INFO 2024/11/05 21:59:40 listening at :7070
Nov 05 21:59:40 evcc[3979]: [XXX-1] DEBUG 2024/11/05 21:59:40 waiting for chargepoint: 5m0s
Nov 05 21:59:45 evcc[3979]: [ocpp  ] DEBUG 2024/11/05 21:59:45 charge point connected: XXX
Nov 05 22:00:15 evcc[3979]: [XXX-1] DEBUG 2024/11/05 22:00:15 failed configuring availability: timeout
Nov 05 22:00:19 evcc[3979]: [ocpp  ] ERROR 2024/11/05 22:00:19 ocpp message (271): NotSupported - Unsupported feature SecurityEventNotification ([2,"271","SecurityEventNotification",{"type":"SettingSystemTime","timestamp":"2024-11-05T21:00:18Z"}])
Nov 05 22:00:23 evcc[3979]: [XXX-1] DEBUG 2024/11/05 22:00:23 failed configuring WebSocketPingInterval: NotSupported
Nov 05 22:00:25 evcc[3979]: [main  ] FATAL 2024/11/05 22:00:25 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'ocpp': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
Nov 05 22:00:25 evcc[3979]: [main  ] FATAL 2024/11/05 22:00:25 will attempt restart in: 15m0s
Nov 05 22:07:23 evcc[3979]: [ocpp  ] DEBUG 2024/11/05 22:07:23 charge point disconnected: XXX
Nov 05 22:07:24 evcc[3979]: [ocpp  ] DEBUG 2024/11/05 22:07:24 charge point connected: XXX
Nov 05 22:07:29 evcc[3979]: [ocpp  ] ERROR 2024/11/05 22:07:29 ocpp message (3): NotSupported - Unsupported feature SecurityEventNotification ([2,"3","SecurityEventNotification",{"type":"SettingSystemTime","timestamp":"2024-11-05T21:07:28Z"}])

Does anyone have any idea what I'm doing wrong?
@premultiply maybe?

andig, 2024-11-05:
The final error is obvious:

> cannot create charger type 'ocpp': sponsorship required

MonsterEnergy-wtf, 2024-11-05:
@andig Holy crap. I completely overlooked this. 

All right. Happy to sponsor this project too, good software deservers sponsoring!
Hopefully my issue will then be gone. 

Thanks!

Rasenkardinal, 2024-12-08:
> @andig Holy crap. I completely overlooked this.
> 
> All right. Happy to sponsor this project too, good software deservers sponsoring! Hopefully my issue will then be gone.
> 
> Thanks!

So, does it work now?

andig, 2024-12-08:
See https://docs.evcc.io/docs/sponsorship

Rasenkardinal, 2024-12-08:
> See https://docs.evcc.io/docs/sponsorship

Sorry, but I don't understand. Do I need sponsorship Token to make it work?

MonsterEnergy-wtf, 2024-12-08:
Nope. Doesn't work as the wallbox simply ignores requests to it via OCPP. I already opened s ticket at AlphaESS for this and it turned out that the box is stupid as hell and ignores incoming requests by design.

Long story short: I removed EVCC and when I replace the wallbox in future, I will consider to buy one which is supported by out of the box by EVCC.

Thanks everyone to help me clearing this up.

MonsterEnergy-wtf, 2024-12-08:
If you want a wallbox to communicate via OCPP then yes - you need a (paid) token.
If you want to communicate with Modbus or TCP direktly with the wallbox then you are good to go.