evcc-io/evcc #13642: Reduktion der Phasenumschaltungen durch Nutzung der Batterie

thse22, 2024-04-29:
Ich schlage vor im Bereich Batterieunterstütztes Laden (butterybuffered) bei automatischer Phasenumschaltung das Downscaling 3p->1p auszusetzen. Bei meiner Anlage ergeben sich ansonsten trotz erhöhten Delay entweder häufige Phasenumschaltungen oder eine Netzeinspeisung. Das Verhalten könnte man mit einem Konfigurationsparameter im yaml ggf. einstellbar machen, da es vl. andere Szenarien gibt, wo man die Batterie dafür nicht nutzen möchte.

thse22, 2024-04-30:
Phasenumschaltung an sich ist erstmal gut. Es geht mir darum eine größere Hysterese zwischen 1p und 3p zu haben um häufige Umschaltungen zu vermeiden. Das aber nur in dem Bereich wo die Batterie zum Einsatz kommen soll. Beim 1p auf 3p klappt das auch mit enable Delay. Mit erhöhtem Disable Delay hatte dennoch den Effekt dass auf 1p zurückgeschaltet wird und dann die Batterie "zu" voll wird. Vl. Habe ich eine ungünstigen Arbeitspunkt erwischt. 
Was meinst du mit extern umsetzen? 
Ich kann ja erstmal bei mir mehr Erfahrung sammeln...

thse22, 2024-04-30:
Premiltiply:
Stimmt auch hatte schon mal in die Richtung Force Discharge gedacht wo man Max Wechselrichterleistung und Max Entladeleistung der Batterie berücksichtigen würde.
Dann hätte man oberhalb butterybuffered 2 Modi: MinPv oder ForceDischarge

Vl mal nur eine Idee wenn man die Batterie noch stärker in das Ladekonzept integrieren möchte. Diese werden ja auch immer größer...

naltatis, 2024-04-30:
Gerade in den letzten Tagen habe ich den Effekt auch beobachtet. Ein Verhalten, wo erstmal der freigegebene Buffer verwendet wird, bevor er von 3p auf 1p zurückschaltet, fände ich gut.

> Hier bräuchte es wenn dann schon Regeln die für alle Szenarien funktionieren und idealerweise ohne weitere Konfigurationsoptionen auskommen.

Wäre hier nicht schon ausreichen, auf den Netzbezug zu schauen und die Batterieleistung zu akzeptieren so lange sie im Bufferbereich ist?

thse22, 2024-04-30:
Schau ich mir heute Abend mal an ob man da eine Regel definieren kann mit der Idee kein Netzbezug und trotzdem stärkere Batterienutzung vor Phasenzurückschaltung...

premultiply, 2024-04-30:
_buffer_ heisst ja heute nur, dass einfach nur nicht abgeschaltet wird (wie Min+PV-Modus) wenn der SoC im Bufferbereich liegt. Sonst ist da keinerlei Magie oder Logik dahinter. Alles andere funktioniert unabhängig davon wie sonst auch.

naltatis, 2024-04-30:
> buffer heisst ja heute nur

Ja, heute ist das so. Ich fänd das aber durchaus nachvollziehbar/erwartungskonform den gleichen Buffer nicht nur für das Abschalten, sondern auch für das Umschalten zu verwenden. Bei den Delays ist es ja auch so, dass für 1p3p keine separaten Werte eingeführt haben, sondern die Disable Werte auch verwenden, um von 3p auf 1p zu wechseln.

Wäre mMn. zumindest eine Überlegung Wert. Dann hätte man das gewünschte Verhalten, ohne eine neue Einstellung einzuführen.

thse22, 2024-05-01:
In loadpoint habe ich die Gridpower nicht gefunden, daher gebe ich den Parameter von sito.go and loadpooint über die diversen Interfaces weiter. 
Zur Entscheidung ob Phaseswitch ausgesetzt wird habe ich überlegt:  batterybuffered && (gridPower < lp.GetDisableThreshold()) zu verwenden, da ein kleiner Netzbezug ja evtl. vorhanden sein könnte.

Schaut mal ob das in eurem Sinne ist (noch nicht getestet):
https://github.com/evcc-io/evcc/compare/master...thse22:evcc:delayed_phase_switch

andig, 2024-05-04:
Mir wäre die Änderung viel zu invasiv für zu kleinen Nutzen.

andig, 2024-05-08:
Klingt als würden wir das so nicht umsetzen wollen- closing.

closed 2024-05-08 by andig