evcc-io/evcc #13649: EVCC Startet ladevorgang nicht (Easee) - "Warte auf Fahrzeug"

chiefhood, 2024-04-30:
### Describe the bug

Hallo,
ich habe seit paar Wochen das Problem, dass EVCC immer anzeigt "Warte auf Fahrzeug" und meine Easee Box ebenfalls sagt "Warte auf Fahrzeug". Wenn ich jedoch in EVCC den Lademodus kurz auf "Aus" und wieder auf "PV" setze. Startet der Ladevorgang.
Kennt das Problem jemand?
Meine Easee Homes sind mit WLAN verbunden (gutes Signal).

Der log zeigt heute Morgen (Ladung wurde nicht gestzartet + Neustart EVCC um 10:50 Uhr).

Grüße!

### Steps to reproduce

1. warten bis Haus Batterie an oder ab und an wenn Fahrzeug neu verbunden wird

### Configuration details

```yaml
sponsortoken: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.1.50
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

# unique installation id
plant: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

interval: 20s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
[... 733 lines cut ...]

2024-04-30T07:25:43.403803653Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 grid powers: [-1089 -1184 -1070]W

2024-04-30T07:25:43.403760800Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 grid meter: -3344W

2024-04-30T07:25:43.403708824Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 battery power: -1515W

2024-04-30T07:25:43.403521836Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 battery soc: 94%

2024-04-30T07:25:43.257730988Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 pv power: 5211W

2024-04-30T07:25:43.179124092Z,stdout,[lp-2  ] DEBUG 2024/04/30 09:25:43 charge power: 0W

2024-04-30T07:25:43.179042900Z,stdout,[lp-1  ] DEBUG 2024/04/30 09:25:43 charge power: 0W

2024-04-30T07:25:43.178392841Z,stdout,[site  ] DEBUG 2024/04/30 09:25:43 ----
```

### What type of operating system are you running?

Docker container

### Version

1.2.5.0

chiefhood, 2024-04-30:
Hier der Log nach dem restart um 10:50 Uhr.

2024-04-30T08:51:26.491841427Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:26 charge power: 0W

2024-04-30T08:51:26.491231202Z,stdout,[site  ] DEBUG 2024/04/30 10:51:26 ----

2024-04-30T08:51:12.878339369Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:12 switched phases: 3p

2024-04-30T08:51:12.835114508Z,stdout,"[lp-1  ] WARN 2024/04/30 10:51:12 charger logic error: phases mismatch (got 0, expected 3)
"
2024-04-30T08:51:11.688046414Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 charge total import: 444.843kWh

2024-04-30T08:51:11.688002254Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 wake-up timer: start

2024-04-30T08:51:11.687957840Z,stdout,[lp-1  ] INFO 2024/04/30 10:51:11 stop charging <-

2024-04-30T08:51:11.687915503Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 charger status: B

2024-04-30T08:51:11.687871575Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 charge total import: 444.843kWh

2024-04-30T08:51:11.687825191Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 charge currents: [0.006 0.005 0.008]A

2024-04-30T08:51:11.687780985Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 site power: -6563W

2024-04-30T08:51:11.687731295Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 grid currents: [-9.72 -10.4 -9.3]A

2024-04-30T08:51:11.687655248Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 grid powers: [-2326 -2504 -2234]W

2024-04-30T08:51:11.687610784Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 grid meter: -7063W

2024-04-30T08:51:11.687559913Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 battery power: 0W

2024-04-30T08:51:11.687394976Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 battery soc: 100%

2024-04-30T08:51:11.568325535Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 pv power: 7830W

2024-04-30T08:51:11.492045123Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:51:11 charge power: 0W

2024-04-30T08:51:11.491968938Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:51:11 charge power: 0W

2024-04-30T08:51:11.491341538Z,stdout,[site  ] DEBUG 2024/04/30 10:51:11 ----

2024-04-30T08:50:56.695291889Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 set session energy limit: 0

2024-04-30T08:50:56.695236298Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 set session soc limit: 0

2024-04-30T08:50:56.695181102Z,stdout,[lp-2  ] INFO 2024/04/30 10:50:56 car disconnected

2024-04-30T08:50:56.695137064Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 charger status: A

2024-04-30T08:50:56.695079581Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 charge total import: 48.758kWh

2024-04-30T08:50:56.695024371Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 charge currents: [0 0 0]A

2024-04-30T08:50:56.694974503Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 site power: -6570W

2024-04-30T08:50:56.694910166Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 grid currents: [-9.74 -10.4 -9.29]A

2024-04-30T08:50:56.694849880Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 grid powers: [-2333 -2504 -2233]W

2024-04-30T08:50:56.694788470Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 grid meter: -7070W

2024-04-30T08:50:56.694586851Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 battery power: 0W

2024-04-30T08:50:56.694382837Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 battery soc: 100%

2024-04-30T08:50:56.567213497Z,stdout,[site  ] DEBUG 2024/04/30 10:50:56 pv power: 7830W

2024-04-30T08:50:56.492045104Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:56 charge power: 0W

[... 106 lines cut ...]
2024-04-30T08:50:41.472558980Z,stdout,[site  ] INFO 2024/04/30 10:50:41     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗

2024-04-30T08:50:41.472518133Z,stdout,[site  ] INFO 2024/04/30 10:50:41   vehicles:

2024-04-30T08:50:41.472468212Z,stdout,[site  ] INFO 2024/04/30 10:50:41     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗

2024-04-30T08:50:41.472423157Z,stdout,[site  ] INFO 2024/04/30 10:50:41     pv 1:      power ✓ energy ✓ currents ✗

2024-04-30T08:50:41.472366192Z,stdout,[site  ] INFO 2024/04/30 10:50:41     grid:      power ✓ energy ✓ currents ✓

2024-04-30T08:50:41.472285562Z,stdout,[site  ] INFO 2024/04/30 10:50:41   meters:      grid ✓ pv ✓ battery ✓

2024-04-30T08:50:41.472118550Z,stdout,[site  ] INFO 2024/04/30 10:50:41 site config:

2024-04-30T08:50:41.456952508Z,stdout,[lp-2  ] DEBUG 2024/04/30 10:50:41 charge total import: 48.758kWh

2024-04-30T08:50:41.454845983Z,stdout,[lp-1  ] DEBUG 2024/04/30 10:50:41 charge total import: 444.843kWh

2024-04-30T08:50:41.152184996Z,stdout,[main  ] ERROR 2024/04/30 10:50:41 creating vehicle Mercedes failed: cannot create vehicle type 'custom': soc: missing configuration

2024-04-30T08:50:38.613829594Z,stdout,[main  ] INFO 2024/04/30 10:50:38 starting ui and api at :7070

2024-04-30T08:50:38.611525398Z,stdout,[main  ] INFO 2024/04/30 10:50:38 using config file: /etc/evcc.yaml

2024-04-30T08:50:38.608950601Z,stdout,[main  ] INFO 2024/04/30 10:50:38 evcc 0.125.0

andig, 2024-04-30:
Bekannter Bug der Easee- bitte Suchfunktion benutzen. /cc @GrimmiMeloni