evcc-io/evcc #13663: Automatische Übernahme des Ladelimits vom Fahrzeug

Captainbrille, 2024-05-01:
Von meinem Fahrzeug kann das Ladelimit ausgelesen werden und dieses wird auch angezeigt. Ich nutze ausschließlich die Ladelimitplanung vom Fahrzeug, daher wäre es hilfreich wenn es in EVCC in der Ladeplanung die Option gibt das Limit vom Fahrzeug automatisch zu übernehmen. Dadurch stimmt z.B. die Angabe der Reichweite bei der Zielladung.

andig, 2024-05-01:
/cc @naltatis im Moment zeigen wir das zusätzlich an und mit Warnung. Hier könnte man sowas wie ein effective limit machen?

naltatis, 2024-05-04:
Das Thema ist leider etwas komplizierter. Bei einigen Fahrzeugen können wir das Fahrzeug-eigene Ladelimit lesen, bei einigen (Tesla) könnte man es auch setzen. Was wäre denn deine Erwartung, wie sich die UI verhält, wenn wir "das Ladelimit vom Fahrzeug übernehmen"? Kein Slider mehr, bzw. nur noch Read-only? Wenn wir das Fahrzeug-Limit aus evcc heraus auch änderbar machen wollen, müssen wir sicherstellen, dass das immer in-sync ist. Wir haben auch Fälle, wo wir das Ladelimit temporär außer Kraft setzen (bspw. bei einem Ladeplan).

Aus diesen Gründen ist es aktuell eine bewusste Entscheidung, dass wir das Fahrzeug-Limit nur anzeigen, ggf. warnen, aber nicht ändern. Siehe auch hier:
https://docs.evcc.io/docs/features/limits#fahrzeugeigenes-limit

Captainbrille, 2024-05-05:
Vielen Dank, dass ihr euch dem annehmt. Grundsätzlich finde ich die Ladeplanung übersichtlich und gut gestaltet  in meinem Fall (Cupra Born) gibt es aber mindestens einen Fall der keinen Sinn macht. Wenn das Fahrzeug auf z.B. 80% begrenzt ist, EVCC auf z.B. 90%, dann hangelt man sich von Begrenzung zu Begrenzung, falls man doch mal auf 100% laden will. Des weiteren fände ich es interessant unter Ladelimit das aktuell wirksame Ladelimit zu sehen.
Ich persönlich bevorzuge die Ladeplanung ausschließlich über das Fahrzeug zu gestalten und kenne leider nicht die Vor- und Nachteile anderer Wallboxen und Fahrzeuge. Vielleicht macht folgendes Sinn:

1. Konfigurationsoptionen: Ladeplanung KFZ; Ladeplanung EVCC
    a) Ladeplanung KFZ: EVCC Schieber übernimmt, falls vorhanden, automatisch den hinterlegten Wert vom KFZ. Einen größeren Wert kann man nicht einstellen, wenn EVCC dies nicht im Fahrzeug setzten kann.
   b) Ladeplanung wie bisher 

Macht das Sinn?

naltatis, 2024-05-06:
> a) Ladeplanung KFZ: EVCC Schieber übernimmt, falls vorhanden, automatisch den hinterlegten Wert vom KFZ. Einen größeren Wert kann man nicht einstellen, wenn EVCC dies nicht im Fahrzeug setzten kann.

Ich glaube ganz hab ich die Idee noch nicht verstanden. Aber ich probier mal: Du möchtest bei einer Ladeplanung kein festes Ziel in evcc eingeben, sondern hier das aktuell eingestellte Limit im Fahrzeug verwenden? Also so etwas:

[image]

Das könnte dann so etwas wie `plan-soc: "auto"` (anstatt `80`) sein. Die Ladeplanung würde dann das zuletzt bekannte Vehicle-Limit für die Planung verwenden. Problem könnte sein, dass wir im Standardfall die Daten nur während der Ladung aktualisieren. Wenn du also nach dem Anstecken noch mal das Fahrzeuglimit erhöhst, bekommen wir das heute nicht immer mit. Es sein denn, du hast Polling für das Fahrzeug aktiviert (battery drain). Hier könnte man natürlich smart sein und evcc das Fahrzeug in plausiblen/sicheren Abständen vor dem Ladeziel noch mal aufwecken lassen.
Geht das in die richtige Richtung?

@andig Für den Planungsalgorithmus wäre es ohnehin eine gute Idee, wenn er den kleinsten Wert aus `plan soc` und `vehicle limit` zur Berechnung annehmen würde. Falls man die Warnung in der UI ignoriert könnte dann noch ein besserer Zeitraum gewählt werden.

closed 2024-06-01 by github-actions[bot]