evcc-io/evcc #13674: Wallbox Pulsar Max is not included in documentation but is supported

teun95, 2024-05-01:
### Describe the bug

The Wallbox Pular Max works well with EVCC using the `pulsarplus` template, but the documention does not mention this model. 

Given that all chargers by the brand "Wallbox" work with the same template I took a gamble and bought the Pulsar Max. See log below to see it working. It works well, with one exception explained below.

**Eco-smart energy meter**
In [this discussion](https://github.com/evcc-io/evcc/discussions/10981) @andig mentions that a modbus meter connected to the Wallbox Pulsar Max would result in the data being reported by the EV charger. In my debug log I see nothing that would indicate that this is the case. I use the [N1CT](https://support.wallbox.com/en/knowledge-base/datasheet-n1-ct/) meter which has a CT clamp that is set up to measure the connection to the grid. It is recognized by the charger and is visible in the manufacturer's app. Given that the Pulsar Max is not mentioned in the documentation, andig's comment might have been about other chargers from the "Wallbox" manufacturer. In any case, the point is that the Pulsar Max works, but a note in the documentation about the modbus meters not being supported would be useful. Of course, it would also be possible to create a separate issue to fix/introduce support for this.

### Steps to reproduce

Given that all chargers by the brand "Wallbox" work with the same template I took a gamble and bought the Pulsar Max and followed the instructions in [the documentation](https://docs.evcc.io/en/docs/devices/chargers#wallbox-pulsar-plus-commander-2-copper-sb).

```
Switch on “Enable OCPP” (myWallbox app) or enable the “OCPP WebSocket connection” switch (myWallbox Portal)
Enable “Improved charger control” (Profile -> Experimental functions) (myWallbox app)
URL: ws://[evcc-adresse]:8887/ (local network connection)
Charge Point Identity: Custom value (e.g. serial number of charger) which is reused in configuration as stationid
```

### Configuration details

```yaml
chargers:
  - type: template
    template: pulsarplus
    timeout: 2m
    name: wallbox4

loadpoints:
  - title: home
    charger: wallbox4
    mode: pv
    vehicle: leaf
    soc:
      estimate: true

```
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/05/01 15:21:53 evcc 0.125.0
[main  ] INFO 2024/05/01 15:21:53 using config file: /config/evcc.yaml
[main  ] INFO 2024/05/01 15:21:53 starting ui and api at :7070
[db    ] INFO 2024/05/01 15:21:53 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/05/01 15:21:53 connecting evcc-2107605143 at tcp://192.168.1.73:1883
[mqtt  ] DEBUG 2024/05/01 15:21:53 tcp://192.168.1.73:1883 connected
[ocpp-1] DEBUG 2024/05/01 15:21:53 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/05/01 15:22:55 charge point connected, registering: yellow_wallbox
[site  ] INFO 2024/05/01 15:23:04 site config:
[site  ] INFO 2024/05/01 15:23:04   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/05/01 15:23:04     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/05/01 15:23:04     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/05/01 15:23:04     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/05/01 15:23:04   vehicles:
[site  ] INFO 2024/05/01 15:23:04     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/05/01 15:23:04 loadpoint 1:
[lp-1  ] INFO 2024/05/01 15:23:04   mode:        pv
[lp-1  ] INFO 2024/05/01 15:23:04   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/05/01 15:23:04   meters:      charge ✗
[lp-1  ] DEBUG 2024/05/01 15:23:04 set smart cost limit: 0
[lp-1  ] DEBUG 2024/05/01 15:23:04 phase timer inactive
[lp-1  ] DEBUG 2024/05/01 15:23:04 pv timer inactive
[lp-1  ] INFO 2024/05/01 15:23:04 vehicle updated: unknown -> Nissan Leaf
[lp-1  ] DEBUG 2024/05/01 15:23:04 set charge mode: pv
[site  ] DEBUG 2024/05/01 15:23:04 ----
[lp-1  ] DEBUG 2024/05/01 15:23:04 charge power: 0W
[site  ] DEBUG 2024/05/01 15:23:04 pv power: 623W
[site  ] DEBUG 2024/05/01 15:23:04 grid meter: -294W
[site  ] DEBUG 2024/05/01 15:23:04 site power: -294W
[lp-1  ] DEBUG 2024/05/01 15:23:04 charger status: B
[lp-1  ] INFO 2024/05/01 15:23:04 car connected
[lp-1  ] DEBUG 2024/05/01 15:23:04 pv timer elapse
[lp-1  ] DEBUG 2024/05/01 15:23:04 pv timer inactive

```
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-05-01:
Closed in https://github.com/evcc-io/evcc/commit/8bbf193781f11b4c0ccc434da61b5419e676c154

closed 2024-05-01 by andig

teun95, 2024-05-01:
@andig could you confirm what the expected behaviour is when an energy meter is connected directly to Wallbox EV chargers? Then I know whether to open a separate issue for this. At the moment I am not sure whether this would be an enhancement or bug.

andig, 2024-05-01:
It would be expected to work!