evcc-io/evcc #13677: Add support for the Glow in home display, allowing for local UK smart meter power readings

teun95, 2024-05-01:
UK smart meters don't have a P1 port and excess energy monitoring tends to rely on CT clamps, which people in the UK would generally have installed by an electrician. UK households are supplied for free with a display that shows live energy usage and some statistics which it received over a closed Zigbee connection with the energy meter. These free displays don't have any external connections and can't be used to share data locally.

[The Glow in home display (IHD)](https://shop.glowmarkt.com/products/display-and-cad-combined-for-smart-meter-customers) replaces the free display and has wifi and support for MQTT. when configured with MQTT server details, it publishes the smart meter data over MQTT every 10 seconds. I am quite sure that it is the only product that allows access to live smart meter data to UK smart meters. Another anticipated device, the [Octopus Home Mini](https://octopus.energy/blog/octopus-home-mini/), that performs a similar task does not have local support as far as I can see.

When I ordered the product I was asked to first install their bright app and add the smart meter there. They then pre-programmed the display and shipped it to me. All that was required then was entering wifi and MQTT server details. I used the MQTT broker addon for Home Assistant as MQTT server. 

The MQTT topic is: `glow/F008D1ECE***` (removed the last three letters) where the sub-topic is the device's wifi MAC address without colons which can be found in the display settings, in the Bright app, and in the router. 

Published on `glow/F008D1ECE***/SENSOR/electricitymeter`

```

{
    "electricitymeter": {
        "timestamp": "2024-05-01T15:00:34Z",
        "energy": {
            "export": {
                "cumulative": 631.267,
                "units": "kWh"
            },
            "import": {
                "cumulative": 2400.895,
                "day": 4.047,
                "week": 22.504,
                "month": 4.047,
                "units": "kWh",
                "mpan": "not available",
                "supplier": "P1",
                "price": {
                    "unitrate": 0.2525,
                    "standingcharge": 0.4785
                }
            }
        },
        "power": {
            "value": -0.078,
            "units": "kW"
        }
    }
}

```

Please note that the `unitrate` might work well for some tariffs, but it does not work well for smart tariffs. I am on Octopus Cozy and it the `unitrate` is does not change with the actual tariff.

While MQTT is the only local option, it seems quite easy to add support for this meter. It can already be supported by using a custom meter in `evcc.yaml` (I don't do this as I re-publish the data in Home Assistant, also using MQTT). Having native support would make adding the meter easier and would add a lot of value for UK users of EVCC.

VolkerK62, 2024-05-01:
not sure, if this will be supported.

>The MQTT topic is: glow/F008D1ECE***

So this will be the default topic after enabling MQTT?

Can you check, if this will work:
```
- name: glow
  type: custom
  power:
    source: mqtt
    topic: glow/F008D1ECE***
    jq: .electricitymeter.power.value
    scale: 1000
    timeout: 30s
```

The template shoud be quit simple
```
- name: my_grid
  mac: F008D1ECE***
```

andig, 2024-05-02:
@teun95 happy to add support. Would you want to open a PR for adding the required template?

closed 2024-05-10 by andig