evcc-io/evcc #13679: Heating: When set to Off old LimitSoc is used as new target temperature

docolli, 2024-05-02:
### Describe the bug

Have a loadpoint/charger configured as standard charger and set a standard limitSoc via the GUI. Then reconfigure this charger with `heating: true`. Now every time you set the loadpoint to "Off" the old limitSoc is used as new target temperature.

I would expect the heating loadpoint just keeps the target temperature when set to "Off".

### Steps to reproduce

1. Define charger without `heating: true`
2. Go to plan and set a standard target Soc (e.g. 60%) and set to PV mode
3. Redefine the charger with `heating: true`
4. Set target temperature to 25°C.
5. Now change mode to "Off"
6. Target temperature will be set to 60°C
...

### Configuration details

```yaml
loadpoints:
- title: Badheizkörper
  charger: mystrom2
  vehicle: heating1

chargers:
- name: mystrom2
  type: template
  template: mystrom
  host: 192.168.1.142
  standbypower: 15
  integrateddevice: true
  heating: true

vehicles:
- name: heating1
  type: custom
  title: Heizstab
  icon: heater
  soc:
    source: http
    uri: http://192.168.1.142/report
    method: GET
    headers:
      - content-type: application/json
    insecure: false
    jq: .temperature
    timeout: 10s
```

### Log details

```text
provided, if needed
```

### What type of operating system are you running?

Linux

### Version

0.125.0

VolkerK62, 2024-05-02:
Ich würde sagen, works as designed.
Du hast das Fahrzeug fest dem Loadpoint zugewiesen.
Durch "off" wird es "abgesteckt" und durch die feste Zuweisung "direkt wieder verbunden" und dann wirkt der eingestellte limitsoc.

docolli, 2024-05-02:
Naja, dann sollte ich auch Zugriff auf limitSoc bei einem Heating loadpoint/charge bekommen um das einstellen zu können. 😉
So ist das inkonsistent. 
Mir schon klar, dass das Heating-Feature noch eine Baustelle ist und ich erwarte noch keine Perfektion. Ist mir halt beim Benutzen aufgefallen, da ich ab und zu den Heizstab manuell ausmache und er mir dann immer wieder ungewollt auf 60°C gesprungen ist.

Entweder wirkt der limitSoc bei einem Heating loadpoint nicht, oder man bekommt Zugriff auf den Wert, der bei "off" gesetzt wird.

closed 2024-05-28 by github-actions[bot]