evcc-io/evcc #13680: Renault total mileage no longer shown in charging logbook

felix-schwarz, 2024-05-02:
### Describe the bug

I noticed that for some time now the total mileage of the vehicle is no longer shown in the charging logbook.

### Steps to reproduce

1. Charge car.
2. Check charging logbook

I assume [`totalMileage`](https://github.com/evcc-io/evcc/blob/81d677b897df7c8c182dd1a470d193f8177d9e06/vehicle/renault/kamereon/types.go#L67) is either no longer returned by Renault - or under a different name.

### Configuration details

```yaml
vehicles:
  - name: eauto
    type: renault
    title: E-Auto
    capacity: 22
    user: ***
    password: ***
    onIdentify:
      mode: pv
```

### Log details

```text
I can't provide this atm.
```

### What type of operating system are you running?

Linux

### Version

0.125.0

andig, 2024-05-02:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-05-02 by andig

felix-schwarz, 2024-05-03:
Sorry for that. I opened a new report after retrieving the log: #13701.