evcc-io/evcc #13683: 0.126 will not start under hass.io

andreasluft76, 2024-05-02:
### Describe the bug

After Upddate from 0.125 to 0.126 add-on under hass.io (running on Synology DS1522 in VM) will not start

### Steps to reproduce

1. did full restart hass.io
2. 
3.
4.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 77d0cca2f6feb6d2ec67acfbfd380f6cedd728f87dce71d0367c85e028f6d553

interval: 10s # control cycle interval

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 192.168.178.197  
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 192.168.178.197  
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 192.168.178.197  
  name: battery3

chargers:
- type: template
  template: keba 
  host: 192.168.178.169  
  serial: XXX
  name: wallbox6

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3368 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

vehicles:
- type: template
  template: hyundai 
  title: Ioniq5  
  icon: car  
  user: XXX  
  password: XXX  
  vin: XXX
  capacity: 72  
  phases: 3  
  cache: 15m  
  language: de  
  mode: minpv  
  minCurrent: 12  
  maxCurrent: 16  
  identifiers:
  - 16 
  name: ev4
- type: template
  template: citroen 
  title: Citroen  
  icon: car  
  user: XXX
  password: XXX  
  vin: XXX
  capacity: 12  
  phases: 1  
  cache: 15m  
  mode: minpv  
  minCurrent: 12  
  maxCurrent: 16  
  name: ev5

loadpoints:
- title: Vorgarten
  charger: wallbox6
  mode: minpv

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/05/02 12:16:13 evcc 0.126.0
[main  ] INFO 2024/05/02 12:16:13 using config file: /config/evcc.yaml
[main  ] INFO 2024/05/02 12:16:13 starting ui and api at :7070
[db    ] INFO 2024/05/02 12:16:13 using sqlite database: /data/evcc.db
[main  ] FATAL 2024/05/02 12:16:14 cannot create vehicle 'ev5': cannot create vehicle type 'template': invalid key: cache
[main  ] FATAL 2024/05/02 12:16:14 will attempt restart in: 5m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/05/02 12:22:20 evcc 0.126.0
[main  ] INFO 2024/05/02 12:22:20 using config file: /config/evcc.yaml
[main  ] INFO 2024/05/02 12:22:20 starting ui and api at :7070
[db    ] INFO 2024/05/02 12:22:20 using sqlite database: /data/evcc.db
[main  ] FATAL 2024/05/02 12:22:21 cannot create vehicle 'ev5': cannot create vehicle type 'template': invalid key: cache
[main  ] FATAL 2024/05/02 12:22:21 will attempt restart in: 5m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/05/02 12:33:04 evcc 0.126.0
[main  ] INFO 2024/05/02 12:33:04 using config file: /config/evcc.yaml
[main  ] INFO 2024/05/02 12:33:04 starting ui and api at :7070
[db    ] INFO 2024/05/02 12:33:04 using sqlite database: /data/evcc.db
[main  ] FATAL 2024/05/02 12:33:06 cannot create vehicle 'ev5': cannot create vehicle type 'template': invalid key: cache
[main  ] FATAL 2024/05/02 12:33:06 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

goebelmeier, 2024-05-02:
Source of the problem is written in the logs:
`[main  ] FATAL 2024/05/02 12:33:06 cannot create vehicle 'ev5': cannot create vehicle type 'template': invalid key: cache`

You need to remove the `cache: 15m` from your Citroen vehicle.

```vehicles:
- type: template
  template: hyundai 
  title: Ioniq5  
  icon: car  
  user: XXX  
  password: XXX  
  vin: XXX
  capacity: 72  
  phases: 3  
  cache: 15m  
  language: de  
  mode: minpv  
  minCurrent: 12  
  maxCurrent: 16  
  identifiers:
  - 16 
  name: ev4
- type: template
  template: citroen 
  title: Citroen  
  icon: car  
  user: XXX
  password: XXX  
  vin: XXX
  capacity: 12  
  phases: 1  
  cache: 15m  
  mode: minpv  
  minCurrent: 12  
  maxCurrent: 16  
  name: ev5```

StefanSchoof, 2024-05-02:
But I thinks this was an unintended effect of #13612.

closed 2024-05-02 by andreasluft76