evcc-io/evcc #13687: e3dc rscp integration battery capacity

Cheppetto, 2024-05-02:
### Describe the bug

looks like the integration only reads one stack of batteries not additional groups

### Steps to reproduce

1. setup battery config for native e3dc
2. unable to set battery capacity manual
3. battery capacity shows 13.1 instead of 26.2 kWh (brutto)
...

### Configuration details

```yaml
configured via UI, no reports in the log
```

### Log details

```text
Homeassistant installation - I got another integration running which also utilizes RSCP for this one the battery capacity is correct - could provide RSCP logs from that one
```

### What type of operating system are you running?

Linux

docolli, 2024-05-02:
Yes, currently the implementation reads only one battery group. If `battery` is not defined, then the value of batteryID=0 is read.
Good to see that somebody is here that has more than 1 battery.
Can you run some linux commands to help us further to implement also E3DC Systems with more than 1 battery?

Cheppetto, 2024-05-02:
I am running evcc as home assistant add-on.
(Evcc in docker inside a VM running on a Linux node) 

I am familiar with Linux, so we can give it a shot.
Could also send you battery output from another rscp integration, if that helps

docolli, 2024-05-02:
@andig 
Es gibt den Tag BAT_REQ_AVAILABLE_BATTERIES, über den könnte man abfragen wieviel Batteriespeicher im System vorhanden sind:

```json
./e3dc '[["BAT_REQ_AVAILABLE_BATTERIES","",""]]' | jq
{
  "BAT_AVAILABLE_BATTERIES": {
    "BAT_BATTERY_SPEC": {
      "BAT_INDEX": 0,
      "BAT_PARAM_BAT_NUMBER": 0,
      "BAT_DEVICE_NAME": "ATL_BAT",
      "BAT_MANUFACTURER_NAME": "ATL",
      "BAT_SERIALNO": 1,
      "BAT_INSTANCE_DESCRIPTOR": "Battery_ATL_BAT_0"
    }
  }
}
```

Bei mir kommt nur was für Index 0, spannend wäre zu sehen, was bei @Cheppetto für diese go-rscp Abfrage raus kommt.
@Cheppetto Are you able to clone a respository and compile the go-code on your system to get the go-rscp command, that we would need? Otherwise let me know your system (linux / arm?) and I can compile the command for you.

Cheppetto, 2024-05-02:
I am using a homeassistant to monitor and control my home and for e3dc i use (based on based on [python-e3dc](https://github.com/fsantini/python-e3dc)): 

`  "integration_manifest": {
    "domain": "e3dc_rscp",
    "name": "E3DC Remote Storage Control Protocol (Git)",
    "codeowners": [
      "@torbennehmer"
    ],
    "config_flow": true,
    "dependencies": [],
    "documentation": "https://github.com/torbennehmer/hacs-e3dc",
    "homekit": {},
    "integration_type": "device",
    "iot_class": "cloud_polling",
    "issue_tracker": "https://github.com/torbennehmer/hacs-e3dc/issues",
    "requirements": [
      "pye3dc==0.9.1"
    ],
    "ssdp": [],
    "version": "v3.7.0",
    "zeroconf": [],
    "is_built_in": false`

      "system-battery-installed-capacity": 26106,

`    "get_batteries_data": [
      {
        "asoc": 98.80000305175781,
        "chargeCycles": 147,
        "current": -0.6000000238418579,
        "dcbCount": 2,
        "dcbs": {
          "0": {
            "current": -0.30000001192092896,
            "currentAvg30s": -0.30000001192092896,
            "cycleCount": 147,
            "designCapacity": 126.0,
            "designVoltage": 51.79999923706055,
            "deviceName": "EM048126P3S",
            "endOfDischarge": 44.5,
            "error": 0,
            "fullChargeCapacity": 126.0,
            "fwVersion": 2,
            "manufactureDate": 0,
            "manufactureName": "LG",
            "maxChargeCurrent": 63.0,
            "maxChargeTemperature": 45.0,
            "maxChargeVoltage": 58.79999923706055,
            "maxDischargeCurrent": 63.0,
            "minChargeTemperature": -10.0,
            "parallelCellCount": 0,
            "sensorCount": 2,
            "seriesCellCount": 0,
            "pcbVersion": 8,
            "protocolVersion": 32,
            "remainingCapacity": 116.30000305175781,
            "serialCode": "221<redacted>",
            "serialNo": 2211102382,
            "soc": 92.30000305175781,
            "soh": 98.80000305175781,
            "status": 8288,
            "temperatures": [
              23.899999618530273,
              23.899999618530273
            ],
            "voltage": 57.0,
            "voltageAvg30s": 57.0,
            "voltages": [],
            "warning": 0
          },
          "1": {
            "current": -0.30000001192092896,
            "currentAvg30s": -0.30000001192092896,
            "cycleCount": 147,
            "designCapacity": 126.0,
            "designVoltage": 51.79999923706055,
            "deviceName": "EM048126P3S",
            "endOfDischarge": 44.5,
            "error": 0,
            "fullChargeCapacity": 126.0,
            "fwVersion": 2,
            "manufactureDate": 0,
            "manufactureName": "LG",
            "maxChargeCurrent": 63.0,
            "maxChargeTemperature": 45.0,
            "maxChargeVoltage": 58.79999923706055,
            "maxDischargeCurrent": 63.0,
            "minChargeTemperature": -10.0,
            "parallelCellCount": 0,
            "sensorCount": 2,
            "seriesCellCount": 0,
            "pcbVersion": 8,
            "protocolVersion": 32,
            "remainingCapacity": 116.0,
            "serialCode": "221<redacted>",
            "serialNo": 2212210725,
            "soc": 92.0999984741211,
            "soh": 98.80000305175781,
            "status": 8288,
            "temperatures": [
              23.100000381469727,
              23.299999237060547
            ],
            "voltage": 57.0,
            "voltageAvg30s": 57.0,
            "voltages": [],
            "warning": 0
          }
        },
        "designCapacity": 252.0,
        "deviceConnected": true,
        "deviceInService": false,
        "deviceName": "LG_BAT",
        "deviceWorking": true,
        "eodVoltage": 41.0,
        "errorCode": 0,
        "fcc": 252.0,
        "index": 0,
        "maxBatVoltage": 57.87600326538086,
        "maxChargeCurrent": 126.0,
        "maxDischargeCurrent": 126.0,
        "maxDcbCellTemp": 23.899999618530273,
        "minDcbCellTemp": 23.100000381469727,
        "moduleVoltage": 57.0,
        "rc": 232.3000030517578,
        "readyForShutdown": true,
        "rsoc": 100.0,
        "rsocReal": 92.19999694824219,
        "statusCode": 0,
        "terminalVoltage": 57.0,
        "totalUseTime": 0,
        "totalDischargeTime": 0,
        "trainingMode": 0,
        "usuableCapacity": 228.05999755859375,
        "usuableRemainingCapacity": 208.36000061035156
      }
    ],`

I fear I don't have the time to acutaly compile and deploy it anywhere else :( - sorry

docolli, 2024-05-02:
>I fear I don't have the time to acutaly compile and deploy it anywhere else :( - sorry
No problem. Is it possible for you to make a folder on a windows machine, put the e3dc.exe into it and make a little .config file? 😉
We just need to find a way how I can send you the code. Can you contact me at evcc@docollipics.de?

Cheppetto, 2024-05-02:
Yeah, guess we can manage, that as i just hopped into the evcc container and tried to enter the command but - haas in the VM in the COnsole in docker has a unknown keyboard layout ... I was not able to test it in there directly

docolli, 2024-05-02:
That will not work anyhow, since the e3dc command is not part of evcc. It is part of the go-rscp library that was used to integrate RSCP into evcc. Send me an email, then I send you the e3dc.exe and instructions how to use.

Cheppetto, 2024-05-02:
Ah that makes total sense in my head it always read as evcc not e3dc, shame on me, you received an email

Cheppetto, 2024-05-02:
`[{"Tag":"BAT_AVAILABLE_BATTERIES","DataType":"Container","Value":[{"Tag":"BAT_BATTERY_SPEC","DataType":"Container","Value":[{"Tag":"BAT_INDEX","DataType":"Int32","Value":0},{"Tag":"BAT_PARAM_BAT_NUMBER","DataType":"Int32","Value":0},{"Tag":"BAT_DEVICE_NAME","DataType":"CString","Value":"LG_BAT"},{"Tag":"BAT_MANUFACTURER_NAME","DataType":"CString","Value":"E3DC"},{"Tag":"BAT_SERIALNO","DataType":"Int32","Value":1},{"Tag":"BAT_INSTANCE_DESCRIPTOR","DataType":"CString","Value":"Battery_LG_BAT_0"}]},{"Tag":"BAT_BATTERY_SPEC","DataType":"Container","Value":[{"Tag":"BAT_INDEX","DataType":"Int32","Value":1},{"Tag":"BAT_PARAM_BAT_NUMBER","DataType":"Int32","Value":1},{"Tag":"BAT_DEVICE_NAME","DataType":"CString","Value":"LG_BAT"},{"Tag":"BAT_MANUFACTURER_NAME","DataType":"CString","Value":"E3DC"},{"Tag":"BAT_SERIALNO","DataType":"Int32","Value":1},{"Tag":"BAT_INSTANCE_DESCRIPTOR","DataType":"CString","Value":"Battery_LG_BAT_1"}]}]}]`

docolli, 2024-05-02:
Reformated the output of `e3dc.exe -output json "[[\"BAT_REQ_AVAILABLE_BATTERIES\"]]"` for easier reading:
```json
[{
"Tag":"BAT_AVAILABLE_BATTERIES","DataType":"Container","Value":[{
	"Tag":"BAT_BATTERY_SPEC","DataType":"Container","Value":[{
		"Tag":"BAT_INDEX","DataType":"Int32","Value":0},{
		"Tag":"BAT_PARAM_BAT_NUMBER","DataType":"Int32","Value":0},{
		"Tag":"BAT_DEVICE_NAME","DataType":"CString","Value":"LG_BAT"},{
		"Tag":"BAT_MANUFACTURER_NAME","DataType":"CString","Value":"E3DC"},{
		"Tag":"BAT_SERIALNO","DataType":"Int32","Value":1},{
		"Tag":"BAT_INSTANCE_DESCRIPTOR","DataType":"CString","Value":"Battery_LG_BAT_0"}]},{
	"Tag":"BAT_BATTERY_SPEC","DataType":"Container","Value":[{
		"Tag":"BAT_INDEX","DataType":"Int32","Value":1},{
		"Tag":"BAT_PARAM_BAT_NUMBER","DataType":"Int32","Value":1},{
		"Tag":"BAT_DEVICE_NAME","DataType":"CString","Value":"LG_BAT"},{
		"Tag":"BAT_MANUFACTURER_NAME","DataType":"CString","Value":"E3DC"},{
		"Tag":"BAT_SERIALNO","DataType":"Int32","Value":1},{
		"Tag":"BAT_INSTANCE_DESCRIPTOR","DataType":"CString","Value":"Battery_LG_BAT_1"}]}]
}]
```

andig, 2024-05-02:
If you have multiple batteries you‘ll need to configure them?

Cheppetto, 2024-05-02:
Nope,

Batteries can be packs of multiples of 16S 18sS 19S

And then packs of those packs, in the end it acts as one battery, packs just define the lowest BMS level, that the management interface can address.

andig, 2024-05-02:
I have no idea what that means in terms of the rscp api.

Cheppetto, 2024-05-02:
Would be the easiest solution, otherwise you need to check each index (max 0 and 1 to my best knowledge) and check + sum  the capacity of both.

docolli, 2024-05-02:
We could implement user capacity information as it was before and if it is not defined, then we ask the E3DC system as we do now. 
A full automatic detection would need the BAT_REQ_AVAILABLE_BATTERIES request and then fill an array with the numbers you get for topic BAT_INDEX for each container BAT_BATTERY_SPEC that you get in the answer. Yes, that needs some parsing. 😉
Then you send the [[\"BAT_REQ_DATA\",[[\"BAT_INDEX\",x],[\"BAT_REQ_SPECIFICATION\"]]]]" for each element in the index for x and summarize all values you get for BAT_SPECIFIED_CAPACITY to get the total battery capacity.

@Cheppetto do you have a S10 Pro System or do you really have two separate battery stacks? I have an idea why this is so, but I will try to explain tomorrow.

Cheppetto, 2024-05-02:
S10pro compact, without external tower.

The python e3dc seems to receive a installed battery capacity, I looked into the repo and to my best knowledge the information looks to be provided by the e3dc rsvp interface

andig, 2024-05-02:
@docolli is it possible to request multiple battery specs in one query?

docolli, 2024-05-03:
Hui, das sieht spannend aus. Damit bekommt man viele Systemwerte mit 1 Abfrage:

`e3dc.exe --output json "[[\"EMS_REQ_GET_SYS_SPECS\"]]"`
```json
{
  "EMS_GET_SYS_SPECS": {
    "EMS_SYS_SPEC": [
      {
        "EMS_SYS_SPEC_INDEX": 15,
        "EMS_SYS_SPEC_NAME": "hybridModeSupported",
        "EMS_SYS_SPEC_VALUE_INT": 1
      },
      {
        "EMS_SYS_SPEC_INDEX": 14,
        "EMS_SYS_SPEC_NAME": "installedBatteryCapacity",
        "EMS_SYS_SPEC_VALUE_INT": 13107
      },
      {
        "EMS_SYS_SPEC_INDEX": 13,
        "EMS_SYS_SPEC_NAME": "maxAcPower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 10,
        "EMS_SYS_SPEC_NAME": "maxBatChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 11,
        "EMS_SYS_SPEC_NAME": "maxBatDischargPower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 0,
        "EMS_SYS_SPEC_NAME": "maxChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 1,
        "EMS_SYS_SPEC_NAME": "maxDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 8,
        "EMS_SYS_SPEC_NAME": "maxFbcChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 9,
        "EMS_SYS_SPEC_NAME": "maxFbcDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 12,
        "EMS_SYS_SPEC_NAME": "maxPvPower",
        "EMS_SYS_SPEC_VALUE_INT": "ERR_NOT_AVAILABLE"
      },
      {
        "EMS_SYS_SPEC_INDEX": 4,
        "EMS_SYS_SPEC_NAME": "maxStartChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 6,
        "EMS_SYS_SPEC_NAME": "maxStartDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 4500
      },
      {
        "EMS_SYS_SPEC_INDEX": 5,
        "EMS_SYS_SPEC_NAME": "minStartChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 0
      },
      {
        "EMS_SYS_SPEC_INDEX": 7,
        "EMS_SYS_SPEC_NAME": "minStartDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 0
      },
      {
        "EMS_SYS_SPEC_INDEX": 2,
        "EMS_SYS_SPEC_NAME": "recommendedMinChargeLimit",
        "EMS_SYS_SPEC_VALUE_INT": 2200
      },
      {
        "EMS_SYS_SPEC_INDEX": 3,
        "EMS_SYS_SPEC_NAME": "recommendedMinDischargeLimit",
        "EMS_SYS_SPEC_VALUE_INT": 1100
      },
      {
        "EMS_SYS_SPEC_INDEX": 16,
        "EMS_SYS_SPEC_NAME": "startChargeDefault",
        "EMS_SYS_SPEC_VALUE_INT": 65
      },
      {
        "EMS_SYS_SPEC_INDEX": 17,
        "EMS_SYS_SPEC_NAME": "startDischargeDefault",
        "EMS_SYS_SPEC_VALUE_INT": 65
      }
    ]
  }
}
```
@Cheppetto Was kommt da bei Dir als Antwort? 
Der Wert von `installedBatteryCapacity` könnte das sein, was wir suchen und der hoffentlich allgemeingültig ist.

Cheppetto, 2024-05-03:
@docolli 

yes it's like that :)
In regards of the two battery "stacks" i guess this is caused by the inverter design to be 2 units

```javascript
{
	"EMS_GET_SYS_SPECS",{
		"EMS_SYS_SPEC":[
		{
			"EMS_SYS_SPEC_INDEX":15,
			"EMS_SYS_SPEC_NAME":"hybridModeSupported",
			"EMS_SYS_SPEC_VALUE_INT":1
			},
			{
			"EMS_SYS_SPEC_INDEX":14,
			"EMS_SYS_SPEC_NAME":"installedBatteryCapacity",
			"EMS_SYS_SPEC_VALUE_INT":26106
			},
			{
			"EMS_SYS_SPEC_INDEX":13
			"EMS_SYS_SPEC_NAME":"maxAcPower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":10,
			"EMS_SYS_SPEC_NAME":"maxBatChargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":11,
			"EMS_SYS_SPEC_NAME":"maxBatDischargPower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":0,
			"EMS_SYS_SPEC_NAME":"maxChargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":1,
			"EMS_SYS_SPEC_NAME":"maxDischargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":8,
			"EMS_SYS_SPEC_NAME":"maxFbcChargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":9,
			"EMS_SYS_SPEC_NAME":"maxFbcDischargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":12,
			"EMS_SYS_SPEC_NAME":"maxPvPower",
			"EMS_SYS_SPEC_VALUE_INT":"Error","ERR_NOT_AVAILABLE"
			},
			{
			"EMS_SYS_SPEC_INDEX":4,
			"EMS_SYS_SPEC_NAME":"maxStartChargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":6,
			"EMS_SYS_SPEC_NAME":"maxStartDischargePower",
			"EMS_SYS_SPEC_VALUE_INT":12000
			},
			{
			"EMS_SYS_SPEC_INDEX":5,
			"EMS_SYS_SPEC_NAME":"minStartChargePower",
			"EMS_SYS_SPEC_VALUE_INT":0
			},
			{
			"EMS_SYS_SPEC_INDEX":7,
			"EMS_SYS_SPEC_NAME":"minStartDischargePower",
			"EMS_SYS_SPEC_VALUE_INT":0
			},
			{
			"EMS_SYS_SPEC_INDEX":2,
			"EMS_SYS_SPEC_NAME":"recommendedMinChargeLimit",
			"EMS_SYS_SPEC_VALUE_INT":4400
			},
			{
			"EMS_SYS_SPEC_INDEX":3,
			"EMS_SYS_SPEC_NAME":"recommendedMinDischargeLimit",
			"EMS_SYS_SPEC_VALUE_INT":2200
			},
			{
			"EMS_SYS_SPEC_INDEX":16,
			"EMS_SYS_SPEC_NAME":"startChargeDefault",
			"EMS_SYS_SPEC_VALUE_INT":65
			},
			{
			"EMS_SYS_SPEC_INDEX":17,
			"EMS_SYS_SPEC_NAME":"startDischargeDefault",
			"EMS_SYS_SPEC_VALUE_INT":65
			}
		]
	}
}
```

docolli, 2024-05-03:
Der Grund für die Aufteilung der 4 Batteriemodule beim S10Pro in 2 Batterieeinheiten im Gegensatz zu 1 Batterieeinheit bei meinem S10E liegt an der Notstromfähigkeit und der Notwendigkeit zur Kalibrierung die Batteriemodule von Zeit zu Zeit vollständig entladen zu müssen. Man kann beim E3DC Batteriesystem eine Notstromreserve (in kWh) vorsehen, die im Normalbetrieb nicht entladen wird.

Cheppetto und ich haben je 4 Batteriemodule A, B, C und D. Bei mir sind diese unter 1 BatterieID "0" zusammengefasst. Zur Kalibrierung muss komplett entladen werden, d.h. bei meinem System habe ich immer wieder mal Zeiten, bei denen kurzfristig die Notstromreserve nicht mehr vorhanden ist. Fällt hier der Netzstrom aus, ist es dumm gelaufen.
Bei S10Pro wurden 2 "virtuelle" Batterieeinheiten definiert und je 2 Module zugeordnet: 0 = A/B, 1 = C/D. Somit kann jede Batterieeinheit (BatterieID) für sich kalibriert werden und die jeweils andere verfügt noch über die Notstromreserve.

@andig Hast du jetzt mit der Abfrage `'[["EMS_REQ_GET_SYS_SPECS"]]'` und der in der Antwort enthaltenen `installedBatteryCapacity` genügend Infos, um den Wert der verbauten Batteriekapazität ermitteln zu können? Ich teste den Code gerne.

maikheinrich, 2024-05-03:
Hallo zusammen,

gerne biete ich mich auch zum testen von RSCP Abfragen an. Ich habe Zugriff auf mehrere E3DC Systemen in unterschiedlichen Ausführungen.

S10 Pro mit Zusatz  Batterieschrank und weiteren Energiezähler (Erzeuger und Verbrauch)
S20 X im Farming Mode
S10X 

Wenn ich etwas zum Project beitragen kann, lasst es mich wissen...

LG Maik

docolli, 2024-05-03:
Hi Maik!
Hast du ein Windows Laptop mit dem du an diese Systeme ran kannst und dort mal deren Ausgaben auf '[["EMS_REQ_GET_SYS_SPECS"]]' ausgeben kannst?
Wenn ja, dann melde dich bei mir unter evcc@docollipics.de Ich schicke Dir die exe zusamen mit einer kleinen Anleitung.

docolli, 2024-05-03:
Schreib mir'ne Mail. 😉

docolli, 2024-05-03:
Werte für `'[["EMS_REQ_GET_SYS_SPECS"]]'` von @mars-wolf, auch hier stimmt installedBatteryCapacity mit dem theroretisch vorhandenen Wert.

```json
[{"Tag":"EMS_GET_SYS_SPECS","DataType":"Container","Value":[
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":15},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"hybridModeSupported"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":1}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":14},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"installedBatteryCapacity"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":19579}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":13},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxAcPower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":12000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":10},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxBatChargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":11},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxBatDischargPower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":0},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxChargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":1},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxDischargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":8},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxFbcChargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":9},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxFbcDischargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Int32","Value":12},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxPvPower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Error","Value":"ERR_NOT_AVAILABLE"}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":4},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxStartChargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":6},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"maxStartDischargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":9000}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":5},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"minStartChargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":0}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":7},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"minStartDischargePower"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":0}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":2},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"recommendedMinChargeLimit"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":3300}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":3},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"recommendedMinDischargeLimit"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":1700}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":16},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"startChargeDefault"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":65}]},
	{"Tag":"EMS_SYS_SPEC","DataType":"Container","Value":[
		{"Tag":"EMS_SYS_SPEC_INDEX","DataType":"Uint32","Value":17},
		{"Tag":"EMS_SYS_SPEC_NAME","DataType":"CString","Value":"startDischargeDefault"},
		{"Tag":"EMS_SYS_SPEC_VALUE_INT","DataType":"Int32","Value":65}]
	}]
}]
```

docolli, 2024-05-03:
Hier die Ausgabe von @maikheinrich und seinem 38kWh System mit 6 Batteriemodulen.

```json
{
  "EMS_GET_SYS_SPECS": {
    "EMS_SYS_SPEC": [
      {
        "EMS_SYS_SPEC_INDEX": 15,
        "EMS_SYS_SPEC_NAME": "hybridModeSupported",
        "EMS_SYS_SPEC_VALUE_INT": 1
      },
      {
        "EMS_SYS_SPEC_INDEX": 14,
        "EMS_SYS_SPEC_NAME": "installedBatteryCapacity",
        "EMS_SYS_SPEC_VALUE_INT": 39160
      },
      {
        "EMS_SYS_SPEC_INDEX": 13,
        "EMS_SYS_SPEC_NAME": "maxAcPower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 10,
        "EMS_SYS_SPEC_NAME": "maxBatChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 11,
        "EMS_SYS_SPEC_NAME": "maxBatDischargPower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 0,
        "EMS_SYS_SPEC_NAME": "maxChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 1,
        "EMS_SYS_SPEC_NAME": "maxDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 8,
        "EMS_SYS_SPEC_NAME": "maxFbcChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 9,
        "EMS_SYS_SPEC_NAME": "maxFbcDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 12,
        "EMS_SYS_SPEC_NAME": "maxPvPower",
        "EMS_SYS_SPEC_VALUE_INT": "ERR_NOT_AVAILABLE"
      },
      {
        "EMS_SYS_SPEC_INDEX": 4,
        "EMS_SYS_SPEC_NAME": "maxStartChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 6,
        "EMS_SYS_SPEC_NAME": "maxStartDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 12000
      },
      {
        "EMS_SYS_SPEC_INDEX": 5,
        "EMS_SYS_SPEC_NAME": "minStartChargePower",
        "EMS_SYS_SPEC_VALUE_INT": 0
      },
      {
        "EMS_SYS_SPEC_INDEX": 7,
        "EMS_SYS_SPEC_NAME": "minStartDischargePower",
        "EMS_SYS_SPEC_VALUE_INT": 0
      },
      {
        "EMS_SYS_SPEC_INDEX": 2,
        "EMS_SYS_SPEC_NAME": "recommendedMinChargeLimit",
        "EMS_SYS_SPEC_VALUE_INT": 6600
      },
      {
        "EMS_SYS_SPEC_INDEX": 3,
        "EMS_SYS_SPEC_NAME": "recommendedMinDischargeLimit",
        "EMS_SYS_SPEC_VALUE_INT": 3300
      },
      {
        "EMS_SYS_SPEC_INDEX": 16,
        "EMS_SYS_SPEC_NAME": "startChargeDefault",
        "EMS_SYS_SPEC_VALUE_INT": 65
      },
      {
        "EMS_SYS_SPEC_INDEX": 17,
        "EMS_SYS_SPEC_NAME": "startDischargeDefault",
        "EMS_SYS_SPEC_VALUE_INT": 65
      }
    ]
  }
}
```

mars-wolf, 2024-05-03:
For the record: Ich habe:
_Batterie 0: dort sind zwei Module drin
Batterie 1: dort ist ein Modul drin_

Insgesamt netto ergeben das ca. 19,5 kWh und E3/DC gibt brutto 17,5 kWh an.

docolli, 2024-05-03:
@andig Vorschlag zur Ermittelung der Batteriekapazität:
Wir nehmen nicht mehr BAT_REQ_DATA/BAT_INDEX 0/BAT_REQ_SPECIFICATION und holen uns aus der Antwort BAT_SPECIFICATION bzw. BAT_SPECIFIED_CAPACITY, sondern du sendest nur noch folgende Anfrage:

Tag:       rscp.EMS_REQ_GET_SYS_SPECS,
DataType: rscp.None,

Als Antwort kommt dann:
[{ EMS_GET_SYS_SPECS Container [{ EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 15 } { EMS_SYS_SPEC_NAME CString hybridModeSupported } { EMS_SYS_SPEC_VALUE_INT Int32 1 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 14 } { EMS_SYS_SPEC_NAME CString installedBatteryCapacity } { EMS_SYS_SPEC_VALUE_INT Int32 13107 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 13 } { EMS_SYS_SPEC_NAME CString maxAcPower } { EMS_SYS_SPEC_VALUE_INT Int32 12000 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 10 } { EMS_SYS_SPEC_NAME CString maxBatChargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 11 } { EMS_SYS_SPEC_NAME CString maxBatDischargPower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 0 } { EMS_SYS_SPEC_NAME CString maxChargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 1 } { EMS_SYS_SPEC_NAME CString maxDischargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 8 } { EMS_SYS_SPEC_NAME CString maxFbcChargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 9 } { EMS_SYS_SPEC_NAME CString maxFbcDischargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Int32 12 } { EMS_SYS_SPEC_NAME CString maxPvPower } { EMS_SYS_SPEC_VALUE_INT Error ERR_NOT_AVAILABLE }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 4 } { EMS_SYS_SPEC_NAME CString maxStartChargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 6 } { EMS_SYS_SPEC_NAME CString maxStartDischargePower } { EMS_SYS_SPEC_VALUE_INT Int32 4500 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 5 } { EMS_SYS_SPEC_NAME CString minStartChargePower } { EMS_SYS_SPEC_VALUE_INT Int32 0 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 7 } { EMS_SYS_SPEC_NAME CString minStartDischargePower } { EMS_SYS_SPEC_VALUE_INT Int32 0 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 2 } { EMS_SYS_SPEC_NAME CString recommendedMinChargeLimit } { EMS_SYS_SPEC_VALUE_INT Int32 2200 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 3 } { EMS_SYS_SPEC_NAME CString recommendedMinDischargeLimit } { EMS_SYS_SPEC_VALUE_INT Int32 1100 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 16 } { EMS_SYS_SPEC_NAME CString startChargeDefault } { EMS_SYS_SPEC_VALUE_INT Int32 65 }] } { EMS_SYS_SPEC Container [{ EMS_SYS_SPEC_INDEX Uint32 17 } { EMS_SYS_SPEC_NAME CString startDischargeDefault } { EMS_SYS_SPEC_VALUE_INT Int32 65 }] }] }]

Aus diesem muss man denjenigen EMS_SYS_SPEC Container ermitteln, dessen EMS_SYS_SPEC_NAME="installedBatteryCapacity" ist und dann den Wert von EMS_SYS_SPEC_VALUE_INT aus diesem Container übernehmen. Vielleicht hat @spali hier einen geschickten Befehl in seiner Bibliothek?

Dies ist dann die theoretische Batteriekapazität, das sollte für evcc genügen.

Man könnte auch zusätzlich den Parameter `capacity` wieder einführen. Ist dieser nicht definiert, so läuft die Kapazitätsermittlung wie oben beschrieben. Ansonsten wird der Wert vom Parameter übernommen und das E3DC System nicht befragt.

andig, 2024-05-04:
Lets return to the basics and keep things simple.

closed 2024-05-04 by andig

mars-wolf, 2024-05-07:
Hallo, bei 0.126.1 wurde mir in der UI 13,5 kWh Kapazität angezeigt (hab aber 19,5 kWh brutto drin). Hier wurde dann ein Weg gefunden, dass bei mehreren Batterien/ Modulen dann die Summe aller gerechnet wird. Jetzt mit 0.162.2 ist keinerlei kWh Angabe mehr in der UI ersichtlich. Was mache ich falsch? Dienst neu gestartet etc. leider ohne Erfolg.