evcc-io/evcc #13690: 0.126 missing from Debian repository

Macrisu, 2024-05-02:
### Describe the bug

I cannot upgrade to 0.126 to my rpi
sudo apt update
sudo apt upgrade -y

The latest version is 0.125

### Steps to reproduce

1.
2.
3.
...
-

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.125

StefanSchoof, 2024-05-02:
For me it is working
```
$ sudo apt update
Get:1 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Hit:2 http://deb.debian.org/debian bullseye InRelease
Get:3 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Hit:5 https://download.docker.com/linux/debian bullseye InRelease
Get:6 https://dl.evcc.io/public/evcc/stable/deb/debian bullseye InRelease
Get:7 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Get:8 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [313 kB]
Get:9 http://archive.raspberrypi.org/debian bullseye/main arm64 Contents (deb) [1,023 kB]
Get:10 http://archive.raspberrypi.org/debian bullseye/main armhf Contents (deb) [1,328 kB]
Get:11 https://dl.evcc.io/public/evcc/stable/deb/debian bullseye/main armhf Packages [16.7 kB]
Get:12 https://dl.evcc.io/public/evcc/stable/deb/debian bullseye/main arm64 Packages [16.7 kB]
Fetched 3,126 kB in 5s (680 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
$ apt list --upgradable
Listing... Done
evcc/bullseye 0.126.0 arm64 [upgradable from: 0.125.0]
```

Are you using the new dl.evcc.io, see #12902?

naltatis, 2024-05-02:
I've tried a clean install on a raspbian system. Worked as expected.

```
$ sudo apt-get install evcc
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  evcc
0 upgraded, 1 newly installed, 0 to remove and 17 not upgraded.
Need to get 23.4 MB of archives.
After this operation, 95.7 MB of additional disk space will be used.
Get:1 https://dl.evcc.io/public/evcc/stable/deb/raspbian bullseye/main armhf evcc armhf 0.126.0 [23.4 MB]
Fetched 23.4 MB in 3s (7,554 kB/s)
Selecting previously unselected package evcc.
(Reading database ... 49860 files and directories currently installed.)
Preparing to unpack .../evcc_0.126.0_armhf.deb ...
Unpacking evcc (0.126.0) ...
Setting up evcc (0.126.0) ...
pi@pihole:~ $
```

naltatis, 2024-05-02:
@Macrisu please provide logs.

Macrisu, 2024-05-02:
https://docs.evcc.io/en/docs/installation/linux

```
sudo apt install -y debian-keyring debian-archive-keyring apt-transport-https curl
curl -1sLf 'https://dl.evcc.io/public/evcc/stable/setup.deb.sh' | sudo -E bash
sudo apt update
audo apt upgrade -y
```
I use bookworm

Macrisu, 2024-05-02:
Now it is working! Downloading 0.126

closed 2024-05-02 by andig