evcc-io/evcc #13695: smaevcharger Steuerung über SEMP

scyomantion, 2024-05-02:
**Is your feature request related to a problem? Please describe.**
Aktuell wird der SMA EV Charger gesteuert, indem die AC-Strom Begrenzung immer wieder neu gesetzt und die Wallbox in den Schnelllademodus geschaltet wird. Dies führt dazu, das im sunnyportal für jede Änderung ein Log-Eintrag geschrieben wird, außerdem bin ich mir nicht sicher ob jede Änderung einen Schreibzyklus im Speicher der Wallbox auslöst und somit dieser unnötig schnell altert. 

**Describe the solution you'd like**
Es ist möglich den SMA EV Charger über das SEMP Protokoll von SMA zu steuern. Dazu muss im Sunnyportal der Haken der Datenübernahme für das Gerät entfernt werden, damit der SMA Home Manager 2.0 dieses nicht mehr versucht zu steuern. Ab dann können von EVCC XML-Posts an http://[sma-ev-ip]/SEMP/ geschickt werden mit dem bekannten SEMP Inhalt:
```XML
<?xml version="1.0" encoding="UTF-8" ?>
<EM2Device xmlns="http://www.sma.de/communication/schema/SEMP/v1">
    <DeviceControl>
        <DeviceID>[device id]</DeviceID>
        <On>true</On>
        <RecommendedPowerConsumption>6000</RecommendedPowerConsumption>
        <Timestamp>0</Timestamp>
    </DeviceControl>
</EM2Device>
```
Dabei kann die DeviceID vorher über einen GET Aufruf an die gleiche URL ermittelt werden. In RecommendedPowerConsumption die gewollte Leistung eintragen. 

Bei meinen Versuchen regelt die Wallbox dann auf einen minimal kleineren Wert als der mitgegebene. Hier weiß ich nicht ob das Fahrzeugabhängig ist, sollte aber kein Problem sein, da ja evcc das automatisch ausgleicht. Ich habe bei mir einen Prototypen gebastelt der das Umsetzt und die ersten Versuche haben gut funktioniert.

Der SMA Home Manager 2.0 schickt zum Regeln der Wallbox, wenn dieser die Datenübernahme aktiv hat, auch noch zusätzlich an die folgende URL das folgende Datenpaket:
URL: http://[sma-ev-ip]/SEMP/Parameters
Daten:
```XML
<?xml version="1.0" encoding="utf-8"?>
<EM2Device xmlns="http://www.sma.de/communication/schema/SEMP/v1" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:schemaLocation="http://www.sma.de/communication/schema/SEMP/v1 SEMP-1.3-charge-api.xsd">
  <Parameters>
    <Parameter>
      <channelId>Setpoint.PlantControl.PCC.ChrgActCnt</channelId>
      <value>1</value>
    </Parameter>
  </Parameters>
</EM2Device>
```
Ich vermute, das damit die Regelung irgendwie beeinflusst wird, genau konnte ich das noch nicht herausfinden. In einen Versuch hat es auch mit dem Wert 0 funktioniert und ich konnte keine Unterschiede feststellen bis jetzt. 

Ein weiterer Vorteil der Lösung ist, dass die Regelung im Intelligenten Modus der Wallbox funktioniert und der Schnelllademodus, wie gewohnt, schnell lädt, wobei das ja dann auch über evcc funktioniert.

**Describe alternatives you've considered**
Wie beschrieben die aktuelle Lösung.

**Additional context**
Es können ein paar Parameter auch ohne Authentifizierung über das SEMP Protokoll ausgelesen werden.
Folgendes hab ich gefunden + Beispiel:

GET URL: http://[sma-ev-ip]/SEMP/Parameters
```XML
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<Device2EM xmlns="http://www.sma.de/communication/schema/SEMP/v1">
    <Parameters>
        <Parameter>
            <channelId>Parameter.Chrg.ActChaMod</channelId>
            <timestamp>2024-04-30T10:50:54.742Z</timestamp>
            <value>4719</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.Plan.StopTm</channelId>
            <timestamp>2024-04-30T10:51:55.000Z</timestamp>
            <value>1714474315</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.Plan.En</channelId>
            <timestamp>2024-04-30T10:51:55.487Z</timestamp>
            <value>0</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.StpWhenFl</channelId>
            <timestamp>2024-04-30T10:50:31.425Z</timestamp>
            <value>1130</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.StpWhenFlTm</channelId>
            <timestamp>2024-04-30T10:50:31.426Z</timestamp>
            <value>15</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Inverter.OutPhs</channelId>
            <timestamp>2024-04-30T10:50:31.443Z</timestamp>
            <value>326</value>
        </Parameter>
        <Parameter>
            <channelId>Setpoint.PlantControl.PCC.ChrgActCnt</channelId>
            <timestamp>2024-04-30T18:45:51.468Z</timestamp>
            <value>0</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Operation.EVeh.Health</channelId>
            <timestamp>2024-04-30T18:45:56.679Z</timestamp>
            <value>307</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Operation.EVeh.ChaStt</channelId>
            <timestamp>2024-04-30T18:45:56.193Z</timestamp>
            <value>200111</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Operation.Health</channelId>
            <timestamp>2024-04-30T15:00:36.347Z</timestamp>
            <value>307</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Operation.Evt.Msg</channelId>
            <timestamp>2024-04-30T15:00:36.347Z</timestamp>
            <value>302</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.ChaSess.WhIn</channelId>
            <timestamp>2024-04-30T15:00:36.213Z</timestamp>
            <value>0.0</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Nameplate.PkgRev</channelId>
            <timestamp>2024-04-30T10:50:53.457Z</timestamp>
            <value>1.2.23.R</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Metering.GridMs.TotWIn.ChaSta</channelId>
            <timestamp>2024-04-30T18:45:53.193Z</timestamp>
            <value>0</value>
        </Parameter>
        <Parameter>
            <channelId>Measurement.Chrg.ModSw</channelId>
            <timestamp>2024-04-30T10:50:54.779Z</timestamp>
            <value>4950</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.ChrgLok</channelId>
            <timestamp>2024-04-30T13:38:13.571Z</timestamp>
            <value>1130</value>
        </Parameter>
        <Parameter>
            <channelId>Parameter.Chrg.ChrgApv</channelId>
            <timestamp>2024-04-30T13:38:13.323Z</timestamp>
            <value>5171</value>
        </Parameter>
    </Parameters>
</Device2EM>
```

GET URL: http://[sma-ev-ip]/SEMP/CustomerMessages?from=2024-04-29T18:40:34.223Z&until=2024-04-30T18:41:59.543Z
```XML
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<Device2EM xmlns="http://www.sma.de/communication/schema/SEMP/v1">
    <CustomerMessages>
        <CustomerMessage>
            <deviceId>[...]</deviceId>
            <timestamp>2024-04-30T05:54:46.587Z</timestamp>
            <accessLevel>USER</accessLevel>
            <messageId>9804</messageId>
            <messageTag>4807</messageTag>
            <traceLevel>INFO</traceLevel>
            <eventTypeExtension>NONE</eventTypeExtension>
            <messageGroupTag>832</messageGroupTag>
            <arguments>
                <position>0</position>
                <value>4721</value>
                <displayType>TagId</displayType>
                <unitTag>0</unitTag>
            </arguments>
            <arguments>
                <position>1</position>
                <value>4719</value>
                <displayType>TagId</displayType>
                <unitTag>0</unitTag>
            </arguments>
        </CustomerMessage>
        <CustomerMessage>
            <deviceId>[...]</deviceId>
            <timestamp>2024-04-30T06:27:44.626Z</timestamp>
            <accessLevel>USER</accessLevel>
            <messageId>9804</messageId>
            <messageTag>4807</messageTag>
            <traceLevel>INFO</traceLevel>
            <eventTypeExtension>NONE</eventTypeExtension>
            <messageGroupTag>832</messageGroupTag>
            <arguments>
                <position>0</position>
                <value>4719</value>
                <displayType>TagId</displayType>
                <unitTag>0</unitTag>
            </arguments>
            <arguments>
                <position>1</position>
                <value>4719</value>
                <displayType>TagId</displayType>
                <unitTag>0</unitTag>
            </arguments>
        </CustomerMessage>
    </CustomerMessages>
</Device2EM>
```

andig, 2024-05-02:
Die Parameter sagen mir nichts. Gibts alles was für api.Charger benötigt wird? Messwerte?

scyomantion, 2024-05-02:
Ich bin leider noch nicht so tief drin in evcc, habe die Parameter aber mal kurz mit denen aus der aktuellen Implementierung abgeglichen. Ich konnte alle finden die er aktuell ausliest bis auf api.PhaseCurrents. 
Also ist auf jedenfalls alles um das Charger interface zu implementieren. Wobei der Strom wieder in Watt umgerechnet werden müsste.

mfuchs1984, 2024-05-03:
Per SEMP ließe sich dann auch die Phasenumschaltung nutzen, oder? Aktuell ist diese in der Doku nicht eingetragen, ich gehe deshalb davon aus, dass das mit dem aktuell verwendeten Protokoll nicht geht. 
Übrigens kann unsere Bender Wallbox über SEMP mit der kommenden Softwareversion auch Phasenumschaltung, andere Wallboxen würden also auch profitieren.

andig, 2024-05-03:
Wir brauchen: https://github.com/evcc-io/evcc/blob/master/api/api.go#L68-L73 und dazu idealerweise noch Messwerte. Steuerung über Leistung können wir noch nicht sooo gut, geht aber.

> Ich konnte alle finden die er aktuell ausliest bis auf api.PhaseCurrents.

Welche sind denn welche? Insbesondere, was ist "enabled" und "status"? Wenns die gäbe könnten wir mal einen Versuch starten!

scyomantion, 2024-05-03:
> Per SEMP ließe sich dann auch die Phasenumschaltung nutzen, oder?

Konnte hierzu leider auch noch nichts finden, wie man das selbst steuern könnte. Aktuell ist es so, dass er automatisch von 1 auf 3 Phasen umschaltet wenn die Leistung eine Zeit X über Wert Y vorgegeben wird. X und Y könnte ich per ausprobieren herausfinden. Ebenso schaltet es wieder zurück auf eine Phase wenn die Leistung eine gewisse Zeit zu niedrig ist. Vor dem Umschalten limitiert er die Leistung bei mir auf ca. 4 KW. Laut Doku kann er auf einer Phase mehr, hält aber natürlich die maximale Schieflast ein.

Ich kann demnächst nochmal den Datenverkehr zwischen Home-Manager und Wallbox abgreifen und schauen ob ich hier noch ein paar versteckte Befehle finden kann.

Die Zuordnung der Parameter kann ich gleich noch raussuchen. Ist aber eigentlich genau gleich wie es aktuell über das Web-Interface abgefragt wird.

mfuchs1984, 2024-05-03:
Ja, soweit ich verstehe, passiert das über die Leistungsvorgabe, nicht explizit per Kommando. Die Logik wäre dann doch anders als bei anderen Wallboxen.

scyomantion, 2024-05-03:
Auf der SMA FAQ Seite zum EV Charger findet man noch folgende Aussage:

> Die automatische Phasenumschaltung des SMA EV Charger ermöglicht die maximale Ausnutzung des selbsterzeugten Solarstroms. Dabei besteht keine Gefahr für den Onboard-Lader des Elektrofahrzeugs, weil zwischen ein- und dreiphasigem Ladevorgang eine Umschaltpause von etwa 120 Sekunden besteht. Der aktive einphasige Ladevorgang wird also zunächst beendet und der dreiphasige Ladevorgang dann erst nach einer kurzen Wartezeit gestartet. Eine integrierte Hysterese vermeidet ständiges Hin- und Herschalten.
>
> Die SMA Phasenumschaltung ist vergleichbar mit einer händischen Unterbrechung des Ladevorgangs und verursacht bei fachgerechter Installation und Konfiguration (Relaisumschaltzeit) keine Fahrzeugschäden.

Ich könnte demnächst auch mal ausprobieren ob sich das auch Manuell so verhält, also das ich mit einer Phase lade, dann 0 Watt vorgebe, kurz Pause, dann eine höhere Leistung. Eventuell lässt er sich so dazu bringen umzuschalten?

Wie ist das denn mit der Phasenumschaltung in evcc? Welche vorteile hat es wenn man hier die Kontrolle von der Wallbox zu evcc bringt? Dadurch das man ja eigentlich nicht mehr mit Stromstärken kommuniziert sondern mit der geforderten Leistung müsste es ja egal sein, richtig? 

Hier noch das Mapping der Parameter:
```
Status() = Measurement.Operation.EVeh.ChaStt
Enabled() = Parameter.Chrg.ActChaMod mappen oder SEMP abfrage und dann auf DeviceStatus>Status == on prüfen
Enable() = Setzen vom Feld On im SEMP Paket + Leistung 0 bzw. > 0
MaxCurrentMillis() = Umrechnen in Watt und dann Feld RecommendedPowerConsumption im SEMP Paket + On = true
CurrentPower() = Measurement.Metering.GridMs.TotWIn
ChargedEnergy() = Measurement.ChaSess.WhIn / 1e3
```
Werte für Measurement.Operation.EVeh.ChaStt: https://github.com/evcc-io/evcc/blob/25f9f6ce5004477ec4be763e3616f904ff6e40d3/charger/smaevcharger/types.go#L9 
Werte für Parameter.Chrg.ActChaMod: https://github.com/evcc-io/evcc/blob/25f9f6ce5004477ec4be763e3616f904ff6e40d3/charger/smaevcharger/types.go#L17

andig, 2024-05-03:
Spannend... müsste man echt mal ausprobieren. Gibts jemanden der per SSH Remotezugang auf seinen Charger zum testen anbietet? Mein Pubkey steht im Github Userprofil!

scyomantion, 2024-05-03:
Möchtest du was bestimmtes testen @andig ? Ich könnte mal schauen ob ich den aus den Hauptnetzwerk raus isoliert bekomme, damit ich mit ruhigem gewissen ein Zugang stellen kann.

andig, 2024-05-04:
Etwas bestimmtes? Die Implementierung!

scyomantion, 2024-05-04:
Dann am besten auch zu ein Zeitpunkt wo das Auto dran hängt? Ich schau mal das ich ein Zugang bereit stelle und dann können wir uns noch absprechen wann es am besten geht

andig, 2024-05-04:
Einfach mal genereller Testzugang. Ohne gibts keine Implementierung. Ich mache hier mal zu bis sich da was abzeichnet.