evcc-io/evcc #13696: goodwe-wifi interrupts data upload to semsportal

dassams, 2024-05-02:
### Describe the bug

While EVCC is running, the GoodWe hybrid inverter no longer uploads data to SEMS portal.

### Steps to reproduce

1. Start EVCC with configured gooodwe-wifi
2. Observe EVCC working properly.
3. SEMS has only 0 values logged
4. Stop EVCC
5. Data is uploaded to SEMS again.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ...

interval: 30s # control cycle interval

sponsortoken: ...

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: goodwe-wifi 
  usage: grid
  uri: 192.168.178.70  
  name: grid1
- type: template
  template: goodwe-wifi 
  usage: pv
  uri: 192.168.178.70  
  name: pv1
- type: template
  template: goodwe-wifi 
  usage: battery
  uri: 192.168.178.70
  name: battery1
  # title: GoodWe LX F9.8-H

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.75  
  name: wallbox1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv

vehicles:
  - name: MB-S205
    title: Mercedes C300 de
    type: mercedes
    account: markus@mwagner.info # Mercedes Me Nutzer-Id (email)
    region: EMEA # MB me Region (EMEA, APAC, NORAM)
    vin: W1K2052131G087806 # Erforderlich, wenn mehr als ein Fahrzeug im Account registriert
    capacity: 13.5 # Akkukapazität in kWh (optional)
    tokens:
      access: ...
      refresh: ...

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
```

### Log details

```text
[main  ] INFO 2024/05/02 22:04:06 evcc 0.126.1
[main  ] INFO 2024/05/02 22:04:06 using config file: /etc/evcc.yaml
[main  ] INFO 2024/05/02 22:04:06 starting ui and api at :7070
[db    ] INFO 2024/05/02 22:04:06 using sqlite database: /root/.evcc/evcc.db
[mercedes] DEBUG 2024/05/02 22:04:06 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/05/02 22:04:06 identity.NewIdentity - token not valid - database token check started
[mercedes] DEBUG 2024/05/02 22:04:06 identity.NewIdentity - token not valid - refreshToken started
[main  ] ERROR 2024/05/02 22:04:06 creating vehicle MB-S205 failed: cannot create vehicle type 'mercedes': token expired
[lp-1  ] DEBUG 2024/05/02 22:04:07 charge total import: 200.928kWh
[site  ] WARN 2024/05/02 22:04:07 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/05/02 22:04:07 site config:
[site  ] INFO 2024/05/02 22:04:07   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/05/02 22:04:07     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/05/02 22:04:07     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/05/02 22:04:07     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/05/02 22:04:07   vehicles:
[site  ] INFO 2024/05/02 22:04:07     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/05/02 22:04:07 loadpoint 1:
[lp-1  ] INFO 2024/05/02 22:04:07   mode:        pv
[lp-1  ] INFO 2024/05/02 22:04:07   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/05/02 22:04:07   meters:      charge ✓
[lp-1  ] INFO 2024/05/02 22:04:07     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/05/02 22:04:07 phase timer inactive
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv timer inactive
[site  ] DEBUG 2024/05/02 22:04:07 ----
[lp-1  ] DEBUG 2024/05/02 22:04:07 charge power: 0W
[lp-1  ] DEBUG 2024/05/02 22:04:07 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/02 22:04:07 pv power: 0W
[site  ] DEBUG 2024/05/02 22:04:07 battery soc: 0%
[site  ] DEBUG 2024/05/02 22:04:07 battery power: 627W
[site  ] DEBUG 2024/05/02 22:04:07 grid meter: -99W
[site  ] DEBUG 2024/05/02 22:04:07 site power: 528W
[lp-1  ] DEBUG 2024/05/02 22:04:07 charge total import: 200.928kWh
[lp-1  ] DEBUG 2024/05/02 22:04:07 charger status: B
[lp-1  ] INFO 2024/05/02 22:04:07 car connected
[lp-1  ] DEBUG 2024/05/02 22:04:07 vehicle api refresh
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv timer elapse
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv timer inactive
[lp-1  ] DEBUG 2024/05/02 22:04:07 charge total import: 200.928kWh
[lp-1  ] DEBUG 2024/05/02 22:04:07 available power -528W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/05/02 22:04:07 phase scale1p in 0s
[lp-1  ] DEBUG 2024/05/02 22:04:07 switched phases: 1p
[lp-1  ] DEBUG 2024/05/02 22:04:07 phase timer inactive
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv charge current: 0A = 0A + -2.29A (528W @ 1p)
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv enable timer reset
[lp-1  ] DEBUG 2024/05/02 22:04:07 pv timer inactive
[site  ] DEBUG 2024/05/02 22:04:37 ----
[lp-1  ] DEBUG 2024/05/02 22:04:37 charge power: 0W
[lp-1  ] DEBUG 2024/05/02 22:04:37 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/02 22:04:37 pv power: 0W
[site  ] DEBUG 2024/05/02 22:04:37 battery soc: 42%
[site  ] DEBUG 2024/05/02 22:04:37 battery power: 627W
[site  ] DEBUG 2024/05/02 22:04:37 grid meter: -99W
[site  ] DEBUG 2024/05/02 22:04:37 site power: 528W
[lp-1  ] DEBUG 2024/05/02 22:04:37 charge total import: 200.928kWh
[lp-1  ] DEBUG 2024/05/02 22:04:37 charger status: B
[lp-1  ] DEBUG 2024/05/02 22:04:37 pv charge current: 0A = 0A + -2.29A (528W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

0.126.1

andig, 2024-05-02:
@ReneNulschDE need to remove the debug messages ;)

andig, 2024-05-02:
@dassams I have no idea what SEMS is. Seems you‘ll need ro chose either evcc or that.

dassams, 2024-05-02:
> @dassams I have no idea what SEMS is. Seems you‘ll need ro chose either evcc or that.
GoodWe's proprietary cloud portal... During test phase and while requiring GoodWe support I'd like to use both. (And then stop using sems and keep the data for me only)