evcc-io/evcc #13697: evcc configure - German only (used to be EN for me)

RTTTC, 2024-05-02:
### Describe the bug

For quite a while now evcc configure shows up in German. It used to be EN. My raspberry localisation is in EN.
Is it a bug or should I check something on my side?

### Steps to reproduce

-

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

v0.126.0

naltatis, 2024-05-03:
Works for me via param. You think the language detection doesnt work correctly?

```
evcc [master] ./evcc configure --lang en
[main  ] INFO 2024/05/03 08:51:56 evcc 0.126.1 (8eea80864)
[main  ] INFO 2024/05/03 08:51:56 evcc 0.126.1 (8eea80864)

The next steps will guide you through the creation of a configuration file for evcc.
Please note that this process cannot cover all possible scenarios.
You can cancel the process by pressing CTRL-C anytime.

NOTE: This functionality is in an experimental state!
  This means that the configuration file created may not work
  in a future update of evcc and would need to be recreated.
  We are looking forward to your feedback at https://github.com/evcc-io/evcc/discussions/

Let's go:

 In which mode should this process guide you through the configuration?  [Use arrows to move, type to filter]
> Standard mode (As simple and quick as possible)
  Advanced mode (Ask more details, requires technical know-how)
```

RTTTC, 2024-05-03:
`evcc configure --lang en` works, thank you @naltatis ! 😎 
I was not aware of this possibilty. (I _think_ it was EN, when i used it a while ago just as `evcc configure` ) 🤷 

Also, this is not related, but it seems that all the vehicle brand options are duplicated? (evcc configure / advanced / single device/vehicle)
I did not check if other deviced such as inverter options are ok, or duplicate too.

> Audi
Audi
BMW
BMW
Citroen
Citroen
...

Again - not sure if I have something screwed up somewhere, or its a bug, or a feature :D

naltatis, 2024-05-03:
double entries are fixed with https://github.com/evcc-io/evcc/pull/13709.

I'll close this one.

closed 2024-05-03 by naltatis