evcc-io/evcc #13701: Renault total mileage no longer shown in charging logbook

felix-schwarz, 2024-05-03:
### Describe the bug

I noticed that for some time now the total mileage of the vehicle is no longer shown in the charging logbook.

### Steps to reproduce

1. Charge car.
2. Check charging logbook

I assume [`totalMileage`](https://github.com/evcc-io/evcc/blob/81d677b897df7c8c182dd1a470d193f8177d9e06/vehicle/renault/kamereon/types.go#L67) is either no longer returned by Renault's API - or returned under a different name.

### Configuration details

```yaml
vehicles:
  - name: eauto
    type: renault
    title: E-Auto
    capacity: 22
    user: ***
    password: ***
    onIdentify:
      mode: pv
```

### Log details

```text
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:19 charge power: 0W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:19 charge currents: [0 0 0]A
evcc  | [site  ] DEBUG 2024/05/03 08:06:19 pv power: 206W
evcc  | [site  ] DEBUG 2024/05/03 08:06:19 grid meter: 49W
evcc  | [site  ] DEBUG 2024/05/03 08:06:19 site power: 49W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:19 charge total import: [private]kWh
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:19 charger status: A
evcc  | [site  ] DEBUG 2024/05/03 08:06:22 ----
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 charge power: 0W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 charge currents: [0 0 0]A
evcc  | [site  ] DEBUG 2024/05/03 08:06:22 pv power: 207W
evcc  | [site  ] DEBUG 2024/05/03 08:06:22 grid meter: 49W
evcc  | [site  ] DEBUG 2024/05/03 08:06:22 site power: 49W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 charge total import: [private]kWh
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 charger status: B
evcc  | [lp-1  ] INFO 2024/05/03 08:06:22 car connected
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 pv timer elapse
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 pv timer inactive
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:22 charge total import: [private]kWh
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 vehicle status: A
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 vehicle soc: 68%
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 vehicle range: 126km
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 pv charge current: 0A = 0A + -0.215A (49W @ 1p)
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 pv enable timer reset
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:23 pv timer inactive
evcc  | [site  ] DEBUG 2024/05/03 08:06:25 ----
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 charge power: 0W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 charge currents: [0 0 0]A
evcc  | [site  ] DEBUG 2024/05/03 08:06:25 pv power: 212W
evcc  | [site  ] DEBUG 2024/05/03 08:06:25 grid meter: 45W
evcc  | [site  ] DEBUG 2024/05/03 08:06:25 site power: 45W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 vehicle odometer: 0km
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 charge total import: [private]kWh
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 charger status: B
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:25 pv charge current: 0A = 0A + -0.194A (45W @ 1p)
evcc  | [site  ] DEBUG 2024/05/03 08:06:28 ----
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:28 charge power: 0W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:28 charge currents: [0 0 0]A
evcc  | [site  ] DEBUG 2024/05/03 08:06:28 pv power: 213W
evcc  | [site  ] DEBUG 2024/05/03 08:06:28 grid meter: 41W
evcc  | [site  ] DEBUG 2024/05/03 08:06:28 site power: 41W
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:28 charge total import: [private]kWh
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:28 charger status: B
evcc  | [lp-1  ] DEBUG 2024/05/03 08:06:28 pv charge current: 0A = 0A + -0.177A (41W @ 1p)
evcc  | [site  ] DEBUG 2024/05/03 08:06:31 ----
```

### What type of operating system are you running?

Linux

### Version

0.126.0

andig, 2024-05-03:
Does the Renault app show it? If yes, please add trace log.

felix-schwarz, 2024-05-03:
I never found the total milage in the Renault app and was pleasantly surprised that this data was accessible and evcc shows it.

andig, 2024-05-03:
Trace log is needed anyway. If not present in the api we can’t produce it.

felix-schwarz, 2024-05-03:
I switched from log level `debug` to `trace`, restarted evcc  and found this in the log:

```
evcc  | [renault] TRACE 2024/05/03 18:01:03 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/[private]/kamereon/kca/car-adapter/v2/cars/[private]/cockpit?country=DE
evcc  | [renault] TRACE 2024/05/03 18:01:03 {"message": "you should not be there but well done for the effort"}
```

Looks like Renault blocks access to the cockpit endpoint.

premultiply, 2024-05-03:
Endpoint has been changed to v1 API

andig, 2024-05-04:
Fixed in https://github.com/evcc-io/evcc/commit/8327ddbb2188a780fa67c990c2ab5017d94dc5d5

closed 2024-05-04 by andig