evcc-io/evcc #13703: tarriff changes

mxwi, 2024-05-03:
### Describe the bug

I am not sure if this is a bug of feature request, but I guess in comparison with Home Assistant the behavior is not expected.

### Steps to reproduce

1. set a value for tarriff-grid
2. load your ev and remember the value for saved €/$
3. reset the tarriff-grid value to a new one (because you have a new contract and the price for kw/h changed)
4. take a look again at your saved €/$ value. It is completly calculated with the new value.
...
The expected behaviour would be to calculate the savings for each day with the value that was configured until it changed.

### Configuration details

```yaml
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2616 # [currency]/kWh
```

### Log details

```text
-
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.125.0

andig, 2024-05-03:
> The expected behaviour would be to calculate the savings for each day with the value that was configured until it changed.

@naltatis das Savings müssten mit aggregiert anstatt live berechnet werden? Gabs dazu nicht schon ein Issue?

naltatis, 2024-05-03:
Yes, its a duplicate to #11160 which seems to have been closed accidentally. Currently, we don't save reference grid price and co2 level for a charging session, but we need to do that to calculate accurate savings.

I've reopened the original issue and will close this as a duplicate.

closed 2024-05-03 by naltatis