evcc-io/evcc #13705: Update auf 0.126.1 fiat nicht mehr nutzbar

unl0ck, 2024-05-03:
### Describe the bug

Hab auf die 0.126.1 upgedated, jetzt erhalte ich folgenden fehler 
```
cannot create vehicle 'fiat'
cannot create vehicle type 'template'
cannot create vehicle type 'fiat'
1 error(s) decoding: * '' has invalid keys: help
```

### Steps to reproduce

1. von 0.125.0
2. auf 0.126.1 updaten
3.
...

### Configuration details

```yaml
chargers:  
  - name: wallbox  
    type: template  
    template: tinkerforge-warp  
    host: 192.168.2.4 # IP Adresse oder der Hostname des MQTT Brokers  
    port: 1883 # MQTT Broker Port (Optional)  
    topic: warp2/21QS # Topic (ohne / am Anfang) (Optional)  
    user: *****  
    password: *****  
  
meters:  
  - name: netzbezug  
    type: template  
    template: vzlogger  
    usage: grid  
    host: 192.168.2.4 # IP-Adresse oder Hostname  
    port: 8084 # Port (Optional)  
    uuid: 14FDB752-A5FC-4174-A20E-698604D3A165  
  - name: einspeisung  
    type: template  
    template: vzlogger  
    usage: min+pv  
    host: 192.168.2.4 # IP-Adresse oder Hostname  
    port: 8084 # Port (Optional)  
    uuid: F22F6F70-88D1-4AD4-A46D-0FAC01F1B85D  
  - name: mqttPV  
    type: custom  
    power:  
      source: mqtt  
      topic: hmd/sensor/Viessmann-Gridbox/Production/state  
  - name: mqttGrid  
    type: custom  
    power:  
      source: mqtt  
      topic: hmd/sensor/Viessmann-Gridbox/Grid/state  
  
vehicles:  
  - name: fiat  
    title: Fiat  
    type: template  
    template: fiat  
    user: *****  
    password: *****  
    pin: *****  
    capacity: 47  
  
loadpoints:  
  - title: Stellplatz  
    charger: wallbox  
    mode: pv  
    vehicle: fiat  
    phases: 3  
    mincurrent: 6  
    maxcurrent: 32  
    enable:  
      delay: 1m  
      threshold: -400 # es wird eingeschaltet, wenn 1 Minute lang 400 W Einspeisung vorhanden sind  
    disable:  
      delay: 30m  
      threshold: 1200 # es wird ausgeschaltet, wenn 30 Minuten lang mehr als 1200 W Netzstrom bezogen wurde  
  
# mqtt message broker  
mqtt:  
  broker: 192.168.2.4:1883  
  topic: evcc # root topic for publishing, set empty to disable publishing  
  # clientid: foo  
  user: *****  
  password: *****  
  
tariffs:  
  currency: EUR # (default EUR)  
  grid:  
    type: fixed  
    price: 0.40 # [currency]/kWh  
  
  feedin:  
    type: fixed  
    price: 0.08 # [currency]/kWh  
interval: 120s  
log: debug  
site:  
  title: Mein Zuhause  
  meters:  
    grid: mqttGrid  
    pv: mqttPV
```

### Log details

```text
[main  ] FATAL 2024/05/03 10:08:31 cannot create vehicle 'fiat': cannot create vehicle type 'template': cannot create vehicle type 'fiat': 1 error(s) decoding:
```

### What type of operating system are you running?

Docker container

### Version

0.126.1

joh3v16, 2024-05-03:
Bei mir auch das Gleiche Problem seit Update auf 126.1 unter Debian Linux
Mai 03 11:10:30 minipc systemd[1]: Started evcc.service - evcc.
Mai 03 11:10:31 minipc evcc[4479]: [main  ] INFO 2024/05/03 11:10:31 evcc 0.126.1
Mai 03 11:10:31 minipc evcc[4479]: [main  ] INFO 2024/05/03 11:10:31 using config file: /etc/evcc.yaml
Mai 03 11:10:31 minipc evcc[4479]: [main  ] INFO 2024/05/03 11:10:31 starting ui and api at :7070
Mai 03 11:10:31 minipc evcc[4479]: [main  ] FATAL 2024/05/03 11:10:31 cannot create vehicle 'Fiat500E': cannot create vehicle type 'template': cannot create vehicle type 'fiat': 1 error(s) decoding:
Mai 03 11:10:31 minipc evcc[4479]: * '' has invalid keys: help
Mai 03 11:10:31 minipc evcc[4479]: [main  ] FATAL 2024/05/03 11:10:31 will attempt restart in: 5m0s

mdkeil, 2024-05-03:
sollte spätestens in der nächsten nightly behoben sein:  https://github.com/evcc-io/evcc/commit/8eea8086450b1b62e1d9a9db1ace47e561a744e3

#13699

closed 2024-05-03 by andig

unl0ck, 2024-05-03:
@mdkeil ahhh da kam das help her, schade wäre meine Chance gewesen für den ersten pull request hier ;-)

andig, 2024-05-04:
@GrimmiMeloni hast du eine Idee, warum die Tests das nicht gefunden haben? Templatefehler sollten wir doch eigentlich abfangen?

GrimmiMeloni, 2024-05-07:
> @GrimmiMeloni hast du eine Idee, warum die Tests das nicht gefunden haben? Templatefehler sollten wir doch eigentlich abfangen?

Zwei Gründe:

1. #10272 hat einen bösen Bug. Der Render Test wird nur bis zum ersten Template mit usages==0 ausgeführt. Bei den Vehicles ist dies leider das erste Template (`aiways`). Ich bin dran und liefer einen Fix.
2. Selbst wenn ich diesen Bock oben nicht geschossen hätte finden wir das nicht, weil unser Test nicht optimal ist.

Details zum Shortcoming im Test:
1. YAML mit unbekannten Feldern wie hier, erkennen wir im Test nicht, weil wir ein `unmarshal()` auf ein `interface{}` machen. Das passt immer - zumindest so lange wie der render Block gültiges YAML enthält (was hier der Fall war). 
2. Wir können solche Situations erst beim Instanziieren sehen - das machen wir aber einerseits nicht bei den Tests in der GitHub CI. Und selbst wenn wir dies machen (oder lokal testen), unser Test nutzt nicht das die values Map aus dem im obigen Schritt aus dem Template gerenderten YAML, sondern nutzt wiederum die values die originär für den Test (die Defaults) bereit stehen. Dort aber stand ja das PIN Feld auch weiterhin ohne Help drin - somit geht auch dieser Test durch. Das könnten wir ändern - ich probiere mal aus.

🤯

andig, 2024-05-07:
Puh... harter Stoff :O