evcc-io/evcc #13712: Eastron meter does not support phase currents in EVCC

hyperbart, 2024-05-03:
### Describe the bug

I started converting my config file to the load balancing config.
I am using an SDM630 with a P11 ethernet module but I am now being confronted with the error while afaik the SDM630 does support phase currents (I am visualising them with Grafana)

```
cannot create circuit 'AutomaatNetaansluiting'
meter does not support phase currents

```

Relevant part of config:

```
meters:
  - name: HuisMeter
    type: mbmd
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.4.184:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter
```

### Steps to reproduce

Create EVCC setup with meter, deploy with circuits/loadbalancing and it shows as:

```
meters:
  - name: HuisMeter
    type: mbmd
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.4.184:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter
```

### Configuration details

```yaml
interval: 10s

site:
- title: Allspark
  meters:
    grid: HuisMeter
    pv:
    - pvHuis
    - pvTuinHuis
  residualpower: 10
  circuit: netaansluiting

circuits:
  - name: AutomaatNetaansluiting
    maxCurrent: 20
    maxPower: 5000 #9.6
    meter: HuisMeter
    parent:
  - name: AutomaatLaadpaalVoortuin
    maxCurrent: 25
    maxPower: 11000 #11kW
    meter:
    parent: AutomaatNetaansluiting
  - name: AutomaatLaadpaalOprit
    maxCurrent: 10
    maxPower: 2400 #11kW
    meter:
    parent: AutomaatNetaansluiting

messaging:
  events:
    connect:
      title: Wagen aangesloten 
      msg: |
        Connect
        Voertuig aangesloten op de {{.title}}-lader.
    disconnect:
      title: Wagen ontkoppeld
      msg: |
        Disconnect
        Voertuig ontkoppeld van de {{.title}}-lader.
    start:
      title: Wagen gestart 
      msg: |
        Start
        {{.vehicleTitle}} is gestart met laden aan de {{.title}}-lader.
        Rijbereik van {{.vehicleTitle}} is {{.vehicleRange}}km met {{.vehicleSoc}}% batterij.
        
    stop:
      title: Wagen gestopt 
      msg: |
        Stop
        {{.vehicleTitle}} is gestopt met laden aan de {{.title}}-lader.
        Er is {{round (divf .chargedEnergy 1000) 2 }} kWh geleverd op {{.chargeDuration}}.
        Rijbereik van {{.vehicleTitle}} is {{.vehicleRange}}km met {{round .vehicleSoc 2}}% batterij.

  services:
    - type: telegram
      token: xxx:xxx
      chats:
        - -xxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: entsoe # Entso-E european market data
    domain: BZN|BE # https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_areas
    securitytoken: xx-xxx-x-xxxxx # api token
    charges: 0 # optional, additional charges per kWh
    tax: 2.3333 # optional, additional tax (0.1 for 10%) 

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.11 # [currency]/kWh

meters:
  - name: HuisMeter
    type: mbmd
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.4.184:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter

  - name: pvHuis
    type: sma 
    uri: 192.168.4.230
    password: x
  - name: pvTuinHuis
    type: sma
    uri: 192.168.4.231
    password: x

sponsortoken: x.x.x-x 

log: debug 
levels:
    core: debug
    cc-main: debug

influx:
  url: http://192.168.4.250:8086
  database: EVCC
  token: "x=="
  org: "x"

chargers:
- name: ChargingPointAlfen
  type: alfen
  uri: 192.168.4.178 # TCP ModBus address
- name: wallbox1
  type: template
  template: pulsarplus
  timeout: 5m

loadpoints:
- title: Voortuin # display name for UI
  charger: ChargingPointAlfen # Wallbox Referenz
  mode: minpv # charge mode (off, now, minpv, pv)
  phases: 0 
  mincurrent: 6 
  maxcurrent: 22
  guardduration: 2m
  enable:
    threshold: 0
    delay: 5m
  disable:
    threshold: 0 # maximum import power (W)
    delay: 3m
  circuit: AutomaatLaadpaalVoortuin
- title: Oprit
  charger: wallbox1
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  circuit: AutomaatLaadpaalOprit

vehicles:
  - name: PolestarBuz
    type: template
    template: polestar
    title: Polestar Buz # Will be displayed in the user interface (optional)
    user: xxxxxx # User account (e.g. email address, user id, etc.)
    password: xxxxxxx # Password of the user account (use single quotes in case of leading zeros)
    vin: xxxxxxx # Required if you own multiple vehicles of the same brand (optional)
    capacity: 64 # Battery capacity in kWh (optional)
```

### Log details

```text
x
```

### What type of operating system are you running?

Docker container

### Version

126

andig, 2024-05-03:
If anything this would need to be added to mbmd. Happy to take a PR there.

closed 2024-05-03 by andig

hyperbart, 2024-05-03:
https://github.com/volkszaehler/mbmd/issues/344