evcc-io/evcc #13718: Solax RCP read failure

longtom99, 2024-05-03:
### Describe the bug

Since upgrade to 126.1 i had this issue the first time!
PV production is not displayed.

### Steps to reproduce

Just see log.

### Configuration details

```yaml
Standard Alfen Wallbox integration
```

### Log details

```text
[lp-1 ] DEBUG 2024/05/03 15:51:41 charge power: 1366W
[lp-1 ] DEBUG 2024/05/03 15:51:41 charge currents: [5.99 0 0]A
[site ] ERROR 2024/05/03 15:51:51 pv 1 power: add[0]: read failed: read tcp 192.168.1.209:49634->192.168.90.99:502: i/o timeout
[site ] DEBUG 2024/05/03 15:51:51 pv power: 0W
[site ] DEBUG 2024/05/03 15:51:51 grid meter: -307W
[site ] DEBUG 2024/05/03 15:51:51 site power: -307W
[lp-1 ] DEBUG 2024/05/03 15:51:51 charge total import: 2716.064kWh
[lp-1 ] DEBUG 2024/05/03 15:51:51 charger status: C
[lp-1 ] DEBUG 2024/05/03 15:51:51 charge start energy: 2.72e+03kWh
[lp-1 ] INFO 2024/05/03 15:51:51 start charging ->
[lp-1 ] DEBUG 2024/05/03 15:51:51 wake-up timer: stop
[lp-1 ] DEBUG 2024/05/03 15:51:51 soc estimated: 35.20% (vehicle: 34.00%)
[lp-1 ] DEBUG 2024/05/03 15:51:51 vehicle soc: 35%
[lp-1 ] DEBUG 2024/05/03 15:51:51 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/05/03 15:51:51 vehicle range: 129km
[lp-1 ] DEBUG 2024/05/03 15:51:51 pv charge current: 7.33A = 6A + 1.33A (-307W @ 1p)
[lp-1 ] DEBUG 2024/05/03 15:51:51 pv timer reset
[lp-1 ] DEBUG 2024/05/03 15:51:51 pv timer inactive
[lp-1 ] DEBUG 2024/05/03 15:51:51 max charge current: 7.33A
[site ] DEBUG 2024/05/03 15:52:11 ----
[lp-1 ] DEBUG 2024/05/03 15:52:11 charge power: 1622W
[lp-1 ] DEBUG 2024/05/03 15:52:11 charge currents: [7.1 0 0]A
[site ] DEBUG 2024/05/03 15:52:12 pv power: 2791W
[site ] DEBUG 2024/05/03 15:52:14 grid meter: -26W
[site ] DEBUG 2024/05/03 15:52:14 site power: -26W
[lp-1 ] DEBUG 2024/05/03 15:52:14 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/03 15:52:14 charge total import: 2716.075kWh
[lp-1 ] DEBUG 2024/05/03 15:52:14 charger status: C
[lp-1 ] DEBUG 2024/05/03 15:52:14 soc estimated: 35.21% (vehicle: 34.00%)
[lp-1 ] DEBUG 2024/05/03 15:52:14 vehicle soc: 35%
[lp-1 ] DEBUG 2024/05/03 15:52:14 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/05/03 15:52:14 vehicle range: 129km
[lp-1 ] DEBUG 2024/05/03 15:52:14 pv charge current: 7.45A = 7.33A + 0.113A (-26W @ 1p)
[lp-1 ] DEBUG 2024/05/03 15:52:14 max charge current: 7.45A
[site ] DEBUG 2024/05/03 15:52:41 ----
[lp-1 ] DEBUG 2024/05/03 15:52:41 charge power: 1652W
[lp-1 ] DEBUG 2024/05/03 15:52:41 charge currents: [7.23 0 0]A
[site ] ERROR 2024/05/03 15:52:51 pv 1 power: add[0]: read failed: read tcp 192.168.1.209:47588->192.168.90.99:502: i/o timeout
[site ] DEBUG 2024/05/03 15:52:51 pv power: 0W
[site ] DEBUG 2024/05/03 15:52:51 grid meter: 19W
[site ] DEBUG 2024/05/03 15:52:51 site power: 19W
[lp-1 ] DEBUG 2024/05/03 15:52:51 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/03 15:52:51 charge total import: 2716.092kWh
[lp-1 ] DEBUG 2024/05/03 15:52:51 charger status: C
[lp-1 ] DEBUG 2024/05/03 15:52:51 soc estimated: 35.23% (vehicle: 34.00%)
[lp-1 ] DEBUG 2024/05/03 15:52:51 vehicle soc: 35%
[lp-1 ] DEBUG 2024/05/03 15:52:51 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/05/03 15:52:51 vehicle range: 129km
[lp-1 ] DEBUG 2024/05/03 15:52:51 pv charge current: 7.37A = 7.45A + -0.0826A (19W @ 1p)
[lp-1 ] DEBUG 2024/05/03 15:52:52 max charge current: 7.37A
[site ] DEBUG 2024/05/03 15:53:11 ----
[lp-1 ] DEBUG 2024/05/03 15:53:11 charge power: 1653W
[... 140 lines cut ...]
[site ] DEBUG 2024/05/03 15:57:41 ----
[lp-1 ] DEBUG 2024/05/03 15:57:41 charge power: 1614W
[lp-1 ] DEBUG 2024/05/03 15:57:41 charge currents: [7.03 0 0]A
[site ] ERROR 2024/05/03 15:57:51 pv 1 power: add[0]: read failed: read tcp 192.168.1.209:48396->192.168.90.99:502: i/o timeout
[site ] DEBUG 2024/05/03 15:57:51 pv power: 0W
[site ] DEBUG 2024/05/03 15:57:51 grid meter: 29W
[site ] DEBUG 2024/05/03 15:57:51 site power: 29W
[lp-1 ] DEBUG 2024/05/03 15:57:51 detected active phases: 1p
[lp-1 ] DEBUG 2024/05/03 15:57:51 charge total import: 2716.227kWh
[lp-1 ] DEBUG 2024/05/03 15:57:51 charger status: C
[lp-1 ] DEBUG 2024/05/03 15:57:51 soc estimated: 35.39% (vehicle: 34.00%)
[lp-1 ] DEBUG 2024/05/03 15:57:51 vehicle soc: 35%
[lp-1 ] DEBUG 2024/05/03 15:57:51 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/05/03 15:57:51 vehicle range: 129km
[lp-1 ] DEBUG 2024/05/03 15:57:51 pv charge current: 7.13A = 7.26A + -0.126A (29W @ 1p)
[lp-1 ] DEBUG 2024/05/03 15:57:52 max charge current: 7.13A
```

### What type of operating system are you running?

Linux

### Version

0.126.1

longtom99, 2024-05-03:
Standard Solax and Alfen Eve integration

mdkeil, 2024-05-03:
Greifst Du noch mit einem anderen Tool parallel auf die Modbus-Schnittstelle zu?

longtom99, 2024-05-03:
Nein, nicht per Modbus-TCP der vom Solax WLAN-Dongle kommt.
Ich greife direkt Modbus-RTU vom Solax WR über die RS485 Com-Schnittstelle für Loxone ab.

mdkeil, 2024-05-03:
Ist ggfs. der WLAN-Empfang gestört?-- Der i/o timeout kommt ja nicht immer..

Poste aber bitte dennoch mal deine config, so wie es standard-mäßig bei einem issue gefordert ist. Ist Dir dieses Verhalten schon bei einer vorherigen Version aufgefallen?

mdkeil, 2024-05-03:
Was mich wundert (kann aber Zufall sein), wenn ich mir das gesamte Log anschaue, dass es erst auftritt, nachdem der Ladevorgang gestartet wurde.

longtom99, 2024-05-03:
Ist mir heute das erste Mal richtig aufgefallen, lade aber erst seit 22.4. mit evcc und habe es am 1.5. auf Synology/Docker von VMware/debian portiert.
WLAN Signal ist gut, Sender ist direkt darüber.
evvc.yaml hier (vin maskiert):

`
meters:
- type: template
  template: solax
  id: 1
  host: 192.168.90.99
  port: 502
  usage: pv
  modbus: tcpip
  name: pv
- name: netz
  type: custom
  power:
    source: modbus
    id: 1
    uri: 192.168.90.90:502
    timeout: 5s
    register:
      address: 5
      type: holding
      decode: int32s

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.1.99
  port: 502
  modbus: tcpip
  name: wallbox

vehicles:
- type: template
  template: vw
  title: ID.4
  user: harald.jung@al-enterprise.com
  password: t7aUouW,r
  vin: WVGZZZE2ZPE......
  capacity: 77
  timeout: 10s
  name: ev

loadpoints:
- title: Parkplatz
  charger: wallbox
  mode: pv

site:
  title: Zaus
  meters:
    grid: netz
    pv:
    - pv
`

andig, 2024-05-03:
IO timeout hat immer Netzwerk oder Gerät zur Ursache. Da können wir nix machen.

closed 2024-05-03 by andig

longtom99, 2024-05-03:
Ok, Danke! Werde es weiter beobachten.