evcc-io/evcc #13725: Publish tariff forecast data via WS/MQTT

naltatis, 2024-05-03:
With https://github.com/evcc-io/evcc/pull/13451 forecast data can be consumed via plugin. It would be nice to expose tariff forecast data via WS/MQTT. Currently only available via HTTP. This is useful when integrating with other automation systems like home assistant. It could also be applicable for future topics like solar forecast.

In MQTT this could be a single entry with JSON payload.

andig, 2024-05-09:
We should probably refactor `publishTariffs` then and convert the output into a struct?

naltatis, 2024-05-09:
Makes sense. Currently, we have quite a few `tariff*` properties. And if I see it correctly, they are all published together (if they exist).

andig, 2024-05-10:
Brauchen wir das `publishDelta` oder ist das verfrühte Optimierung?

naltatis, 2024-05-10:
Hm, gute Frage. Kann dir nicht mehr genau sagen ob es dafür eine technische Notwendigkeit gab. Jetzt gerade sehe ich keine. Ich würd sagen "mach raus, und wir gucken was passiert" :D

closed 2025-02-09 by andig