evcc-io/evcc #13732: Darstellungsfehler iPhone 11

AniGerm, 2024-05-03:
### Describe the bug

Auf dem iPhone 11 ist seit einigen Versionen der Menüknopf (die Striche oben in der Ecke) hinter dem Batterie und Wifi Symbol versteckt und fast nicht mehr bedienbar. 
Kann man das Menü bitte wieder etwas tiefer positionieren? 
Vielen Dank!

### Steps to reproduce

1. immer in der Vollbild UI- Ansicht im Safari Browser, ich habe eine Verlinkung zum Homebildschirm erstellt. 

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

V0.126.0

TobiasNeuschulz, 2024-05-03:
Mmmh, entferne bitte mal die Verlinkung auf dem Homebildschirm und erstelle eine Neue. Ich kann mich an eine Diskussion hier erinnern, dass der Fehler anschließend behoben war.

naltatis, 2024-05-03:
Ja, dieser Effekt tauchte bei einigen Nutzer seit dem letzten größeren iOS Update auf. Neu anlegen sollte es beheben. Sonst gerne hier noch mal mit Screenshot und mehr Details schreiben.

closed 2024-05-03 by naltatis

AniGerm, 2024-05-04:
Super! Vielen Dank, hat funktioniert!