evcc-io/evcc #13733: Anleitung: das "docker-compose" Beipiel ist nicht korrekt

martin8310, 2024-05-03:
### Describe the bug

https://docs.evcc.io/docs/installation/docker

Der Punkt unter  " volumes: - /etc/evcc.yaml:/etc/evcc.yaml" kann nicht funktionieren, da in dem Fall ein Ordener angelegt wird und die .yaml Dateil dort hingelangt.

```
version: "3"
services:
  evcc:
    command:
      - evcc
    container_name: evcc
    image: evcc/evcc:latest
    ports:
      - 7070:7070/tcp
      - 8887:8887/tcp
      - 7090:7090/udp
      - 9522:9522/udp
    volumes:
      - /etc/evcc.yaml:/etc/evcc.yaml
      - /home/[user]/.evcc:/root/.evcc
    restart: unless-stopped
```

Bei mir im Portainer Stack funktiokniert:

```
version: "3"
services:
  evcc:
    command:
      - evcc
    container_name: evcc
    image: evcc/evcc:latest
    ports:
      - 7070:7070/tcp
      - 8887:8887/tcp
      - 7090:7090/udp
      - 9522:9522/udp
    volumes:
      - /docker_folder/evcc/evcc.yaml:/etc/
      - /docker_folder/evcc/.evcc:/root/.evcc
    restart: always
```

### Steps to reproduce

1. Man sollte versuchen das Docker Image im Portainer zu starten, dann wird man sehen, dass es nicht funktioniert.
2.
3.
...

### Configuration details

```yaml
version: "3"
services:
  evcc:
    command:
      - evcc
    container_name: evcc
    image: evcc/evcc:latest
    ports:
      - 7070:7070/tcp
      - 8887:8887/tcp
      - 7090:7090/udp
      - 9522:9522/udp
    volumes:
      - /etc/evcc.yaml:/etc/evcc.yaml
      - /home/[user]/.evcc:/root/.evcc
    restart: unless-stopped
```

### Log details

```text
/app # evcc --log debug
[main  ] INFO 2024/05/03 22:53:55 evcc 0.126.1
[main  ] INFO 2024/05/03 22:53:55 missing config file - switching into demo mode
[main  ] INFO 2024/05/03 22:53:55 starting ui and api at :7070
[db    ] INFO 2024/05/03 22:53:55 using sqlite database: /root/.evcc/evcc.db
[site  ] WARN 2024/05/03 22:53:55 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/05/03 22:53:55 vehicle 'blauer e-Golf' does not support automatic detection
[site  ] WARN 2024/05/03 22:53:55 vehicle 'grüner Honda e' does not support automatic detection
[site  ] WARN 2024/05/03 22:53:55 vehicle 'schwarzes VanMoof' does not support automatic detection
[site  ] WARN 2024/05/03 22:53:55 vehicle 'Wärmepumpe' does not support automatic detection
[site  ] INFO 2024/05/03 22:53:55 site config:
[site  ] INFO 2024/05/03 22:53:55   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/05/03 22:53:55     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/05/03 22:53:55     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/05/03 22:53:55     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/05/03 22:53:55   vehicles:
[site  ] INFO 2024/05/03 22:53:55     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/05/03 22:53:55     vehicle 2: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2024/05/03 22:53:55     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/05/03 22:53:55     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/05/03 22:53:55     vehicle 5: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/05/03 22:53:55 loadpoint 1:
[lp-1  ] INFO 2024/05/03 22:53:55   mode:        pv
[lp-1  ] INFO 2024/05/03 22:53:55   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/05/03 22:53:55   meters:      charge ✓
[lp-1  ] INFO 2024/05/03 22:53:55     charge:    power ✓ energy ✗ currents ✗
[lp-2  ] INFO 2024/05/03 22:53:55 loadpoint 2:
[lp-2  ] INFO 2024/05/03 22:53:55   mode:        off
[lp-2  ] INFO 2024/05/03 22:53:55   charger:     power ✗ energy ✗ currents ✗ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/05/03 22:53:55   meters:      charge ✓
[lp-2  ] INFO 2024/05/03 22:53:55     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] DEBUG 2024/05/03 22:53:55 phase timer inactive
[lp-1  ] DEBUG 2024/05/03 22:53:55 pv timer inactive
[lp-1  ] INFO 2024/05/03 22:53:55 vehicle updated: unknown -> blauer e-Golf
[js-shared] DEBUG 2024/05/03 22:53:55 state: {"batterySoc":55,"batterypower":-750,"gridpower":-1000,"loadpoints":[{"chargepower":0,"enabled":true,"maxcurrent":6,"phases":1,"vehicleSoc":62},{"chargepower":0,"enabled":false,"maxcurrent":0,"phases":3,"vehicleSoc":22}],"pvpower":5000,"residualpower":500}
[lp-1  ] DEBUG 2024/05/03 22:53:55 max charge current: 6A
[lp-2  ] DEBUG 2024/05/03 22:53:55 phase timer inactive
[lp-2  ] DEBUG 2024/05/03 22:53:55 pv timer inactive
[lp-2  ] INFO 2024/05/03 22:53:55 vehicle updated: unknown -> weißes Model 3
[site  ] WARN 2024/05/03 22:53:55 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/05/03 22:53:55 ----
[main  ] FATAL 2024/05/03 22:53:55 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
[lp-1  ] DEBUG 2024/05/03 22:53:55 charge power: 1380W
/app # ^C
```

### What type of operating system are you running?

Docker container

### Version

all

mdkeil, 2024-05-04:
Die Anleitung ist korrekt.. es handelt sich um ein bind-mount, sprich du erstellst eine evcc.yaml und 'verlinkst' diese gegen die evcc.yaml-version im docker-container, was diese ersetzt. Verzeichnis und Datei muss auf dem Host-System vorhanden sein, bevor der Container gestartet wird..

[image]

andig, 2024-05-04:
@mdkeil meld dich bitte mal bei info@evcc.io

closed 2024-05-04 by andig

martin8310, 2024-05-04:
Hallo, ja dann kann der Punkt hier geschlossen werden ;-) Auch wenn bei mir es so nicht funktioniert.

mdkeil, 2024-05-04:
@martin8310 gem. deinem Log liegen die Probleme bei dir woanders. Wenn Du nicht weiter kommst, mach einfach mal eine Diskussion auf mit allen nötigen Details und ich schau mal rüber.

martin8310, 2024-05-04:
> @martin8310 gem. deinem Log liegen die Probleme bei dir woanders. Wenn Du nicht weiter kommst, mach einfach mal eine Diskussion auf mit allen nötigen Details und ich schau mal rüber.

Danke, bei mir läuft das einwandfrei, aber danke für das Angebot, sehr nett. Es läuft halt nur nicht so wie im Wiki vorgestellt;-)