evcc-io/evcc #13738: Nachträgliches Ändern des geladenen Fahrzeugs in Ladehistorie nicht möglich

julred, 2024-05-04:
### Describe the bug

Hallo,
es ist nicht mehr möglich im Nachgang das geladene Fahrzeug in der Ladehistorie zu ändern.
Bei einem Klick auf bspw. Gastfahrzeug wird dieses nicht übernommen / abgespeichert.
Habe das Problem das erste mal in V0.125 gesehen.

### Steps to reproduce

1. Ladehistorie öffnen
2. Ladeintrag auswählen
3. Klick auf Fahrzeug -> Gastfahrzeug auswählen
[image]

### Configuration details

```yaml
unabhängig
```

### Log details

```text
betrifft nur UI
```

### What type of operating system are you running?

Linux

### Version

0.126.1

andig, 2024-05-04:
Welcher Aufruf steht denn in der Browserkonsole?

andig, 2024-05-04:
@naltatis ich sehe

```
PUT /api/sessions/id
{"vehicle":null}
```

Das kann nicht funktionieren, oder? Allerdings gibts auch kein HTTP 400, das ist doof.

andig, 2024-05-04:
https://gorm.io/docs/update.html#Update-Changed-Fields:

> NOTE When updating with struct, GORM will only update non-zero fields. You might want to use map to update attributes or use Select to specify fields to update

andig, 2024-05-04:
Closed in https://github.com/evcc-io/evcc/commit/64e45488e853fe188ccba933c32089a65497a83d

closed 2024-05-04 by andig