evcc-io/evcc #13739: Tinkerforge WARP3 Charger Ladeleistung wird nicht angesteuert.

Logirius, 2024-05-04:
### Describe the bug

Der WARP 3 Charger lässt sich mit der neuen Version verbinden. Allerdings steuert er die Ladeleistung nicht an. Es kommt der Fehler [lp-1  ] ERROR 2024/05/04 09:16:27 charge currents: json: cannot unmarshal object into Go value of type []float64

### Steps to reproduce

1. Modus PV
2. Auto anstecken
3. 
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.XXX.XXX
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
[... 516 lines cut ...]
[lp-2  ] INFO 2024/05/04 09:29:43   meters:      charge ✓
[lp-2  ] INFO 2024/05/04 09:29:43     charge:    power ✓ energy ✓ currents ✗
[site  ] DEBUG 2024/05/04 09:29:43 set priority soc: 60
[lp-1  ] DEBUG 2024/05/04 09:29:43 phase timer inactive
[lp-1  ] DEBUG 2024/05/04 09:29:43 pv timer inactive
[lp-2  ] DEBUG 2024/05/04 09:29:43 phase timer inactive
[lp-2  ] DEBUG 2024/05/04 09:29:43 pv timer inactive
[site  ] WARN 2024/05/04 09:29:43 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/05/04 09:29:43 ----
[lp-1  ] DEBUG 2024/05/04 09:29:43 charge power: 0W
[lp-1  ] ERROR 2024/05/04 09:29:43 charge currents: json: cannot unmarshal object into Go value of type []float64
[lp-2  ] DEBUG 2024/05/04 09:29:43 charge power: 0W
[site  ] DEBUG 2024/05/04 09:29:45 pv power: 1800W
[site  ] DEBUG 2024/05/04 09:29:46 battery soc: 97%
[site  ] DEBUG 2024/05/04 09:29:46 battery power: 230W
[site  ] DEBUG 2024/05/04 09:29:47 grid meter: -0W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.1

andig, 2024-05-04:
Es braucht ein trace log

Logirius, 2024-05-04:
Ich hoffe das hier ist richtig. Ich habe log=trace gesetzt

warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.9499969,229.4299927,229.4400024,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.6066742,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825820441,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825820441}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/evse/low_level_state: '{"val":"{\"led_state\":1,\"cp_pwm_duty_cycle\":1000,\"adc_values\":[3909,3464,177,234,2114,3738,175],\"voltages\":[12000,9132,-12065,-11691,1703,12049,-12071],\"resistances\":[2691,696],\"gpio\":[false,false,false,true,false,false,false,true,false,true,false,true,true,true,false,false,false,false,true,false,false,false,false,false],\"charging_time\":0,\"time_since_state_change\":151742,\"uptime\":259381365,\"time_since_dc_fault_check\":1877159,\"temperature\":4137,\"phases_current\":3,\"phases_requested\":3,\"phases_state\":0,\"phases_info\":0,\"dc_fault_pins\":0,\"dc_fault_sensor_type\":1}","ack":true,"ts":1714825836808,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825836808}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/evse/low_level_state: '{"val":"{\"led_state\":1,\"cp_pwm_duty_cycle\":1000,\"adc_values\":[3910,3465,177,234,2113,3738,175],\"voltages\":[12007,9138,-12065,-11691,1702,12049,-12071],\"resistances\":[2692,695],\"gpio\":[false,false,false,true,false,false,false,true,false,true,false,true,true,true,false,false,false,false,true,false,false,false,false,false],\"charging_time\":0,\"time_since_state_change\":96682,\"uptime\":259326305,\"time_since_dc_fault_check\":1822099,\"temperature\":4137,\"phases_current\":3,\"phases_requested\":3,\"phases_state\":0,\"phases_info\":0,\"dc_fault_pins\":0,\"dc_fault_sensor_type\":1}","ack":true,"ts":1714825781656,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825781656}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.8999939,229.4599915,229.4400024,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.5999908,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825859586,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825859586}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.9299927,229.4299927,229.4199982,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.5933228,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825838342,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825838342}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.9499969,229.4199982,229.4400024,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.6033325,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825847532,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825847532}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.9299927,229.4099884,229.4199982,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.5866547,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825828903,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825828903}'
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/evse/low_level_state: '{"val":"{\"led_state\":1,\"cp_pwm_duty_cycle\":1000,\"adc_values\":[3909,3464,177,234,2110,3737,175],\"voltages\":[12000,9132,-12065,-11691,1700,12046,-12071],\"resistances\":[2691,693],\"gpio\":[false,false,false,true,false,false,false,true,false,true,false,true,true,true,false,false,false,false,true,false,false,false,false,false],\"charging_time\":0,\"time_since_state_change\":163601,\"uptime\":259393224,\"time_since_dc_fault_check\":1889018,\"temperature\":4137,\"phases_current\":3,\"phases_requested\":3,\"phases_state\":0,\"phases_info\":0,\"dc_fault_pins\":0,\"dc_fault_sensor_type\":1}","ack":true,"ts":1714825848594,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825848594}'
[modbus] TRACE 2024/05/04 14:31:05 modbus: recv 00 5e 00 00 00 05 01 03 02 01 7e
[modbus] TRACE 2024/05/04 14:31:05 modbus: send 00 5f 00 00 00 06 01 03 06 9a 00 02
[... cut ...]
[warp  ] TRACE 2024/05/04 14:31:05 recv warp3/CP/meter/all_values: '{"val":"[229.9299927,229.4599915,229.4299927,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.6066437,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714825850317,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714825850317}'

andig, 2024-05-04:
@rtrbt soweit ich das verstehe war `meter/all_values` mal ein array, ist jetzt aber ein struct:

```
{
    "val": "[229.9499969,229.4299927,229.4400024,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,229.6066742,0,0,0,0,0,0,0,null,127.5299988,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,127.5299988,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]",
    "ack": true,
    "ts": 1714825820441,
    "q": 0,
    "from": "system.adapter.mqtt.0",
    "user": "system.user.admin",
    "lc": 1714825820441
}
```

Ist das eine gewollte Änderung? Liegt sie an der Warp oder (Idee?) auch wieder an merkwürdigen Dingen im ioBroker?

@Logirius Die Werte sehe **SEHR** komisch aus. Verwendest Du ioBroker? Sind da irgendwelche Konvertierungen eingestellt?

andig, 2024-05-04:
Ich zitiere:

> ...da die Implementierung von MQTT in ioBroker nicht standardkonform ist. Statt einem nicht standardkonformen Server mehrere nicht standardkonforme Server laufen zu lassen, macht meiner Meinung nach keinen Sinn. Was hilft, ist, den MQTT-Broker von ioBroker komplett zu deaktivieren und einen standardkonformen MQTT-Broker wie z. B. Mosquitto zu verwenden.

Logirius, 2024-05-04:
Nein alles läuft in Homeassistant. Ich habe dort den Mosquito Broker installiert. Der evcc läuft auf dem Homeassistant als Addon. Der Homeassistant selbst läuft mit dem HA OS. Diese VM des Homeassistant läuft wiederum auf einem Proxmox.

Logirius, 2024-05-04:
[image]
[image]

Logirius, 2024-05-04:
Ich glaube das hilft dir vermutlich nicht, aber zur Sicherheit. Ich wüsste nicht wo ich da irgendwelche Konvertierung eingestellt haben könnte.

Logirius, 2024-05-04:
[image]

andig, 2024-05-04:
Ich denke wir müssen rausfinden woher diese komischen JSON Nachrichten kommen. Ich glaube irgendwie nicht, dass die WARP die sendet.

Logirius, 2024-05-04:
Ich habe gerade mit dem MQTT Analyzer geschaut. Hilft so etwas? warp3/CP/meters/0/values {"val":"[226.2299957,225.9599915,225.8399963,11.71999931,11.65999985,11.63999939,2648,2628,2625,2653.306641,2635.907715,2630.260498,167.7259674,204.0232849,166.2693634,0.998000026,0.997000039,0.998000026,0.063256636,0.077479318,0.06325569,226.0100098,11.67333221,35.01999664,7902,7919.474609,538.0186157,0.997000039,0.203991652,135.6299896,0,0.071602747,135.6299896,135.6299896,135.6299896,0]","ack":true,"ts":1714830105305,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714830105305}

Logirius, 2024-05-04:
Leider kenne ich mich viel zu wenig aus wie ich da helfen könnte um die notwendigen Infos zu beschaffen

Logirius, 2024-05-04:
Wenn ich im Mosquito auf Topic „warp3/CP/meter/all_values„ zuhören gehe kommt diese Ausgabe 

Nachricht 979 empfangen auf warp3/CP/meter/all_values um 15:45:
{
    "val": "[226.2999878,226.0399933,225.8300018,11.64999962,11.64999962,11.64999962,2631,2629,2627,2646.881348,2634.268555,2632.264404,289.5168457,166.5232697,166.394104,0.994000018,0.998000026,0.998000026,0.109599739,0.063256636,0.06325569,226.0566559,11.66333294,34.98999786,7890,7913.414063,622.4342041,0.997000039,0.236112058,null,136.0500031,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0.040217306,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,136.0500031,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]",
    "ack": true,
    "ts": 1714830298974,
    "q": 0,
    "from": "system.adapter.mqtt.0",
    "user": "system.user.admin",
    "lc": 1714830298974
}

Ich habe dort den Button angeschalten Json Inhalt formatieren. Ohne den Button siehts so aus

Nachricht 1108 empfangen auf warp3/CP/meter/all_values um 15:46:
{"val":"[227.0399933,224.7099915,226.1100006,11.71999931,11.67999935,11.64999962,2657,2620,2629,2662.324463,2627.883545,2636.910645,168.2928925,203.4010773,204.0992432,0.998000026,0.997000039,0.997000039,0.063254751,0.077478543,0.077478543,225.9533234,11.68333149,35.04999542,7908,7927.118652,575.7932129,0.997000039,0.218211845,null,136.2599945,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,0.060515366,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,136.2599945,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]","ack":true,"ts":1714830394640,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1714830394640}
QoS: 0 - Retain: false

andig, 2024-05-04:
Es hilft nicht weiter zu spekulieren. Die Werte sind Schrott, wo auch immer die her kommen.

Logirius, 2024-05-05:
Guten Morgen, ich habe gestern noch das neueste Release aufgespielt die 0.126.2 danach ist der Fehler scheinbar verschwunden. Allerdings regelt er die Leistung noch nicht wirklich auf mehr als 8 Ampere hoch. Er lädt immer noch in die Hausbatterie obwohl diese auf 50 % konfiguriert wurde und sie schon einen SOC über 80 hat. Man muss dazu sagen es handelt sich um eine Nulleinspeisungsanlage, er erkennt aber die überschüssige Leistung anhand der Leistung die in die Batterie fließen würde. Das hatte in Tests mit einem mobilen Ladegerät (an Homematic Schaltsteckdose) sehr gut funktioniert bis die Batterie auf SOC 100 war. Dann kann er den Überschuss natürlich nicht mehr erkennen. Anbei ein Tracelog. Ich entschuldige mich vorab falls das hier falsch und unverschämt ist. 

[homematic] TRACE 2024/05/05 08:37:21 <?xml version="1.0" encoding="UTF-8"?> <methodCall><methodName>getParamset</methodName><params><param><value><string>0034DF29A3621E:3</string></value></param><param><value><string>VALUES</string></value></param></params></methodCall> -- <?xml version='1.0' encoding='ISO-8859-1'?><methodResponse><params><param><value><struct><member><name>SECTION_STATUS</name><value><i4>0</i4></value></member><member><name>PROCESS</name><value><i4>0</i4></value></member><member><name>STATE</name><value><boolean>0</boolean></value></member><member><name>SECTION</name><value><i4>0</i4></value></member></struct></value></param></params></methodResponse>
[site ] DEBUG 2024/05/05 08:37:21 ----
[lp-1 ] DEBUG 2024/05/05 08:37:21 charge power: 1812W
[lp-1 ] DEBUG 2024/05/05 08:37:21 charge currents: [8.07 0 0]A
[lp-2 ] DEBUG 2024/05/05 08:37:21 charge power: 0W
[modbus] TRACE 2024/05/05 08:37:21 modbus: send c4 e6 [00 00 00 06 01 03 05](tel:00 00 00 06 01 03 05) c4 00 01
[modbus] TRACE 2024/05/05 08:37:22 modbus: recv c4 e6 [00 00 00 05 01 03 02](tel:00 00 00 05 01 03 02) 00 20
[modbus] TRACE 2024/05/05 08:37:22 modbus: send c4 e7 [00 00 00 06 01 03 06](tel:00 00 00 06 01 03 06) 86 00 02
[warp ] TRACE 2024/05/05 08:37:22 recv warp3/CP/evse/low_level_state: '{"led_state":4,"cp_pwm_duty_cycle":133,"adc_values":[3906,2985,178,200,2264,3734,177],"voltages":[11981,6043,-12046,-11923,1824,12036,-12058],"resistances":[826,805],"gpio":[false,false,false,true,false,false,false,false,false,false,true,false,true,false,false,false,false,false,true,false,false,false,false,false],"charging_time":4187515,"time_since_state_change":4187515,"uptime":324585138,"time_since_dc_fault_check":50297662,"temperature":2775,"phases_current":1,"phases_requested":1,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":1}'
[warp ] TRACE 2024/05/05 08:37:22 recv warp3/CP/meter/all_values: '[224.6100006,230.7200012,231.6599884,8.069999695,0,0,1812,0,0,1815.631226,0,0,114.7725983,0,0,0.998000026,0,0,0.06325569,0,0,228.9966583,2.689999819,8.069999695,1812,1815.631226,114.7725983,0.998000026,0.06325569,null,145.1699982,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,8.069999695,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,145.1699982,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]'
[modbus] TRACE 2024/05/05 08:37:22 modbus: recv c4 e7 [00 00 00 07 01 03 04](tel:00 00 00 07 01 03 04) 00 00 fd ec
[site ] DEBUG 2024/05/05 08:37:22 pv power: 3200W
[modbus] TRACE 2024/05/05 08:37:22 modbus: send c4 e8 [00 00 00 06 01 03 06](tel:00 00 00 06 01 03 06) 06 00 01
[modbus] TRACE 2024/05/05 08:37:23 modbus: recv c4 e8 [00 00 00 05 01 03 02](tel:00 00 00 05 01 03 02) 00 3c
[modbus] TRACE 2024/05/05 08:37:23 modbus: send c4 e9 [00 00 00 06 01 03 06](tel:00 00 00 06 01 03 06) 9a 00 02
[warp ] TRACE 2024/05/05 08:37:23 recv warp3/CP/evse/low_level_state: '{"led_state":4,"cp_pwm_duty_cycle":133,"adc_values":[3907,2987,178,198,2264,3734,177],"voltages":[11987,6056,-12052,-11923,1824,12036,-12058],"resistances":[829,805],"gpio":[false,false,false,true,true,false,false,false,false,false,true,false,true,false,false,false,false,false,true,false,false,false,false,false],"charging_time":4188538,"time_since_state_change":4188538,"uptime":324586161,"time_since_dc_fault_check":50298685,"temperature":2775,"phases_current":1,"phases_requested":1,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":1}'
[warp ] TRACE 2024/05/05 08:37:23 recv warp3/CP/meter/all_values: '[224.6100006,230.7200012,231.6799927,8.069999695,0,0,1812,0,0,1815.631226,0,0,114.7725983,0,0,0.998000026,0,0,0.06325569,0,0,229.0033417,2.689999819,8.069999695,1812,1815.631226,114.7725983,0.998000026,0.06325569,null,145.1699982,0,null,null,null,null,null,null,null,null,null,null,null,null,null,null,8.069999695,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,145.1699982,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null]'
[modbus] TRACE 2024/05/05 08:37:23 modbus: recv c4 e9 [00 00 00 07 01 03 04](tel:00 00 00 07 01 03 04) 00 00 62 ee
[modbus] TRACE 2024/05/05 08:37:23 modbus: send c4 ea [00 00 00 06 01 03 06](tel:00 00 00 06 01 03 06) 08 00 01
[warp ] TRACE 2024/05/05 08:37:24 recv warp3/CP/evse/low_level_state: '{"led_state":4,"cp_pwm_duty_cycle":133,"adc_values":[3907,2986,179,199,2264,3734,177],"voltages":[11987,6050,-12052,-11929,1824,12036,-12058],"resistances":[827,805],"gpio":[false,false,false,true,false,false,false,true,false,false,true,false,true,false,true,false,false,fals [... cut ...] null,null,null,null,null,null,null,null,null,null,null,null]'
[modbus] TRACE 2024/05/05 08:37:25 modbus: recv c4 ec [00 00 00 05 01 03 02](tel:00 00 00 05 01 03 02) 00 00
[modbus] TRACE 2024/05/05 08:37:25 modbus: send c4 ed 00 [00 00 06 01 03 04](tel:00 00 06 01 03 04) 9e 00 01
[modbus] TRACE 2024/05/05 08:37:25 modbus: recv c4 ed 00 [00 00 05 01 03 02 00](tel:00 00 05 01 03 02 00) 00
[modbus] TRACE 2024/05/05 08:37:25 modbus: send c4 ee [00 00 00 06 01 03 04](tel:00 00 00 06 01 03 04) a9 00 01

Logirius, 2024-05-05:
Anbei noch das Log vom WarpCharger. Ich glaube zu erkennen das der EVCC Currents um die 8,07 einstellen will. Das wird im WarpCharger auch nicht übernommen, er steht auf 8,000 A. Es scheint als ob die mA Regelung auch nicht greift. Es steht ganz unten auch eine Fehlermeldung das er eine JSON Anforderung nicht interpretieren kann. 

2024-05-05T07-00-04-949Z
Scroll down for event log!

{"uptime": 190241,
 "free_heap_bytes":72428,
 "largest_free_heap_block":65524,
 "devices": [ {"UID":"2aP4", "DID":2167, "port":"A"},{"UID":"29sG", "DID":286, "port":"B"}],
 "error_counters": [ {"port": "A", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0},{"port": "B", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0},{"port": "C", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0},{"port": "D", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0},{"port": "E", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0},{"port": "F", "SpiTfpChecksum": 0, "SpiTfpFrame": 0, "TfpFrame": 0, "TfpUnexpected": 0}],
 "info/modules": {"event_log":true,"warp_esp32_i2c":true,"warp_esp32_rtc":false,"esp32_ethernet_brick":true,"watchdog":true,"uptime_tracker":true,"debug_protocol":true,"evse_v2":true,"evse_common":true,"evse_led":false,"network":true,"ethernet":true,"wifi":true,"ntp":true,"rtc":false,"rtc_bricklet":false,"mqtt":true,"mqtt_auto_discovery":true,"http":true,"ws":true,"event":true,"firmware_update":true,"meters":true,"meters_evse_v2":true,"meters_api":true,"meters_modbus_tcp":false,"meters_sun_spec":true,"meters_legacy_api":true,"require_meter":true,"charge_manager":true,"cm_networking":true,"charge_tracker":true,"power_manager":true,"users":true,"nfc":true,"device_name":true,"modbus_tcp":true,"wireguard":true,"ocpp":true,"charge_limits":true,"coredump":true,"pwa":true,"certs":true,"automation":true},
 "info/features": ["evse","cp_disconnect","button_configuration","ethernet","meters","nfc","phase_switch","meter","meter_all_values","meter_phases","rtc"],
 "info/version": {"firmware":"2.3.0-660fed4b","config":"2.3.0","config_type":"warp"},
 "event_log/boot_id": {"boot_id":3757170406},
 "esp32/temperature": {"temperature":3643},
 "info/last_boots": [{"reset_reason":3,"boot_count":13,"timestamp_min":28570429,"uptime":16864,"uptime_overflows":0},{"reset_reason":3,"boot_count":14,"timestamp_min":28570515,"uptime":5161904,"uptime_overflows":0},{"reset_reason":3,"boot_count":2,"timestamp_min":28580117,"uptime":240301747,"uptime_overflows":0},{"reset_reason":3,"boot_count":3,"timestamp_min":28580136,"uptime":1146639,"uptime_overflows":0},{"reset_reason":3,"boot_count":4,"timestamp_min":28580432,"uptime":17704921,"uptime_overflows":0},{"reset_reason":3,"boot_count":5,"timestamp_min":28580454,"uptime":1325771,"uptime_overflows":0},{"reset_reason":3,"boot_count":6,"timestamp_min":28580513,"uptime":3554449,"uptime_overflows":0},{"reset_reason":3,"boot_count":8,"timestamp_min":28580520,"uptime":37023,"uptime_overflows":0},{"reset_reason":3,"boot_count":9,"timestamp_min":28580539,"uptime":1175272,"uptime_overflows":0},{"reset_reason":3,"boot_count":10,"timestamp_min":28580731,"uptime":11514747,"uptime_overflows":0}],
 "evse/identity": {"uid":"2aP4","connected_uid":"0","position":"A","hw_version":"1.0.0","fw_version":"2.2.3","device_identifier":2167},
 "evse/state": {"iec61851_state":2,"charger_state":3,"contactor_state":13,"contactor_error":0,"allowed_charging_current":8000,"error_state":0,"lock_state":0,"dc_fault_current_state":0},
 "evse/hardware_configuration": {"jumper_configuration":3,"has_lock_switch":false,"evse_version":30,"energy_meter_type":6},
 "evse/low_level_state": {"led_state":4,"cp_pwm_duty_cycle":133,"adc_values":[3906,2985,179,198,2264,3734,177],"voltages":[11981,6043,-12046,-11923,1824,12036,-12058],"resistances":[826,805],"gpio":[false,false,false,true,false,false,false,true,false,true,true,false,true,false,false,false,false,false,true,false,false,false,false,false],"charging_time":5551027,"time_since_state_change":5551027,"uptime":325948650,"time_since_dc_fault_check":51661174,"temperature":2850,"phases_current":1,"phases_requested":1,"phases_state":0,"phases_info":0,"dc_fault_pins":0,"dc_fault_sensor_type":1},
 "evse/button_state": {"button_press_time":0,"button_release_time":0,"button_pressed":false},
 "evse/slots": [{"max_current":16000,"active":true,"clear_on_disconnect":false},{"max_current":20000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":false},{"max_current":32000,"active":true,"clear_on_disconnect":false},{"max_current":8000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":true},{"max_current":0,"active":false,"clear_on_disconnect":true},{"max_current":15000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":false},{"max_current":32000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":true,"clear_on_disconnect":false},{"max_current":32000,"active":false,"clear_on_disconnect":false}],
 "evse/indicator_led": {"indication":-1,"duration":0,"color_h":0,"color_s":0,"color_v":0},
 "evse/external_current": {"current":15000},
[... cut ...]
2024-05-05 08:57:08,707 | wifi             | Got IP address: 192.168.xxx.xxx/24. Own MAC address: 58:xx:xx:xx:xx:44
2024-05-05 08:57:16,829 | ntp              | NTP synchronized at 22,056!
2024-05-05 08:57:22,154 | mqtt             | Connected to broker.
2024-05-05 08:57:22,305 | mqtt             | On evse/external_current_update: Failed to deserialize: JSON payload was longer than expected and possibly contained unknown keys.
2024-05-05 08:57:22,605 | mqtt             | Topic power_manager/external_control_update is an action. Ignoring retained message (data_len=147).

Logirius, 2024-05-05:
Ich habe ihn jetzt beobachtet, 1Phasen auf 3 Phasenumschaltung funktioniert. Aktuell habe ich wechselnde Bewölkung. Er regelt aber nur im ganzen Amperebereich 6 A, 7A, 8A… Dann kann es natürlich nicht mit der feinen Ausregelung klappen.

closed 2024-05-05 by andig

andig, 2024-05-05:
Über MQTT kommen plötzlich ganz andere Nachrichten als vorher. Da Du anscheinend neben evcc an anderen Einstellungen rumbastelst, aber nichts davon hier transparent ist habe ich zugemacht. Anscheinend kein Thema von evcc sondern Deiner sonstigen Infrastruktur.