evcc-io/evcc #13752: Doku Templates: `advanced` Optionen ebenfalls anzeigen

VolkerK62, 2024-05-04:
Es kommen gelegentlich Nachfragen/"Fehlermeldungen" (Beispiel: Victron `minsoc` oder `capacity` bei Batteriespeichern) die u. U. nicht aufkommen würden, wenn diese erweiterten Einstellungen ebenfalls in der Doku sichtbar wären.
Deshalb die Idee, diese dort ebenfalls zu zeigen. Evtl abgesetzt mit einem entsprechenden Hinweis.

andig, 2024-05-05:
@naltatis gab es nicht die "erweiterte" Konfiguration sonst immer auf einem zweiten Tab? Bin mir nicht mehr sicher.

naltatis, 2024-05-05:
Die Advanced Optionen haben wir bislang nie angezeigt. Das war ursprünglich mal der Plan. Mein Vorschlag wäre, dass wir das Advanced Flag einfach abschaffen und die Werte einfach zu "echten" optionalen Feldern machen. 

> gab es nicht die "erweiterte" Konfiguration sonst immer auf einem zweiten Tab?

Hatte wir damals mal drüber gesprochen. Wir verwenden die Tabs in der Doku schon für "Usage".

closed 2024-05-31 by github-actions[bot]