evcc-io/evcc #13764: Batterie sperren funktioniert nicht (mehr)

stna1981, 2024-05-05:
### Describe the bug

Die Option, beim schnellen Laden die Hausbatterie gegen Entladen zu sperren funktioniert nicht mehr. Früher hat das funktioniert, aber als ich es zuletzt getestet habe, wurde die Batterie trotzdem entladen.

### Steps to reproduce

Ich nutze v0.124.10, weil 0.125 bei mir ein anderes Problem verursacht. Ich verwende einen Kostal PLENTICORE 5.5 mit angeschlossenem BYD HVS 5.5, evcc läuft auf einem Raspberry, Wallbox ist die ABB Terra AC, die über einen Elfin EW11 per ModBus angesprochen wird. Alle Firmwares sind jeweils aktuell.

Witzigerweise zeigt er im UI sogar an, dass die Batterie gesperrt ist, sie entlädt aber trotzdem. Sieht so aus, als würde das Setzen des minSOC im WR nicht richtig funktionieren?

### Configuration details

```yaml
meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.178.187
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.178.187
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 192.168.178.187
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3
- type: template
  template: kostal-piko-legacy
  usage: pv
  host: 192.168.178.231
  user: xxx
  password: xxx
  name: pv4

chargers:
- type: template
  template: abb
  id: 1
  host: 192.168.178.55
  port: 502
  modbus: rs485tcpip
  name: wallbox6

vehicles:
- type: template
  template: renault
  title: Mégane E-TECH Electric
  user: xxx
  password: xxx
  vin: xxx
  capacity: 60
  phases: 3
  icon: car
  cache: 5m
  mode: pv
  mincurrent: 6
  maxcurrent: 16
  name: ev5
[... 178 lines cut ...]
[site  ] DEBUG 2024/05/05 14:33:20 set battery mode: normal
[site  ] DEBUG 2024/05/05 14:33:32 ----
[lp-1  ] DEBUG 2024/05/05 14:33:32 charge power: 0W
[site  ] DEBUG 2024/05/05 14:33:32 pv power: 2529W
[site  ] DEBUG 2024/05/05 14:33:33 battery soc: 26%
[site  ] DEBUG 2024/05/05 14:33:33 battery power: -134W
[site  ] DEBUG 2024/05/05 14:33:33 grid meter: 29W
[site  ] DEBUG 2024/05/05 14:33:33 site power: -5W
[lp-1  ] DEBUG 2024/05/05 14:33:33 charge voltages: [236 236 235]V
[lp-1  ] DEBUG 2024/05/05 14:33:33 detected connected phases: 3p
[lp-1  ] DEBUG 2024/05/05 14:33:33 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/05/05 14:33:33 charger status: B
[lp-1  ] INFO 2024/05/05 14:33:33 stop charging <-
[lp-1  ] DEBUG 2024/05/05 14:33:33 soc estimated: 37.32% (vehicle: 37.00%)
[lp-1  ] DEBUG 2024/05/05 14:33:33 vehicle soc: 37%
[lp-1  ] DEBUG 2024/05/05 14:33:33 vehicle range: 137km
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-05-05:
>  Früher hat das funktioniert, aber als ich es zuletzt getestet habe, wurde die Batterie trotzdem entladen.

Dann brauchts bitte ein trace log von

    evcc meter --battery-mode hold

von beiden Versionen.

stna1981, 2024-05-05:
Ziemlich seltsam das ganze...mit der alten Version 0.123.2 funktioniert es jetzt auch nicht mehr. Vielleicht gabs da Änderungen in der FW bei Kostal? Trotzdem ist das Verhalten scheinbar ein wenig anders als mit 0.126.2, auch der Trace Log sieht anders aus, im alten Release kommt deutlich mehr zurück.

Was seltsam ist: beim Test mit der 0.123.2 ist mir aufgefallen, dass er nur 1,4kW aus dem Akku zieht und den Rest aus dem Netz. Also irgendwie ein Zwitterverhalten, das für mich gar keinen Sinn ergibt. Denn warum zieht er nicht alles aus dem Akku, mind. bis zu seiner Entladegrenze von 5.5kW?

[image]

Hier das Log von beiden Versionen, wobei auch 0.123.2 den Speicher nicht mehr korrekt sperrt:
[evcc_log_0.123.2 .txt](https://github.com/evcc-io/evcc/files/15213096/evcc_log_0.123.2.txt)
[evcc_log_0.126.2 .txt](https://github.com/evcc-io/evcc/files/15213097/evcc_log_0.126.2.txt)

0.124.10 verhält sich im Übrigen gleich zu 0.126.2., gerade getestet

andig, 2024-05-05:
/cc @premultiply

pustelbaer, 2024-05-06:
may be related to issue [#13069](https://github.com/evcc-io/evcc/issues/13069#issuecomment-2043589423)

andig, 2024-05-07:
> Vielleicht gabs da Änderungen in der FW bei Kostal?

Das wird nur Kostal beantworten können. Anscheinend war die Implementierung ja mal korrekt. Mangels neuer Informationen können wir hier nur zu machen bis jemand eine Idee hat, _wie_ Batteriesperre bei Kostal erneut implementiert werden kann.

stna1981, 2024-05-07:
@andig Das jetzt einfach so zu closen, ohne dass sich um eine Lösung bemüht wird, finde ich jetzt schon etwas ernüchternd. Da muss ich mir dann echt überlegen, ob ich meine monatlichen Sponsor-Zahlungen einstelle. Ich hätte mir zumindest mal gewünscht, dass es eine Rückmeldung zu den Logs oder sowas gibt.

andig, 2024-05-07:
Was erwartest Du denn? Ging mal, jetzt nicht mehr. Ergo Ursache ausserhalb. Wir können leider nicht hellsehen und es gibt auch keine neue Doku von Kostal die bekannt wäre. 

Was und wo sollen wir Deiner Meinung nach denn bitte suchen???

Du hingegen bist Kunde von Kostal und kannst Dich an diese wenden. Die Frage ist ganz einfach: warum funktioniert die Batteriesperre nicht? Wie sie umgesetzt ist kann im Template nachgelesen werden.

stna1981, 2024-05-07:
Naja du hast geschrieben, dass "wir hier nur zu machen" können, bis jemand eine Idee hat. Wenn damit was andere gemeint war, dann sorry, dann habe ich das falsch verstanden.

Hellsehen natürlich nicht. Ich habe ja die beiden Logs angehängt, geben diese denn keinen Anhaltspunkt, woran es scheitert? Die Logs unterscheiden sich ja doch deutlich.

Ich werde mal bei Kostal nachfragen, ob man von denen eine Antwort bekommt ist aber immer Glückssache und am Ende können die sich immer drauf herausreden, dass sie keine Drittanbieter-Lösungen supporten (müssen). Das mit dem Firmware-Update war ja auch nur so eine Idee von mir, da ich es mir anders nicht erklären kann.

andig, 2024-05-08:
Bitte mit Antwort von Kostal wieder melden!

closed 2024-05-08 by andig