evcc-io/evcc #13769: Solar mode nutzt Netz mit voller Leistung anstatt auf Sonne zu warten

elepete, 2024-05-05:
### Describe the bug

Im Solar Mode siehe Bild wurde in der Zeit der gelben Markierung (über den Graphen), die Batterie gelockt und mit max. Power (in meinem Fall 2 Phasen) geladen. Dieses Verhalten ist mir neu.
In den Einstellunen war der Solar Mode aktiv ohne einen Ladeplan 80% zu erreichen.
Das Batterieunterstützte Laden ist in der GUI auf 100% gestellt so dass nur Sonne genutzt werden solle. Wobei ich bisher 90% eingestellt hatte.

Hier ist die gelble Klammer in deren Zeit die Batterie gelockt ist (Ladezustand konstant über die Zeit):
[image]

Hier noch der der Screenshot von EVCC und dieser Ladung: 
[image]

### Steps to reproduce

Ich beobachte gerne und werde zuerst auf 90% Bat. Unterstütztes Laden zurückgehen.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2807
  feedin:
    type: fixed
    price: 0.079
  co2:
    type: grÃ¼nstromindex
    zip: xxx
#    zone: DE

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: xxx
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: xxx
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: xxx
  port: 502
  usage: battery  
  modbus: tcpip  
  name: battery3
#- type: custom
#  power:
#    source: modbus
#    uri: xxx
#    id: 1
#    register:
#      address: 4122
#      type: input
#      decode: float32s
#  name: myaux1

chargers:
- type: template
  template: ellipro 
  ski: xxx  
  ip: xxx  
  name: wallbox5

vehicles:
- type: template
  template: vw
  title: e-up!
  user: xxx
  password: xxx
  vin: xxx
  capacity: 32
  phases: 2
  cache: 15m
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  name: ev4
- type: template
  template: offline
  title: generic 2-phase
  phases: 2
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  name: ev6

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: 
  meters:
    grid: grid1
    pv:
      - pv2
    battery:
      - battery3
#    aux:
#      - myaux1
  residualPower: 200
  maxGridSupplyWhileBatteryCharging: 50

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          -----END CERTIFICATE-----
```

### Log details

```text
[main  ] INFO 2024/05/05 19:59:49 evcc 0.126.2
[main  ] INFO 2024/05/05 19:59:49 using config file: /home/peter/evcc.yaml
[main  ] INFO 2024/05/05 19:59:49 starting ui and api at :7070
[db    ] INFO 2024/05/05 19:59:50 using sqlite database: /home/peter/.evcc/evcc.db
[eebus ] INFO 2024/05/05 19:59:50 Local SKI:  xxx
[eebus ] DEBUG 2024/05/05 19:59:50 mdns: announce
[eebus ] DEBUG 2024/05/05 19:59:50 mdns: using avahi
[eebus ] DEBUG 2024/05/05 19:59:50 starting websocket server on :4712
[eebus ] DEBUG 2024/05/05 19:59:50 mdns: start search
[eebus ] DEBUG 2024/05/05 19:59:51 ski: xxx name: Elli-Wallbox-2136A1LZ0J brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-xxx register: false host: wallbox-xxx.local port: 4712 addresses: [192.168.xxx]
[eebus ] DEBUG 2024/05/05 19:59:51 delaying connection to xxx by 2.641s to minimize double connection probability
[eebus ] DEBUG 2024/05/05 19:59:53 trying to connect to xxx at 192.168.xxx
[eebus ] DEBUG 2024/05/05 19:59:53 initiating connection to 33ff1fdf127772f89613fe197260bb6d0dc1cef3 at 192.168.8.29:4712
[main  ] FATAL 2024/05/05 19:59:54 cannot create vehicle 'ev4': cannot create vehicle type 'template': invalid key: minsoc
[main  ] FATAL 2024/05/05 19:59:54 will attempt restart in: 5m0s
[eebus ] DEBUG 2024/05/05 19:59:54 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:54 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement read 1365080 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:54 Send: d:_i:47859_Elli-Wallbox-xxx reply 2 1365080 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:54 Send: d:_i:47859_Elli-Wallbox-xxx result 3 1365080 ResultData 0
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement reply 1365082 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/05/05 19:59:55 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx read 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx read 7 NodeManagementUseCaseData
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement call 1365085 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx result 8 1365085 ResultData 0
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to DeviceClassification reply 1365088 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement result 1365091 4 ResultData 0
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to DeviceDiagnosis reply 1365092 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement reply 1365094 7 NodeManagementUseCaseData
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement read 1365096 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx reply 9 1365096 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx result 10 1365096 ResultData 0
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-xxx: to NodeManagement read 1365097 NodeManagementDestinationListData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx reply 11 1365097 NodeManagementDestinationListData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx result 12 1365097 ResultData 0
[eebus ] DEBUG 2024/05/05 19:59:55 Recv: d:_i:47859_Elli-Wallbox-XXX: to NodeManagement read 1365098 NodeManagementUseCaseData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx reply 13 1365098 NodeManagementUseCaseData
[eebus ] DEBUG 2024/05/05 19:59:55 Send: d:_i:47859_Elli-Wallbox-xxx result 14 1365098 ResultData 0
^C[eebus ] DEBUG 2024/05/05 20:00:23 mdns: stop announcement
[eebus ] DEBUG 2024/05/05 20:00:23 mdns: stop announcement
[eebus ] DEBUG 2024/05/05 20:00:23 xxx websocket read error:  read tcp 192.168.8.221:55576->192.168.xxx:4712: use of closed network connection
[eebus ] DEBUG 2024/05/05 20:00:23 restarting mdns search
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.2

andig, 2024-05-05:
Ausser Kommunikationsfehlern ist da nichts zu sehen.

closed 2024-05-05 by andig

elepete, 2024-05-06:
Sorry, bei dem Log ist der Teil zu dem der Fehler passierte nicht drin - my bad
Ich werde versuchen dieses Verhalten nochmal zu provozieren um dann besseres Feedback - es ist CO2 kontraproduktiv dass der Solarmodus sich so verhalten kann