evcc-io/evcc #13771: Connection to wallbox after update to 0.126.2 not possible

ThomasPloski, 2024-05-06:
### Describe the bug

After update from 0.125 to 0.125.2, the connection to the already defined wallbox (eon Professional) refused.

The defined ip address and port to wallbox has not been changed after update to 0.126.2 - furthtermore, I am able to logon to the wallbox backend using the mentioned IP address. 

Would you please assist me? 

### Steps to reproduce

1. Update evcc from 0.125 to 0.126.2
2. restart evcc
3. see entries in logs
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 5s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: [exists]

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true
[... 185 lines cut ...]
[lp-1 ] ERROR 2024/05/06 05:58:41 charger status: dial tcp 192.168.178.100:502: connect: connection refused
[site ] DEBUG 2024/05/06 05:58:41 ----
[lp-1 ] ERROR 2024/05/06 05:58:42 charge power: dial tcp 192.168.178.100:502: connect: connection refused
[lp-1 ] ERROR 2024/05/06 05:58:42 charge currents: dial tcp 192.168.178.100:502: connect: connection refused
[site ] DEBUG 2024/05/06 05:58:42 pv power: 0W
[site ] DEBUG 2024/05/06 05:58:42 grid meter: 875W
[site ] DEBUG 2024/05/06 05:58:42 site power: 875W
[lp-1 ] ERROR 2024/05/06 05:58:42 charge total import: dial tcp 192.168.178.100:502: connect: connection refused
[lp-1 ] ERROR 2024/05/06 05:58:42 charger status: dial tcp 192.168.178.100:502: connect: connection refused
[site ] DEBUG 2024/05/06 05:58:45 ----
[lp-1 ] DEBUG 2024/05/06 05:58:45 charge power: 0W
[lp-1 ] DEBUG 2024/05/06 05:58:45 charge currents: [0.03 0.03 0.03]A
[site ] DEBUG 2024/05/06 05:58:45 pv power: 0W
[site ] DEBUG 2024/05/06 05:58:45 grid meter: 864W
[site ] DEBUG 2024/05/06 05:58:45 site power: 864W
[lp-1 ] ERROR 2024/05/06 05:59:46 charge total import: EOF
```

### What type of operating system are you running?

Docker container

### Version

0.126.2

StefanSchoof, 2024-05-06:
Did you try to downgrade to make sure evcc still works with the old version?

ThomasPloski, 2024-05-06:
Not yet, but this will be my fallback, in case there is not quick solution for my issue.

StefanSchoof, 2024-05-06:
Please try. In many cases where user think a evcc update broke something, something else has changed and only the restart of evcc for the update made it visible.

ThomasPloski, 2024-05-06:
With v0.125.0 evcc works fine. Connection to wallbox has been established successfully.

premultiply, 2024-05-06:
Running multiple instances at the same time?

ThomasPloski, 2024-05-06:
No, just one at the same time.

andig, 2024-05-06:
[lp-1 ] ERROR 2024/05/06 05:31:44 charge total import: EOF

Sieht aus wie https://github.com/evcc-io/evcc/issues/13631.

@samuelbrucksch fällt Dir dazu etwas ein? Sonst müssten wir https://github.com/evcc-io/evcc/commit/da4af01d736503b77901648d8a03f45516a99591 zurück rollen.

SamuelBrucksch, 2024-05-06:
Habe es bei mir auch nicht zum laufen bekommen. Am besten zurückrollen, ich richte mir dann mal eine manuelle Morbus Verbindung ein zum testen...

@SamuelBrucksch fällt Dir dazu etwas ein? Sonst müssten wir da4af01 zurück rollen.

andig, 2024-05-06:
Closed in 7d67b502a7600a8231aabb98cc6c144990d62445

closed 2024-05-06 by andig

Logocad, 2024-05-10:
Hallo andig

Ist dieses Problem mit der innogy eBox Professional nun gelöst? Oder kommt die Lösung mit der nächsten Version.
Hab das Problem ebenfalls mit der 0.126.2 wenn ich die 0.125.0 installiere ist wieder alles top,

StefanSchoof, 2024-05-10:
Der Fix ist bisher nur in dem nightly.

Logocad, 2024-05-10:
Danke für die schnelle Antwort. Werde ich dann mal installieren und testen.