evcc-io/evcc #13779: Externe Quelle fehlt  e3dc-rscp

winfotiker, 2024-05-06:
### Describe the bug

Nach Umstellung vom template e3dc auf das e3dc-rscp wird nur noch die Leistung vom internen Wechselrichter angezeigt und nicht mehr die Leistung vom externen Wechselrichter. 
Ebenso fehlt seit dem der Hausverbrauch.
[image]

### Steps to reproduce

Änderung der evcc.yaml entsprechend Anleitung zum e3dc

### Configuration details

```yaml
meters:
- type: template
#  template: e3dc 
  template: e3dc-rscp
  usage: grid  
  host: 192.168.178.20  
#  port: 502  
  port: 5033
  name: grid1
  user: ***
  password: ***
  key: ****
- type: template
  template: e3dc-rscp 
  usage: pv 
  host: 192.168.178.20  
  port: 5033  
  name: pv2
  user: ***
  password: ***
  key: ****
- type: template
  template: e3dc-rscp 
  usage: battery  
  host: 192.168.178.20  
  port: 5033  
  name: battery3
  user: ***
  password: ***
  key: ****
```

### Log details

```text
[main  ] INFO 2024/05/06 12:11:03 evcc 0.126.2
[main  ] INFO 2024/05/06 12:11:03 using config file: /etc/evcc.yaml
[main  ] INFO 2024/05/06 12:11:03 starting ui and api at :7070
[db    ] INFO 2024/05/06 12:11:03 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2024/05/06 12:11:03 Local SKI:  d1e1654222435487ddd585473051fb2cd055386d
[eebus ] DEBUG 2024/05/06 12:11:03 mdns: announce
[eebus ] DEBUG 2024/05/06 12:11:03 mdns: using zeroconf
[eebus ] DEBUG 2024/05/06 12:11:03 starting websocket server on :4712
[eebus ] DEBUG 2024/05/06 12:11:03 mdns: start search
[eebus ] DEBUG 2024/05/06 12:11:03 ski: e6fbee384f24fac2f8cfe1369ffb932d841502f2 name: Elli-Wallbox-2109A184ND brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2109A184ND register: false host: wallbox-2109A184ND.local. port: 4712 addresses: [192.168.178.83]
[eebus ] DEBUG 2024/05/06 12:11:03 delaying connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 by 2.899s to minimize double connection probability
[eebus ] DEBUG 2024/05/06 12:11:04 ski: c04cc6e65dec6a3c308f178017bef67693856a78 name: Elli-Wallbox-2109A181H6 brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2109A181H6 register: false host: wallbox-2109A181H6.local. port: 4712 addresses: [192.168.178.91]
[eebus ] DEBUG 2024/05/06 12:11:04 delaying connection to c04cc6e65dec6a3c308f178017bef67693856a78 by 220ms to minimize double connection probability
[eebus ] DEBUG 2024/05/06 12:11:04 trying to connect to c04cc6e65dec6a3c308f178017bef67693856a78 at 192.168.178.91
[eebus ] DEBUG 2024/05/06 12:11:04 initiating connection to c04cc6e65dec6a3c308f178017bef67693856a78 at 192.168.178.91:4712
[eebus ] DEBUG 2024/05/06 12:11:04 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/06 12:11:04 Recv: d:_i:47859_Elli-Wallbox-2109A181H6: to NodeManagement read 2750854 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/06 12:11:04 Send: d:_i:47859_Elli-Wallbox-2109A181H6 reply 2 2750854 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/06 12:11:04 Send: d:_i:47859_Elli-Wallbox-2109A181H6 result 3 2750854 ResultData 0
[eebus ] DEBUG 2024/05/06 12:11:04 Recv: d:_i:47859_Elli-Wallbox-2109A181H6: to NodeManagement reply 2750856 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/05/06 12:11:04 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/05/06 12:11:04 Send: d:_i:47859_Elli-Wallbox-2109A181H6 read 5 DeviceClassificationManufacturerData
[... 129 lines cut ...]
time="2024-05-06T12:11:13+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-05-06T12:11:13+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-05-06T12:11:13+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-05-06T12:11:13+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2024-05-06T12:11:20+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2024-05-06T12:11:20+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2024-05-06T12:11:20+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2024-05-06T12:11:20+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
[eebus ] DEBUG 2024/05/06 12:11:20 trying to connect to e6fbee384f24fac2f8cfe1369ffb932d841502f2 at 192.168.178.83
[eebus ] DEBUG 2024/05/06 12:11:20 initiating connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 at 192.168.178.83:4712
[eebus ] DEBUG 2024/05/06 12:11:20 connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 failed:  dial tcp 192.168.178.83:4712: connect: connection refused
[eebus ] DEBUG 2024/05/06 12:11:20 trying to connect to e6fbee384f24fac2f8cfe1369ffb932d841502f2 at wallbox-2109A184ND.local.
[eebus ] DEBUG 2024/05/06 12:11:20 initiating connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 at wallbox-2109A184ND.local.:4712
[eebus ] DEBUG 2024/05/06 12:11:20 connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 failed: dial tcp: lookup wallbox-2109A184ND.local. on 192.168.178.1:53: no such host
[eebus ] DEBUG 2024/05/06 12:11:20 restarting mdns search
[eebus ] DEBUG 2024/05/06 12:11:20 delaying connection to e6fbee384f24fac2f8cfe1369ffb932d841502f2 by 24.456s to minimize double connection probability
```

### What type of operating system are you running?

Linux

### Version

0.126.2

andig, 2024-05-06:
Arg, ich erinnere mich. @docolli wie wollten wir die analog Modbus Template mit abfragen?

docolli, 2024-05-06:
Komisch, das tust du doch im Code? Du frägst PV und ADD Power an

```go
	case templates.UsagePV:
		res, err := m.conn.SendMultiple([]rscp.Message{
			*rscp.NewMessage(rscp.EMS_REQ_POWER_PV, nil),
			*rscp.NewMessage(rscp.EMS_REQ_POWER_ADD, nil),
		})
		if err != nil {
			return 0, err
		}

		values, err := rscpValues(res, cast.ToFloat64E)
		if err != nil {
			return 0, err
		}

		return lo.Sum(values), nil
```
Ich habe noch die 0.126.0 im Einsatz und ich habe keine external Power zum Test.

Die Anfragen sollte aber so wie im Codce funktionieren:
```json
./e3dc '[["EMS_REQ_POWER_PV","",""],["EMS_REQ_POWER_ADD","",""]]' | jq
{
  "EMS_POWER_PV": 551,
  "EMS_POWER_ADD": 0
}
```

maikheinrich, 2024-05-06:
Hallo zusammen,

ich könnte das testen für Euch, Ich habe externe Erzeuger und Verbrauchszähler an meinem E3DC. Allerdings habe ich vor Do keine Möglichkeit auf das System zuzugreifen.

Ich meine allerdings, dass der EMS_POWER_ADD Tag negative Werte zurück liefert. Das seit Ihr bei der Modus Integration auch schon drüber gestolpert...  https://github.com/evcc-io/evcc/issues/6433

LG Maik

docolli, 2024-05-07:
@maikheinrich Ja, wäre gut, wenn du mit dem Kommandozeilentool von go-rscp die Ausgabewerte der beiden Befehle testen könntest, damit man sieht, wie sich die Vorzeichen verhalten.

Mir fällt aber oben im Log auf, dass es nur 1 Ausgabe von `pv power: 8510W` gibt, nach dem `[lp-1  ] ERROR 2024/05/06 12:11:11 charger status: timeout` gibt es zwar das Senden von RSCP Anfragen, aber keine weitere Debug Ausgabe von Werten. Vielleicht ist das Problem auch ganz woanders.

andig, 2024-05-07:
Trace log sollte reichen.

docolli, 2024-05-07:
Blöde Frage, aber ich bekomme es grad nicht hin, die Log Ausgaben vom go-rscp zu bekommen, auch wenn ich in der config.yaml `site: trace` setze. Geht das nur, wenn ich evcc manuell aufrufe und dann einen Log-Level übergebe? Welchen?

andig, 2024-05-07:
Closed in https://github.com/evcc-io/evcc/commit/3590574aa0a5f5ce7b8b4c43fb8d740efe29a3e9

closed 2024-05-07 by andig