evcc-io/evcc #13786: Octopus Product code/Tariff code not accpeted

realist625, 2024-05-06:
### Describe the bug

On upgrading from 0.125.0 to  0.126.2 I find that the information from Octopus Energy reading unit price  in no longer available.
The error I get is 'failed configuring grid tariff: cannot create tariff type 'octopusenergy': missing product code', but as you can see, it is there and I have made no changes from previous version.   I have verified the product code with the Octopus API, and I can manually see the relevant tariff rate by: 
https://api.octopus.energy/v1/products/GO-VAR-22-10-14/electricity-tariffs/E-1R-GO-VAR-22-10-14-A/standard-unit-rates/
or
https://api.octopus.energy/v1/products/GO-VAR-22-10-14

Looking at the code in octopus.go and my use case is where I *do not* use the API key, so looking on how that error in generated, well I am not sure how that part of the code works, but it feels like something is amiss here.

### Steps to reproduce

1. Upgrade from 0.125.0 to 0.126.2
2.
3.
...

### Configuration details

```yaml
<details><summary>Konfiguration (/etc/evcc.yaml)</summary>

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 5s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: eastron-sdm220_230
  id: 1
  device: /dev/serial0
  baudrate: 9600
  comset: 8N1
  usage: grid
  modbus: rs485serial
  name: grid1
- type: template
  template: solaredge-inverter
  id: 1
  host: 10.184.63.81
  port: 1502
  usage: pv
  modbus: tcpip
  timeout: 10s
  name: pv1

- type: template
  template: victron-energy
  usage: battery
  host: 10.184.63.123
  port: 502
  capacity: 9
  name: battery1

chargers:
- type: template
  template: keba
  host: 10.184.63.116
  rfid: A63DB84B
  name: BMW_Wallbox
- name: plug_charger
  type: template
  template: tapo
  host: 10.184.63.105 # IP-Adresse oder Hostname
  user: *****
  password: *****
  standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional)type: template

loadpoints:
- title: BMW Wallbox
  charger: BMW_Wallbox
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 32
- title: Plug Charger
  charger: plug_charger
  mode: off

site:
  title: FM1
  meters:
    grid: grid1
    pv:
     - pv1
    battery:
     - battery1
  residualPower: -350

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
#tariffs:
#  currency: GBP # three letter ISO-4217 currency code (default EUR)
tariffs:
  currency: GBP
  grid:
    type: octopusenergy
#    tariff: E-1R-GO-VAR-22-10-14-A
    tariff: GO-VAR-BB-23-02-07
#    tariff: GO-VAR-22-10-14
    region: A

#  grid:
    # either static grid price
#    type: fixed
#    price: 0.41653 # GBP/kWh

    # # or variable via tibber
    # type: tibber
    # cheap: 0.2 # EUR/kWh
    # token: *****
    # homeid: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c" # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # cheap: 0.2 # EUR/kWh
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.00 # GBP/kWh

mqtt:
  broker: 10.184.63.51:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user: *****
  # password: *****

</details>
Version: `0.126.2`
```

### Log details

```text
main  ] INFO 2024/05/06 19:33:55 evcc 0.126.2
[main  ] INFO 2024/05/06 19:33:55 using config file: /etc/evcc.yaml
[main  ] INFO 2024/05/06 19:33:55 starting ui and api at :7070
[db    ] INFO 2024/05/06 19:33:55 using sqlite database: /home/jraubal/.evcc/evcc.db
[mqtt  ] INFO 2024/05/06 19:33:55 connecting evcc-907283016 at tcp://10.184.63.51:1883
[mqtt  ] DEBUG 2024/05/06 19:33:55 tcp://10.184.63.51:1883 connected
[tapo  ] DEBUG 2024/05/06 19:33:56 SMART.TAPOPLUG P110 connected (fw:1.3.0 Build 230905 Rel.152200,hw:1.0,mac:98-25-4A-93-6A-DE)
[main  ] ERROR 2024/05/06 19:33:56 failed configuring grid tariff: cannot create tariff type 'octopusenergy': missing product code
[lp-1  ] DEBUG 2024/05/06 19:33:56 charge total import: 14162.473kWh
```

### What type of operating system are you running?

Linux

### Version

0.126.2

realist625, 2024-05-06:
Just. tested using API key, and that configuration works as expected.  It seems to be only an issue with a config where API key is not used and the product code and region code is used.

andig, 2024-05-07:
/cc @duckfullstop

duckfullstop, 2024-05-07:
Apologies for the regression, I'll have a fix for this tonight. 🙇

closed 2024-05-07 by andig

duckfullstop, 2024-05-07:
@realist625 Just an FYI that the fix simply allows the deprecated `tariff` flag to work: please switch to `productCode` as per the deprecation warning you (should!) now be receiving. (this'll be fixed in documentation with https://github.com/evcc-io/docs/pull/509)

PR #13799 (2024-05-07): tariff/octopusenergy: Fix parsing of Tariff variable
This PR resolves a silly typo I made that causes usage of the old `tariff` configuration variable to fail hard instead of with a deprecation warning.

Also includes a test for parsing (feel free to expand, I kinda ran out of time implementing)

Closes #13786