evcc-io/evcc #13791: different DB for tokens?

Macrisu, 2024-05-07:
**Is your feature request related to a problem? Please describe.**
See [this](https://github.com/evcc-io/evcc/discussions/13758) problem.
We always must delete the evcc.db if we want to renew the tokens. This is confusing and frustrating as we lose all info about charges.

**Describe the solution you'd like**
Use a different DB just for tokens, like **tokens.db** or give the option to purge the expired tokens from the interface

**Additional context**
I cannot add a `mercedes vehicle from the interface. I get this error:

`cannot create vehicle type 'template': invalid key: icon
`
[image]

Sillium007, 2024-05-07:
You can query the database and delete entries according to this comment:
https://github.com/evcc-io/evcc/discussions/13685#discussioncomment-9297402

So there should be no need to delete the whole database

andig, 2024-05-07:
There is no reason to delete either database or tokens from database afaik.