evcc-io/evcc #13796: enphase API working in HTTPS

jaykayone, 2024-05-07:
### Describe the bug

Your enphase template uses the http protocol: templates/definition/meter/enphase.yaml

I have enphase working locally and it only works in https, http sends a "301 moved permanently" code.

Could you add a parameter to specify the protocol to use so that it stays does not break things for people on other versions of the API ?

### Steps to reproduce

1. Add enphase pv to config
2. It tries to communicate with enphase via http instead of https
3. jq gets empty results
...

### Configuration details

```yaml
meters:
  - name: my_pv
    type: template
    template: enphase
    usage: pv
    host: envoy.local 
    token: xxxx
```

### Log details

```text
jq: empty result
```

### What type of operating system are you running?

Linux

### Version

v0.126.2

andig, 2024-05-07:
Should not be necessary if Enphase sends a redirect header.

jaykayone, 2024-05-07:
Yes it follows the redirect according to the trace.

but there is now data shown as you seen in the log, whNow is 205 but the ui shows no data and the log in the UI says: 
pv 1 power: jq: empty result

andig, 2024-05-07:
Please run with --log-trace

jaykayone, 2024-05-07:
i added the log in trace mode in the description

andig, 2024-05-07:
> Please run with --log-trace

jaykayone, 2024-05-07:
```
pi@raspberrypi:~ $ evcc meter --log-headers
[main  ] INFO 2024/05/07 21:25:20 evcc 0.126.2
[main  ] INFO 2024/05/07 21:25:20 using config file: /etc/evcc.yaml
[db    ] INFO 2024/05/07 21:25:20 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT * FROM `settings` 1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/05/07 21:25:20 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[http  ] TRACE 2024/05/07 21:25:20 GET http://envoy.local/production.json
[http  ] TRACE 2024/05/07 21:25:21 
GET /production.json HTTP/1.1
Host: envoy.local
User-Agent: Go-http-client/1.1
Authorization: Bearer eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJhdWQiOiIxMjIzMDMwNDYzNDQiLCJpc3MiOiJFbnRyZXoiLCJlbnBoYXNlVXNlciI6Im93bmVyIiwiZXhwIjoxNzQ2NTMyMzEwLCJpYXQiOjE3MTQ5OTYzMTAsImp0aSI6ImQ4MmQ5MDlhLTI5ZjEtNDFjNi04ZTdiLTBhNGY3MDBhMWYzZSIsInVzZXJuYW1lIjoiamVmZkBrb25uZW4ubHUifQ.rM_A3Gs2EBStEhC6d7XMp0UTGYO-OnUBchAIX4XtobHC8dUZIchQ9IVhxyiP6_hKzBWzQ3LZWpkLnAOI2KmRrQ
Accept-Encoding: gzip

HTTP/1.1 301 Moved Permanently
Content-Length: 175
Connection: keep-alive
Content-Type: text/html
Date: Tue, 07 May 2024 19:25:47 GMT
Location: https://envoy.local/production.json
Server: openresty/1.17.8.1

<html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
[http  ] TRACE 2024/05/07 21:25:21 GET https://envoy.local/production.json
[http  ] TRACE 2024/05/07 21:25:23 
GET /production.json HTTP/1.1
Host: envoy.local
User-Agent: Go-http-client/1.1
Authorization: Bearer eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.eyJhdWQiOiIxMjIzMDMwNDYzNDQiLCJpc3MiOiJFbnRyZXoiLCJlbnBoYXNlVXNlciI6Im93bmVyIiwiZXhwIjoxNzQ2NTMyMzEwLCJpYXQiOjE3MTQ5OTYzMTAsImp0aSI6ImQ4MmQ5MDlhLTI5ZjEtNDFjNi04ZTdiLTBhNGY3MDBhMWYzZSIsInVzZXJuYW1lIjoiamVmZkBrb25uZW4ubHUifQ.rM_A3Gs2EBStEhC6d7XMp0UTGYO-OnUBchAIX4XtobHC8dUZIchQ9IVhxyiP6_hKzBWzQ3LZWpkLnAOI2KmRrQ
Referer: http://envoy.local/production.json
Accept-Encoding: gzip

HTTP/1.1 200 OK
Transfer-Encoding: chunked
Cache-Control: no-cache
Connection: keep-alive
Content-Type: application/json
Date: Tue, 07 May 2024 19:25:49 GMT
Expires: 1
Pragma: no-cache
Server: openresty/1.17.8.1
Set-Cookie: sessionId=TAO9egg7mmJjtB7uzTnKbEkmWOzZc0pc; Secure; HttpOnly; path=/
Strict-Transport-Security: max-age=63072000; includeSubdomains
X-Content-Type-Options: nosniff
X-Frame-Options: DENY

c9
{"production":[{"type":"inverters","activeCount":19,"readingTime":1715108566,"wNow":2,"whLifetime":461493}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[... 28 lines cut ...]
User-Agent: Go-http-client/1.1
[... cut ...]
Pragma: no-cache
Server: openresty/1.17.8.1
Set-Cookie: sessionId=TAO9egg7mmJjtB7uzTnKbEkmWOzZc0pc; Secure; HttpOnly; path=/
Strict-Transport-Security: max-age=63072000; includeSubdomains
X-Content-Type-Options: nosniff
X-Frame-Options: DENY

c9
{"production":[{"type":"inverters","activeCount":19,"readingTime":1715108566,"wNow":2,"whLifetime":461493}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
0
Power:  jq: empty result
Energy: jq: empty result
```

andig, 2024-05-07:
Redirect works as expected. Only there is no `.measurementType == "production"`. Seems the api has changed, too.

Lightmanster, 2024-05-09:
Workaround for now is to use:

meters:
- name: Enphase
  type: custom
  power:
    source: http
    uri: https://ip_envoy/production.json
   method: GET
    auth:
      type: bearer
      password: token
    insecure: true
    jq: .production[0].wNow
    timeout: 60s

andig, 2024-05-09:
Feel free to add a new template. V2 api is not compatible.

BeneWilh, 2024-05-26:
Hi, 

maybe not directly related but I am getting an 

Mai 26 10:24:26 raspi-evcc evcc[6188]: [site  ] ERROR 2024/05/26 10:24:26 grid meter: unexpected status: 401 (Unauthorized)

some hours after configuration. I added the meter (grid & pv) in evcc.yaml and also tried configuration via UI (..and deleted the yaml configuration part).

After putting a new token from enphase data are retrieved but again after some hours I am getting this error.. 

Any hints or workarounds? Thank you!

andig, 2024-05-26:
Maybe the token expires. Impossible to say without logfile and api documentation.

BeneWilh, 2024-05-26:
> Maybe the token expires. Impossible to say without logfile and api documentation.

yes, semms to be the case... haven't found any documentation how/where to put credentials so that the token could get renewed by the system.

What logfile do you need?

olikra, 2024-06-01:
@BeneWilh
Will start using evcc in combination with enphase next week - when new car is in place.

But:

I use the local enphase-API for some data-analytics. The token was valid for quite a long time (+/- 8 month). Since April 2024 the token expires randomly every 6-8 weeks - last expiration yesterday evening at 19:53. At 23:15 the gateway rebooted automatically. A newly generated token does not work today.

The firmware on the gateway is "D7.6.358". What's yours?

andig, 2024-06-01:
Ask Enphase for documentation

Lightmanster, 2024-06-01:
As an installer you could also generate user token. Goto https://entrez.enphaseenergy.com/login
Multiple ways to create tokens of installer and user 
check https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication

Check token in jwt.io and it will tell you expire date.

forum6691, 2024-06-01:
I have the same problem with token have a lifetime only 12h00 si your account enphase is "Installer".

So to avoid this problem, it would be use  an api call with enphase account. 

Is EVCC has this type of access method available for enphase ?

olikra, 2024-06-01:
Found some time to investigate: The certificate was invalid + the local self signed certificate (signed by Enphase) was recreated yesterday evening and did not match with the certificate used by the requesting script. So all okay now - will see how evcc integration will work next week.