evcc-io/evcc #13806: When automatic phase change is activated the "Fast Charge" option stays in the current Phase 1 instead of switich to 3 and going to max charge

felsnerino, 2024-05-08:
### Describe the bug

When it was on PV Charge with automatic phase change and was charging in 1 phase the switch to fast charge is not working properly.
When switching to fast charge it only make it fast within the currently running phases (1<->3). Even though it could switched to 3 phases and fast charge. There is now way currently to switch to 3 phases and fast charge.
Configuration:
[image]

Charging on fast:
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistantprox.knx
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval

#plant: FH2
# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: warn
levels:
  site: warn
  lp-1: warn
  lp-2: warn
  cache: warn
  db: warn

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
#modbusproxy:
  #  - port: 5200
[... 213 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
No error as such.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.126.2

felsnerino, 2024-05-08:
One addition - maybe 2 buttons would be the best - Fast charge 1 phase - Fast charge 3 phases?

mdkeil, 2024-05-08:
pls provide a log, because the some timers in the background, which could prevent phase shifting

`guardDuration: 5m` is deprecated

felsnerino, 2024-05-08:
Added the log from the add-on menu in Home Assistant.

mdkeil, 2024-05-08:
Does your config works with past evcc-versions?

felsnerino, 2024-05-08:
Usually I only use it in PV mode or PV min mode. Only recently I started using 1 phase automatic change and yesterday due to bad weather/little PV excess wanted to charge in Fast Charge mode... so unfortunately I cannot tell. This config I am using for about 1 year through all kinds versions...

andig, 2024-05-08:
Wallbox shows F which is an error state. Fix these errors first, then retry.

closed 2024-05-08 by andig

felsnerino, 2024-05-08:
Sorry, that is misleading - my wife left today at ~9:00 but did not remove the cable on the wallbox - only on the car - which provokes this error. but as you can see in the log - the error was not happening before.

felsnerino, 2024-05-08:
how can I generate the logs in Home Assistant AddOn - here I only see one page - also within the System ->Logs.... If I read another issue correctly there is currently no way to get more logs?!

felsnerino, 2024-05-08:
@andig could you please reopen it? The error you mentioned is not related. If you need more info - please let me know how - I will provide them.

andig, 2024-05-08:
> The log doesn't show anything. We'll need the entire log, not just the resulting error message.

felsnerino, 2024-06-12:
[evcc-20240612-161644-debug.log](https://github.com/user-attachments/files/15806567/evcc-20240612-161644-debug.log)

@andig now with the attached debug protocol. Now the evcc was not able itself to switch - so it was loading with 1p and then wanted to switch to 3p - but got the error. can you please reopen?

andig, 2024-06-12:
> Now the evcc was not able itself to switch 

Which timestamp are you looking at?

felsnerino, 2024-06-12:
yes of course:
First time:  [lp-1  ] WARN 2024/06/12 16:09:29 charger logic error: phases mismatch (got 1, expected 3)
Second time: [lp-1  ] WARN 2024/06/12 16:09:29 charger logic error: phases mismatch (got 1, expected 3)
And yes it was (and stayed) in automatic (PV) Mode. I just saw it by coincidence and took the opportunity... 
The sun was coming and going so that it started with 1phase and then as it got brighter it wanted to change to 3phase as there was enough excess energy...

andig, 2024-06-13:
Kurz vorher hat eine Phasenumschaltung stattgefunden:

```
[lp-1  ] DEBUG 2024/06/12 16:07:29 phase scale3p in 0s
[lp-1  ] DEBUG 2024/06/12 16:07:29 switched phases: 3p
[lp-1  ] DEBUG 2024/06/12 16:07:29 phase timer inactive
[lp-1  ] DEBUG 2024/06/12 16:07:29 pv charge current: 12.9A = 10.6A + 2.27A (-1564W @ 3p)
[lp-1  ] DEBUG 2024/06/12 16:07:29 max charge current: 12.9A

[lp-1  ] WARN 2024/06/12 16:09:29 charger logic error: phases mismatch (got 1, expected 3)
[lp-1  ] DEBUG 2024/06/12 16:09:29 pv charge current: 32.4A = 21.9A + 10.5A (-2426W @ 1p)
```

Diese scheint nicht zu funktionieren da der Charger auf 1p bleibt. Warum auch immer. Bitte Alfen anfragen warum das scheitern kann/sollte.

rudybroersma, 2024-06-13:
I have the same issue. 1p/3p switching works fine while using Solar charging, but doesn't switch to 3p when selecting 'Fast'

[evcc.log](https://github.com/user-attachments/files/15826961/evcc.log)

Relevant entries:
[lp-1 ] WARN 2024/06/13 20:11:30 charger logic error: phases mismatch (got 1, expected 3)  <------------ Phase mismatch
[lp-1 ] DEBUG 2024/06/13 20:11:30 switched phases: 3p  <------- IT DIDNT! REMAINS AT 1p
[site ] DEBUG 2024/06/13 20:11:59 ----
[lp-1 ] DEBUG 2024/06/13 20:11:59 charge power: 5097W
[lp-1 ] DEBUG 2024/06/13 20:11:59 charge currents: [23.6 0 0]A  <-- Still 1 phase.

andig, 2024-06-18:
@rudybroersma 

> Diese scheint nicht zu funktionieren da der Charger auf 1p bleibt. Warum auch immer. Bitte Alfen anfragen warum das scheitern kann/sollte.

You'll need to investigate with Alfen why the phase switching is no longer working. Seems it did in an older version?

rudybroersma, 2024-06-18:
> @rudybroersma
> 
> You'll need to investigate with Alfen why the phase switching is no longer working. Seems it did in an older version?

I'm new to EVCC since about a week or two. I do not have any experiences with older versions. I'm running 0.126.5
Phase switching *is* working. It works fine when using 'Solar'. 

My guess is that when I select 'Fast' and the previous phase setting of the charger was '1p' things go wrong. When EVCC starts to run in 'Fast mode'  it assumes the charger is in 3p state. It throws an error that there's a phase mismatch.

EVCC never recovers from this. Even when I select 'Solar'  again, it will never start 3p charging. I need to powercycle the Alfen charger to get things working again. After that switching to 1p and 3p using 'Solar mode'  works like a breeze. It's only the 'Fast mode'  that breaks things.

Today I noticed that EVCC suddenly thought 2 phases were active for a short while, while my car was charging on 1p. I haven't deep-dived into EVCC yet as I find all the .go files a little overwhelming for now, but this made me believe that EVCC does the phase check based on consumption per phase? eg. if a phase is using >10W or some arbitrary value it defines that phase as active? If for some reason there is some current flowing on an unused phase (eg. incorrect measurements) EVCC gets confused about the phase switching in the charger?

Whatever it is.. it isn't the Alfen that's at fault here. Phase switching works fine in Solar mode but goes haywire in Fast mode. The only difference between those modes is the EVCC internals.

andig, 2024-06-20:
> When EVCC starts to run in 'Fast mode' it assumes the charger is in 3p state. It throws an error that there's a phase mismatch.

Switching to FAST switches to 3p. Seems that doesn‘t (always) work. Why is unclear. You can verify in the log that switch is triggered. Only Alfen would be able to tell us why box doesn‘t follow.

andig, 2024-06-20:
> Whatever it is.. it isn't the Alfen that's at fault here. Phase switching works fine in Solar mode but goes haywire in Fast mode. The only difference between those modes is the EVCC internals.

@rudybroersma any proof for this claim? We switch to 3p on fast mode and have been doing so for ages. 

My best guess: Alfen is unreliable. In PV mode it will sooner of later like to go to 1p again and then restart. In Fast mode it stays on 3p.

I can only repeat: you‘ll need to open a ticket with Alfen.

rudybroersma, 2024-06-29:
Not really sure if it's proof, but here is what I noticed.. I made photos and screenshots.
This doesn't always happen, but sometimes EVCC WebUI shows 2 or 3 phases when only 1 is actually in use.

EVCC shows 2 phases:
[image]

And here is a bigger image with a couple of cropped photos.
First you see the 3 phases of my Modbus meter that's connected to the EV charger
Then a pic of the Tesla App showing it's charging at 1 phase, 23A.
And a pic of the Victron App showing there's little to no load on phases 2 and 3.

[image]

Yet.. EVCC shows the car is charging at 2 phases. On the photo above you also see a yellow exclamation mark showing an error: "Alfen: ignoring inconsistent phases: 1p < 2p observed active"

premultiply, 2024-06-30:
evcc detects phase usage on phase currents and phase voltages.

andig, 2024-06-30:
Instead of screenshots, look at /api/state, it will show the currents. Probably p2 shows a current above our detection threshold and hence we treat it as active. If that's the case that is actually a very good explanation for why evcc might not switch to 3p in cases it thinks that it already _is_ on 2p (or 3p). The question then really is- where does that p2 consumption come from when the charger really is at 1p???

rudybroersma, 2024-06-30:
Alright, will post when I have another occurrence

felsnerino, 2024-07-03:
But why use such a (complicated/unreliable) logic to derive how many phase are active with alfen? There is a modbus register for that... modbus.2.holdingRegisters.1.**41216**_Charge_phases

andig, 2024-07-03:
>> evcc detects phase usage on phase currents and phase voltages.

> But why use such a (complicated/unreliable) logic to derive how many phase are active with alfen? There is a modbus register for that... 41216 Charge_phases

Which is what we're doing (Reg 1215). Seems that value is sometimes wrong, i.e. Alfen lies. To verify we'll need a modbus trace log.

felsnerino, 2024-07-06:
I don't really understand. I have modbus log all values also the phase change, and the interesting part is, that the change from 3 -> 1 works. But the change back doesn't (I thought, I remember that in the past that worked already). But then it says switching back - it even thinks it is 3 phases, but on Modbus I still see "1"... so not sure where it gets the values from, when it says "active phases: 3"... 
Please Ignore this one spike - I tried to change once the phases through modbus directly to 3 which worked, I then set it back to 1. the one strange thing is, that the initial switch from 3 to 1 actually worked... 
[image]
DEBUG Log File: 
[evcc-20240705-102154-debug.log](https://github.com/user-attachments/files/16109136/evcc-20240705-102154-debug.log)

andig, 2024-07-06:
> I have modbus log all values also the phase change, and the interesting part is, that the change from 3 -> 1 works. But the change back doesn't (I thought, I remember that in the past that worked [...]

Could you extract the relevant part of the log? Include `trace` for Alfen so we can see the modbus.