evcc-io/evcc #13816: Entladestopp Batterie Alpha Ess Smile 5 / Modbus

AWR69, 2024-05-08:
GUI: Batterieeinstellung

Für einige Batterietypen wird ein Entladestopp unterstützt. Ist es möglich eine Alpha-Ess-Smile 5 zu unterstützen in einem zukünftigen Release. 

Anschlusstyp ist Modbus (rs485).

Problem: Schnelles entladen der Batterie, wenn nachts geladen werden muss.

Danke

andig, 2024-05-08:
Wenn Du Informationen hast, wie das bei diesem Gerät gehen soll?

AWR69, 2024-05-08:
Hallo, gute Frage… in der GUI des Systems oder der Website kann man einen Entladestopp fest einstellen. Also gibt es die Funktion bereits. 
Das PDF beschreibt nur die Modbus Definition https://alphaess.de/Public/Uploads/uploadfile/files/20230926/AlphaESS-Handbuch_SMILE_ModBus_RTU_TCP_V21.pdf

Es gibt eine API für das System:
https://github.com/alphaess-developer/alphacloud_open_api

Registrieren kann ich mich und bekomme einen API Key. 
Mehr konnte ich nicht herausfinden. 
Was genau müsste man wissen, dann versuche ich direkt bei AlphaESS Support anfragen.

premultiply, 2024-05-08:
Nichts dabei was auch nur annähernd dafür dienen könnte und zudem alles read-only.

Ich mache hier erstmal. Sollte sich an der Firmware oder Doku nochmal was ändern kann man nochmal hier weitermachen. Aber erstmal gibt es hier nichts zu tun.

AWR69, 2024-05-14:
Der Modbus hat auch beschreibbare Register:

https://alphaess.de/Public/Uploads/uploadfile/files/20230926/AlphaESS_Register_Parameter_List.pdf

LG, Andreas

andig, 2024-05-14:
…und hilft etwas davon?

AWR69, 2024-05-15:
Hallo,Seite 14 / Time Period Control könnte die benötigten Register steuern…Das ist das Menü in der GUI der App:Gruß / Regards,Andreassent from mobile device

AWR69, 2024-05-15:
Hallo,

Seite 14 / Time Period Control könnte die benötigten Register steuern…

Das ist das Menü in der GUI der App:
[image]
[image]

hier der Link zum Dokument:

https://alphaess.de/Public/Uploads/uploadfile/files/20230926/AlphaESS_Register_Parameter_List.pdf

Gruß / Regards,
Andreas

sent from mobile device

softcat, 2024-09-22:
I published a PR for that feature.

https://github.com/evcc-io/evcc/pull/16280

closed 2024-09-27 by premultiply

Nitrox912, 2024-10-07:
> I published a PR for that feature.
> 
> #16280

Hallo, ich verstehe die Funktionsweise deiner Erweiterung nicht ganz. Wozu wird mit den Zeit-Slots gearbeitet? Reicht es nicht das Entladen und die Ladefreigabe über den minSoC (Register 2128) zu realisieren? Ich habe das Ganze in Home Assistant umgesetzt und arbeite ausschließlich mit diesem Wert. 

MinSoC 100% = Netzladen (z. B. bei günstigen Tibber-Preisen), ohne den Akku zu entleeren.
MinSoC 10% = EV-Ladung inkl. Nutzung des Akkus (Ladestopp bei Erreichung des MinSoC bzw. auch darüber hinaus)

andig, 2024-10-07:
#16280 ist schon gemerged. Ich wäre allerdings dafür- wenn das funktioniert- den auf die deutlich einfachere `limitsoc` Variante umzubauen. @Nitrox912 magst Du PR dazu machen?

softcat, 2024-10-07:
Also bei meinen vorherigen Tests (allerdings mit einem anderen Softwarestand vor einigen Monaten) hat das Setzen von `minSoC` alleine sind zum Entladestopp geführt (ist damals auch lange im Herstellerforum diskutiert worden). 

Ich werde es jetzt neu mal neu testen. Falls es ausreicht mache ich einen neuen PR, wäre ja eleganter.

Nitrox912, 2024-10-07:
> #16280 ist schon gemerged. Ich wäre allerdings dafür- wenn das funktioniert- den auf die deutlich einfachere `limitsoc` Variante umzubauen. @Nitrox912 magst Du PR dazu machen?

Leider fehlt mir dazu die Ahnung den Code richtig anzupassen und mit GitHub umzugehen. Aber @softcat will sich das ja nochmals angucken.

@softcat Ich nutze die Funktion seit Anfang des Jahres. 

Meine Automation in HA sieht so aus. Vielleicht bringt es dich ja etwas weiter:

`#Helper for setting the SOCminhold
input_number:
  alpha_ess_helper_socminholding_value:
    name: "Setting SOC min holding"
    min: 5
    max: 100
    icon: mdi:battery-minus-variant
    unit_of_measurement: "%"
    mode: slider
    step: 1.0

#Automations
automation:
  - id: "Alpha_ESS_Setting_SOCminholding"
    alias: PV_Set_SOCminholding_Value
    description: "With slide the SOCminholding rate is set and send to System via modbus"
    trigger:
      - platform: state
        entity_id:
          input_number.alpha_ess_helper_socminholding_value
    condition: []
    action:
      - service: modbus.write_register
        data:
          hub: modbuspvsystem
          address: 2128
          slave: 85
          value: "{{ (states('input_number.alpha_ess_helper_socminholding_value') | int)}}"
    mode: single`

softcat, 2024-10-07:
Ich teste es mal bis die Sonne hier nicht mehr so scheint. 

Aber: der PR ist ja neben dem Entladestopp auch auf für die geplante Aufladung der Batterie zuständig. Dafür brauch man die Zeitslots, das hast du ja in Homeassistant so nicht umgesetzt.

Nitrox912, 2024-10-07:
> Ich teste es mal bis die Sonne hier nicht mehr so scheint.
> 
> Aber: der PR ist ja neben dem Entladestopp auch auf für die geplante Aufladung der Batterie zuständig. Dafür brauch man die Zeitslots, das hast du ja in Homeassistant so nicht umgesetzt.

Ja, da hast du absolut recht.

AWR69, 2024-10-07:
Hi, bei mir steht der SOC dauerhaft auf 13%. Ich kann ab Samstag auch testen, wenn Ihr möchtet. Gruß / Regards,Andreassent from mobile device

andig, 2024-10-07:
Das klang in https://github.com/evcc-io/evcc/issues/13816#issuecomment-2396333605 Aber noch anders („Netzladen“)?

Nitrox912, 2024-10-07:
> Das klang in [#13816 (comment)](https://github.com/evcc-io/evcc/issues/13816#issuecomment-2396333605) Aber noch anders („Netzladen“)?

Ich sprach von Netzladen in Bezug auf Strom vom Netz ins Auto. Nicht Strom vom Netz in den Speicher.

Bzgl. der Speicherladung aus dem Netz gibt es noch eine Funktion, die sich "USV-Reserve aktivieren" nennt. Damit habe ich bisher aber noch nicht experimentiert. So wie ich das verstehe, lässt sich damit der Speicher auf den eingestellten MinSoC aus dem Netz laden. Also unabhängig von irgendwelchen Ladezeiten.

[image]

softcat, 2024-10-07:
Die Option gibt es bei mir nicht (SMILE-B3)

[image]

AWR69, 2024-10-07:
Hi, USV reserve ist auch ein losgelöstes Feature. Gruß / Regards,Andreassent from mobile device

IMG_1068.PNG (view on web)

AWR69, 2024-10-07:
Hi, ich habe eben den SOC bei „ Entladen stoppt bei folgendem Ladezustand“ auf 99 eingestellt. Die Batterie war auf 100 und am entladen. Erst bei unterschreiten von 100 hat das Entladen gestoppt und die Verbraucher bedienten sich am Netzstrom. 
Zurück auf 13 eingestellt, beziehen die Verbraucher wieder Strom aus der Batterie. 

Ich hoffe das hilft weiter. 

[image]
[image]
[image]

softcat, 2024-10-07:
Ich hatte die Nutzungsmöglichkeit von Modbus Register `2128` (entspricht dem obigen Screenshot) ja auch bereits im PR erwähnt (https://github.com/evcc-io/evcc/pull/16280#issuecomment-2367536069)

Ich habe es jetzt nochmals getestet und ein paar Erkenntnisse mitgenommen:

*  Entladestopp ist darüber möglich, **aber**, zumindest bei mir, muss für dieses Verhalten exakt der aktuelle SoC eingetragen werden. Wenn der eingestellte minSoC größer ist als der SoC wird der Akku **geladen**.  Er kann also nicht auf feste Zielwerte wie 10% oder 100% eingestellt werden.

*  Ich war unsicher ob dieses mit einem evcc template umzusetzen ist. @andig hatte aber geschrieben (https://github.com/evcc-io/evcc/pull/16280#issuecomment-2367615675), dass dies möglich sei. Allerdings nur wenn darüber auch `charge` nutzbar wäre was ist damals fälschlicherweise verneint hatte. 

* Da ja scheinbar `charge` auch über dieses Register gesteuert werden kann wäre es eine Alternative ohne die Zeitsteuerung zu nutzen. Es hat allerdings den Nachteil, dass ggf. ein vom Nutzer selbst definierter Min-SOC überschrieben würde.

DIe Logik wäre also 

`normal`: `minSoC=10%` (Hausakku wird genutzt, Normalzustand)
`hold`: `minSoc=Aktueller SoC` (Entladestopp bei Ladung Auto)
`charge`: `minSoc=maxSoc` (Geplante Ladung des Hausakku, z.B. bei dynamischen Preisen)

ohne Nutzung der Zeitsteuerung.

Wenn ich da jetzt keinen Denkfehler habe und mir @andig verraten würde wie ich im template auf den aktuellen SoC zugreifen kann würde ich es testweise umsetzen und bei Erfolg einen PR erstellen.

AWR69, 2024-10-07:
Hallo,

normal: minSoC=10%(Hausakku wird genutzt, Normalzustand)
—> das sollte frei einstellbar sein. 

hold: minSoc=Aktueller SoC (Entladestopp bei Ladung Auto)
—> bei mir minSoC=Aktueller SoC-1%. Sieht aus als müsste dieser Wert von SOC hoch nach tief erreicht werden, um wirksam zu werden. 

Firmware bei meiner Anlage:

Gruß,
Andreas

sent from mobile device

> Am 07.10.2024 um 21:33 schrieb SoftCat ***@***.***>:
> 
> normal: minSoC=10% (Hausakku wird genutzt, Normalzustand)
> hold: minSoc=Aktueller SoC (Entladestopp bei Ladung Auto)

softcat, 2024-10-07:
> bei mir minSoC=Aktueller SoC-1%

Hier bei mir hört die Entladung sofort bei minSoC=aktueller SoC auf. Scheint also Gerät / Firmware abhängig zu sein. 

Schon die bestehende Implementierung ausprobiert? (der aktuellen evcc Version).

andig, 2024-10-07:
> Wenn ich da jetzt keinen Denkfehler habe und mir @andig verraten würde wie ich im template auf den aktuellen SoC zugreifen kann würde ich es testweise umsetzen und bei Erfolg einen PR erstellen

Du brauchst limitsoc statt batterymode. Schau mal in die anderen Templates. Einfacher geht es nicht…

Nitrox912, 2024-10-07:
@softcat Ich habe einen Hi10 Speicher. Deiner wird eine Generation älter sein. 

Ich setzte grundsätzlich meinen MinSoC auf 100% um den Speicher zu sperren. Damit stelle ich den Netzbezug aller Verbraucher samt Wallbox sicher. 

Aktiviere ich die USV-Reserve und mein SoC ist unter MinSoC, kann ich nach meinem Test auch den Akku zielgerichtet laden. Hier müsste also bei der Implementierung zwischen den Modellgenerationen unterschieden werden.

softcat, 2024-10-08:
@Nitrox912 Dann gibt es definitiv unterschiedliches Verhalten zwischen den Geräten. minSoC 100% ist für `hold` bei mir nicht nutzbar da es für eine Ladung des Speichers führt. 

Daher meine Frage: wie verhält sich die aktuelle Implementierung (evcc 0.130.13) bei euch (@Nitrox912, @AWR69) ?

Wenn es funktioniert sehe ich den Vorteil nicht die Ansteuerung umzustellen. Um dann gleichzeitig die Notwendigkeit zu haben verschiedene Geräte und schlimmstenfalls verschiedene Firmwareversionen unterscheiden und implementieren zu müssen.

AWR69, 2024-10-08:
Hi, 

die aktuelle Implementierung (evcc 0.130.13) zeigt keine Wirkung. Nächtliches geplantes Laden des E-Autos wird aus der Batterie+Netz bedient. 

Gruß / Regards,
Andreas

sent from mobile device

> Am 08.10.2024 um 06:57 schrieb SoftCat ***@***.***>:
> 
> die aktuelle Implementierung (evcc 0.130.13)

andig, 2024-10-08:
@AWR69  Die Voraussetzungen gem. Doku hast Du erfüllt?

AWR69, 2024-10-08:
ja Gruß / Regards,Andreassent from mobile device

Nitrox912, 2024-10-08:
Ich konnte es bisher nicht testen. Die Modbus-Schnittstelle des Speichers lässt nur eine Verbindung zu und die blockiert derzeit Home Assistant. Bräuchte also noch etwas Zeit, mich darum zu kümmern.

Nitrox912, 2024-10-08:
Kurzer Hinweis: Bzgl. des Schreibens in die Register hatten wir gerade eine Diskussion im Alpha ESS-Forum, da die Daten auf einem Flashspeicher abgelegt werden und dieser wohl relativ schnell altert. Hierzu gibt es Seitens des Herstellers nun Restriktionen:
https://github.com/alphaess-developer/alphacloud_open_api/issues/54

Die betrifft wohlmöglich alle Register, die unter "Time period control" fallen.

andig, 2024-10-08:
> Recently, it was found that the following two interfaces are set very frequently, which seriously affects the FLASH flash data of the device. After internal confirmation by the team, the startup limit is set once a day, and it can only be set 10 min after it is set.
https://openapi.alphaess.com/api/updateDisChargeConfigInfo
https://openapi.alphaess.com/api/updateChargeConfigInfo

Trifft das denn hier zu? Und wenn: was schlagen sie stattdessen vor?