evcc-io/evcc #13819: Ladesteuerung für Hausakku mit Tibber

winfotiker, 2024-05-08:
Wenn das Fahrzeug schon mit günstigem Strom von Tibber geladen werden kann wäre es toll wenn auch Akkus (zum Beispiel von e3dc) über Tibber geladen werden können wenn der Strom besonders günstig ist und keine Sonne hinreichend zur Verfügung steht.

Funktion soll über die Oberfläche aktiviert oder deaktiviert werden können. (Um diese beispielsweise nur im Winter zu nutzen)

premultiply, 2024-05-08:
Mindestens in Deutschland spricht momentan noch das EEG eindeutig gegen eine solche Variante, da dann s.g. Graustrom im Speicher landen könnte.

winfotiker, 2024-05-09:
Nee, das wurde doch jetzt mit dem Solarpaket 1 geändert. Damit darf der
eigene Speicher auch aus dem Netz geladen werden.

winfotiker, 2024-05-09:
Siehe Seite 6 letzter Absatz 😃

https://www.bmwk.de/Redaktion/DE/Downloads/S-T/solarpaket-im-ueberblick.pdf?__blob=publicationFile&v=8

closed 2024-05-09 by andig

winfotiker, 2024-05-09:
Na Guck. Das sieht ja gut aus. Muss ich nur noch den Anbieter wechseln wenn der Vertrag ausläuft ☺️
Wäre aber vielleicht trotzdem an der Zeit das in den produktiven build zu übernehmen wo es ja jetzt offiziell erlaubt ist, oder? 
😇