evcc-io/evcc #13825: Begrenzung der Wallbox-Ladeleistung im PV-Modus

marsu81, 2024-05-09:
Hallo zusammen,

ersteinmal möchte ich mich herzlich für dieses großartige Projekt und all denen bedanken, die sich hierfür einbringen. Ich habe EVCC über Homeassistant in unser Energiemanagement integriert und dies funktioniert ganz großartig! :)

Bzgl. der Erweiterung der möglichen Funktionalitäten hätte ich folgende Anfrage:
Im PV-Modus ist es aktuell so, dass man eine Priorisierung zwischen E-Auto laden und Hausbatterie nur "sequentiell" über das Ladelevel der Hausbatterie steuern kann (über die Einstellungen unter "Hausbatterie" => Vorrang Haus oder E-Auto).
Jetzt wäre es natürlich ganz wunderbar, wenn die Möglichkeit bestünde, dass im PV-Modus eine Begrenzung der maximalen Ladeleistung von x% oder xkW auf Basis des aktuell verfügbaren PV-Überschusses möglich wäre. Dies hätte den Vorteil, dass die Hausbatterie und das E-Auto parallel geladen werden könnten und damit am Abend der Hausakku nicht leer ist bzw. einfach nicht geladen wurde, weil die ganze PV-Leistung in die Wallbox geflossen ist. Aktuell lässt sich das nur dadurch regeln, indem man die Ladegrenze für das Auto limitiert (sodass dieses bspw. um 14Uhr erreicht wird) bzw. die maximale Ladeleistung in den Einstellungen für die Wallbox reduziert (und damit ein höherer PV-Überschuss in die Hausbatterie geladen würde). Beide Varianten sind jedoch "statisch" und orientieren sich nicht am dynamisch verfügbaren PV-Überschuss.

Besteht die Möglichkeit, eine solche Funktion zu implementieren bzw. ist derartiges evtl. bereits sogar für die Zukunft geplant?

Besten Dank & viele Grüße!

VolkerK62, 2024-05-09:
schau mal hier https://github.com/evcc-io/evcc/discussions/13500
hört sich sehr ähnlich an

andig, 2024-05-09:
Dann bitte dort weiter :)

marsu81, 2024-05-11:
Ich verstehe das so, dass man die Batterieeinstellungen in evcc auch von "extern" verstellen kann (also nicht nur über die UI), der Algorithmus dazu für die zu übergebenden Werte aber in HA integriert ist, die dann per mqtt an evcc gesendet werden. Das hört sich sehr spannend an, damit muss ich mich genauer beschäftigen. Ich lese aktuell bereits die evcc-parameter via mqqt in HA zurück (z.B. Wallbox-Ladeleistung, SoC, ...), anders herum hab ich das allerdings noch nie versucht / mir war auch nicht klar, dass das möglich ist. Besten Dank für den Link, ich werde mir das anschauen! :)