evcc-io/evcc #13828: Ladeplan startet immer sofort

kvray, 2024-05-09:
### Describe the bug

Hallo liebes Team,

erstmal danke: Tolle Software!

Ich habe seit Anfang an (März 2024) das Problem, dass bei einer Ladeplanung immer sofort angefangen wird zu laden, obwohl die Ladezeit richtig berechnet wird. Egal ob in 1P, 3P und  unabhängig vom Ladezustand des PV-Akkus, falls das relevant ist.
Mache ich da irgendwas falsch? Hier nur ein Beispiel:
[image]

### Steps to reproduce

1. Ladeplanung erstellen
2. Ladung startet sofort, statt sinnvollerweise zu einem späteren Zeitpunkt

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXX

interval: 10s # control cycle interval

sponsortoken: XXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.2.60  
  port: 502  
  minsoc: 20  
  maxsoc: 100  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.2.60  
  port: 502  
  minsoc: 20  
  maxsoc: 100  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.2.60  
  port: 502  
  minsoc: 20  
  maxsoc: 100  
  name: battery3

vehicles:
- name: my_car1
  type: template
  template: bmw
  title: BMW iX1 # Wird in der Benutzeroberfläche angezeigt (optional)
  user: XXX # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: XXX # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
  vin: XXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
  capacity: 64.7 # Akkukapazität in kWh (optional) 
  identifiers:
  - 050a92b8

chargers:
- type: template
  template: keba-modbus 
  id: 255  
  host: 192.168.0.50  
  port: 502  
  modbus: tcpip  
  name: wallbox4

loadpoints:
- title: Parkplatz
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 32

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.345 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: XXX # For post codes with leading zeros, enclose in ""
```

### Log details

```text
[site  ] DEBUG 2024/05/09 10:37:54 battery power: -163W
[site  ] DEBUG 2024/05/09 10:37:54 grid meter: 63W
[site  ] DEBUG 2024/05/09 10:37:54 site power: 0W
[lp-1  ] DEBUG 2024/05/09 10:37:54 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:37:54 charge total import: 335.815kWh
[lp-1  ] DEBUG 2024/05/09 10:37:54 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:37:54 soc estimated: 72.13% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:37:54 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:37:54 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:37:54 pv charge current: 28.1A = 28.1A + -0A (0W @ 1p)
[site  ] DEBUG 2024/05/09 10:38:04 ----
[lp-1  ] DEBUG 2024/05/09 10:38:04 charge power: 5934W
[lp-1  ] DEBUG 2024/05/09 10:38:04 charge currents: [26.4 0.004 0]A
[site  ] DEBUG 2024/05/09 10:38:04 pv power: 6901W
[site  ] DEBUG 2024/05/09 10:38:04 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:04 battery power: -28W
[site  ] DEBUG 2024/05/09 10:38:04 grid meter: 147W
[site  ] DEBUG 2024/05/09 10:38:04 site power: 219W
[lp-1  ] DEBUG 2024/05/09 10:38:04 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:04 charge total import: 335.832kWh
[lp-1  ] DEBUG 2024/05/09 10:38:04 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:04 soc estimated: 72.16% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:04 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:04 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:04 pv charge current: 27.2A = 28.1A + -0.952A (219W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:04 max charge current: 27.2A
[site  ] DEBUG 2024/05/09 10:38:14 ----
[lp-1  ] DEBUG 2024/05/09 10:38:14 charge power: 5747W
[lp-1  ] DEBUG 2024/05/09 10:38:14 charge currents: [25.6 0.003 0]A
[site  ] DEBUG 2024/05/09 10:38:14 pv power: 6912W
[site  ] DEBUG 2024/05/09 10:38:14 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:14 battery power: 62W
[site  ] DEBUG 2024/05/09 10:38:14 grid meter: -177W
[site  ] DEBUG 2024/05/09 10:38:14 site power: -15W
[lp-1  ] DEBUG 2024/05/09 10:38:14 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:14 charge total import: 335.848kWh
[lp-1  ] DEBUG 2024/05/09 10:38:14 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:14 soc estimated: 72.18% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:14 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:14 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:14 pv charge current: 27.2A = 27.2A + 0.0652A (-15W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:14 max charge current: 27.2A
[site  ] DEBUG 2024/05/09 10:38:24 ----
[lp-1  ] DEBUG 2024/05/09 10:38:24 charge power: 5760W
[lp-1  ] DEBUG 2024/05/09 10:38:24 charge currents: [25.6 0.003 0]A
[site  ] DEBUG 2024/05/09 10:38:24 pv power: 6915W
[site  ] DEBUG 2024/05/09 10:38:24 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:24 battery power: -79W
[site  ] DEBUG 2024/05/09 10:38:24 grid meter: -69W
[site  ] DEBUG 2024/05/09 10:38:24 site power: -48W
[lp-1  ] DEBUG 2024/05/09 10:38:24 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:24 charge total import: 335.864kWh
[lp-1  ] DEBUG 2024/05/09 10:38:24 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:24 soc estimated: 72.20% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:24 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:24 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:24 pv charge current: 27.4A = 27.2A + 0.209A (-48W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:24 max charge current: 27.4A
[site  ] DEBUG 2024/05/09 10:38:33 ----
[lp-1  ] DEBUG 2024/05/09 10:38:34 charge power: 5799W
[lp-1  ] DEBUG 2024/05/09 10:38:34 charge currents: [25.8 0.003 0]A
[site  ] DEBUG 2024/05/09 10:38:34 pv power: 6922W
[site  ] DEBUG 2024/05/09 10:38:34 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:34 battery power: -135W
[site  ] DEBUG 2024/05/09 10:38:34 grid meter: 9W
[site  ] DEBUG 2024/05/09 10:38:34 site power: -26W
[lp-1  ] DEBUG 2024/05/09 10:38:34 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:34 charge total import: 335.880kWh
[lp-1  ] DEBUG 2024/05/09 10:38:34 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:34 soc estimated: 72.22% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:34 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:34 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:34 pv charge current: 27.5A = 27.4A + 0.113A (-26W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:34 max charge current: 27.5A
[site  ] DEBUG 2024/05/09 10:38:44 ----
[lp-1  ] DEBUG 2024/05/09 10:38:44 charge power: 5837W
[lp-1  ] DEBUG 2024/05/09 10:38:44 charge currents: [25.9 0.004 0]A
[site  ] DEBUG 2024/05/09 10:38:44 pv power: 6926W
[site  ] DEBUG 2024/05/09 10:38:44 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:44 battery power: -90W
[site  ] DEBUG 2024/05/09 10:38:44 grid meter: -33W
[site  ] DEBUG 2024/05/09 10:38:44 site power: -23W
[lp-1  ] DEBUG 2024/05/09 10:38:44 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:44 charge total import: 335.896kWh
[lp-1  ] DEBUG 2024/05/09 10:38:44 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:44 soc estimated: 72.25% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:44 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:44 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:44 pv charge current: 27.6A = 27.5A + 0.1A (-23W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:44 max charge current: 27.6A
[site  ] DEBUG 2024/05/09 10:38:54 ----
[lp-1  ] DEBUG 2024/05/09 10:38:54 charge power: 5828W
[lp-1  ] DEBUG 2024/05/09 10:38:54 charge currents: [26 0.003 0]A
[site  ] DEBUG 2024/05/09 10:38:54 pv power: 6931W
[site  ] DEBUG 2024/05/09 10:38:54 battery soc: 40%
[site  ] DEBUG 2024/05/09 10:38:54 battery power: -50W
[site  ] DEBUG 2024/05/09 10:38:54 grid meter: -30W
[site  ] DEBUG 2024/05/09 10:38:54 site power: 20W
[lp-1  ] DEBUG 2024/05/09 10:38:54 detected active phases: 1p
[lp-1  ] DEBUG 2024/05/09 10:38:54 charge total import: 335.912kWh
[lp-1  ] DEBUG 2024/05/09 10:38:54 charger status: C
[lp-1  ] DEBUG 2024/05/09 10:38:54 soc estimated: 72.27% (vehicle: 72.00%)
[lp-1  ] DEBUG 2024/05/09 10:38:54 vehicle soc: 72%
[lp-1  ] DEBUG 2024/05/09 10:38:54 vehicle range: 254km
[lp-1  ] DEBUG 2024/05/09 10:38:54 pv charge current: 27.6A = 27.6A + -0.087A (20W @ 1p)
[lp-1  ] DEBUG 2024/05/09 10:38:54 max charge current: 27.6A
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.126.2

andig, 2024-05-09:
Ich sehe da keine Ladeplanung im Log?

kvray, 2024-05-09:
Ja, zu dem Zeitpunkt war auch keine aktiv. Die Vorschau zeigt ja schon, dass was nicht stimmt. Sobald ich heute wieder Zuhause bin, aktiviere ich eine Planung und lade den log nochmal hoch.

kvray, 2024-05-09:
Hier ein Beispiel mit aktivierter Planung.
Zuletzt hatte ich den Plan immer nach 18Uhr gemacht und er fängt dann direkt mit dem laden an.
[image]
`[site  ] DEBUG 2024/05/09 13:56:34 ----
[lp-1  ] DEBUG 2024/05/09 13:56:34 charge power: 0W
[lp-1  ] DEBUG 2024/05/09 13:56:34 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/09 13:56:34 pv power: 11374W
[site  ] DEBUG 2024/05/09 13:56:34 battery soc: 96%
[site  ] DEBUG 2024/05/09 13:56:34 battery power: -3564W
[site  ] DEBUG 2024/05/09 13:56:34 grid meter: -4551W
[site  ] DEBUG 2024/05/09 13:56:34 site power: -8015W
[lp-1  ] DEBUG 2024/05/09 13:56:34 charge total import: 342.852kWh
[lp-1  ] DEBUG 2024/05/09 13:56:34 charger status: B
[lp-1  ] DEBUG 2024/05/09 13:56:34 plan: charge 52m45s between 2024-05-09 18:00:00 +0200 CEST until 2024-05-10 05:00:00 +0200 CEST (power: 7360W, avg cost: 108.000)
[site  ] DEBUG 2024/05/09 13:56:43 ----
[lp-1  ] DEBUG 2024/05/09 13:56:44 charge power: 0W
[lp-1  ] DEBUG 2024/05/09 13:56:44 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/09 13:56:44 pv power: 11364W
[site  ] DEBUG 2024/05/09 13:56:44 battery soc: 96%
[site  ] DEBUG 2024/05/09 13:56:44 battery power: -3682W
[site  ] DEBUG 2024/05/09 13:56:44 grid meter: -4398W
[site  ] DEBUG 2024/05/09 13:56:44 site power: -7980W
[lp-1  ] DEBUG 2024/05/09 13:56:44 charge total import: 342.852kWh
[lp-1  ] DEBUG 2024/05/09 13:56:44 charger status: B
[lp-1  ] DEBUG 2024/05/09 13:56:44 plan: charge 52m45s between 2024-05-09 18:00:00 +0200 CEST until 2024-05-10 05:00:00 +0200 CEST (power: 7360W, avg cost: 108.000)
[site  ] DEBUG 2024/05/09 13:56:53 ----
[lp-1  ] DEBUG 2024/05/09 13:56:54 charge power: 0W
[lp-1  ] DEBUG 2024/05/09 13:56:54 charge currents: [0 0 0]A
[site  ] DEBUG 2024/05/09 13:56:54 pv power: 11366W
[site  ] DEBUG 2024/05/09 13:56:54 battery soc: 96%
[site  ] DEBUG 2024/05/09 13:56:54 battery power: -3600W
[site  ] DEBUG 2024/05/09 13:56:54 grid meter: -4434W
[site  ] DEBUG 2024/05/09 13:56:54 site power: -7934W
[lp-1  ] DEBUG 2024/05/09 13:56:54 charge total import: 342.852kWh
[lp-1  ] DEBUG 2024/05/09 13:56:54 charger status: B
[lp-1  ] DEBUG 2024/05/09 13:56:54 plan: charge 52m45s between 2024-05-09 18:00:00 +0200 CEST until 2024-05-10 05:00:00 +0200 CEST (power: 7360W, avg cost: 108.000)`

kvray, 2024-05-09:
Ich habe das Problem jetzt selbst "gelöst". Es hängt mit dem Grünstromindex zusammen. Obwohl ich "keine" bei der Grenze eingestellt habe und es eigentlich bei der Ladeplanung nicht berücksichtigt werden sollte (laut Text), passiert da irgendwas.
Grünstromindex ist raus, nun passt der Startzeitpunkt.

andig, 2024-05-09:
> Die Vorschau zeigt ja schon, dass was nicht stimmt. 

Ich verstehe nur Bahnhof. Die Vorschau zeigt, dass 16:00 Uhr morgen geladen werden soll. Das Logfile zeigt gar nichts. Da gelöst mache ich hier zu...

closed 2024-05-09 by andig

VolkerK62, 2024-05-09:
>Ich habe das Problem jetzt selbst "gelöst". Es hängt mit dem Grünstromindex zusammen. Obwohl ich "keine" bei der Grenze eingestellt habe und es eigentlich bei der Ladeplanung nicht berücksichtigt werden sollte (laut Text), passiert da irgendwas.
Grünstromindex ist raus, nun passt der Startzeitpunkt.

Die Grenze bei "Sauberes Netzladen" hat nichts mit der Ladeplanung zu tun.
Der KOmmentar bei Ladeplanung sagt lediglich, dass eine gesetzte Grenze ignoriert (und somit auch überschritten) wird, um den Plan zu erfüllen.

@naltatis   diese "falsche" Interpretation aufgrund des Hinweises bei der Ladeplanung, kam jetzt schon öfter vor.

naltatis, 2024-05-09:
@kvray ja, in deinem Screenshot ist ein Zustand den es eigentlich nicht geben sollte.

[image]

Kannst du das noch mal nachstellen, damit wir das Problem beheben können? Was hilft wäre ein Auszug von `http://evcc.local:7070/api/state`. Falls Grünstromindex wieder reinnehmen zum vorherigen Fehler führt, wäre es super, wenn du das noch mal heraussuchen könntest. Sieht so aus als ob die Preis-/CO2-Grenze einen unerwarteten Wert hat.

kvray, 2024-05-10:
Klar, kein Problem. Ich habe es wieder rein genommen und es zeigt sich das selbe Bild.
Hier sieht man auch, dass er wieder sofort angefangen hat zu laden, obwohl der Plan mehr als 24h in der Zukunft ist.

Wenn Du noch etwas brauchst, melde Dich gerne.

PS: Danke, dass Du das ernst nimmst ;) Auch wenn ich mich vielleicht nicht ganz klar ausgedrückt habe...

[image]

> [lp-1  ] DEBUG 2024/05/10 14:57:26 pv charge current: 8.06A = 9.19A + -1.13A (259W @ 1p)
> [lp-1  ] DEBUG 2024/05/10 14:57:26 max charge current: 8.06A
> [site  ] DEBUG 2024/05/10 14:57:36 ----
> [lp-1  ] DEBUG 2024/05/10 14:57:36 charge power: 1693W
> [lp-1  ] DEBUG 2024/05/10 14:57:36 charge currents: [7.44 0 0]A
> [site  ] DEBUG 2024/05/10 14:57:36 pv power: 2475W
> [site  ] DEBUG 2024/05/10 14:57:36 battery soc: 80%
> [site  ] DEBUG 2024/05/10 14:57:36 battery power: 244W
> [site  ] DEBUG 2024/05/10 14:57:36 grid meter: -324W
> [site  ] DEBUG 2024/05/10 14:57:36 site power: 20W
> [lp-1  ] DEBUG 2024/05/10 14:57:36 detected active phases: 1p
> [lp-1  ] DEBUG 2024/05/10 14:57:36 charge total import: 371.212kWh
> [lp-1  ] DEBUG 2024/05/10 14:57:36 charger status: C
> [lp-1  ] DEBUG 2024/05/10 14:57:36 soc estimated: 79.03% (vehicle: 79.00%)
> [lp-1  ] DEBUG 2024/05/10 14:57:36 vehicle soc: 79%
> [lp-1  ] DEBUG 2024/05/10 14:57:36 vehicle range: 290km
> [lp-1  ] DEBUG 2024/05/10 14:57:36 pv charge current: 7.98A = 8.06A + -0.087A (20W @ 1p)
> [lp-1  ] DEBUG 2024/05/10 14:57:36 max charge current: 7.98A
> [site  ] DEBUG 2024/05/10 14:57:43 set my_car1 plan soc: 100 @ 2024-05-11 05:00:00 +0200 CEST
> [site  ] DEBUG 2024/05/10 14:57:46 ----
> [lp-1  ] DEBUG 2024/05/10 14:57:46 charge power: 1691W
> [lp-1  ] DEBUG 2024/05/10 14:57:46 charge currents: [7.45 0 0]A
> [site  ] DEBUG 2024/05/10 14:57:46 pv power: 2522W
> [site  ] DEBUG 2024/05/10 14:57:46 battery soc: 80%
> [site  ] DEBUG 2024/05/10 14:57:46 battery power: -85W
> [site  ] DEBUG 2024/05/10 14:57:46 grid meter: -105W
> [site  ] DEBUG 2024/05/10 14:57:46 site power: -90W
> [lp-1  ] DEBUG 2024/05/10 14:57:46 detected active phases: 1p
> [lp-1  ] DEBUG 2024/05/10 14:57:46 charge total import: 371.217kWh
> [lp-1  ] DEBUG 2024/05/10 14:57:46 charger status: C
> [lp-1  ] DEBUG 2024/05/10 14:57:46 soc estimated: 79.03% (vehicle: 79.00%)
> [lp-1  ] DEBUG 2024/05/10 14:57:46 vehicle soc: 79%
> [lp-1  ] DEBUG 2024/05/10 14:57:46 vehicle range: 290km
> [lp-1  ] DEBUG 2024/05/10 14:57:46 plan: charge 2h31m48s between 2024-05-10 15:00:00 +0200 CEST until 2024-05-11 05:00:00 +0200 CEST (power: 7360W, avg cost: 110.466)
> [lp-1  ] DEBUG 2024/05/10 14:57:46 pv charge current: 8.37A = 7.98A + 0.391A (-90W @ 1p)
> [lp-1  ] DEBUG 2024/05/10 14:57:46 max charge current: 8.37A
> [site  ] DEBUG 2024/05/10 14:57:56 ----
> [lp-1  ] DEBUG 2024/05/10 14:57:56 charge power: 1777W
> [lp-1  ] DEBUG 2024/05/10 14:57:56 charge currents: [7.76 0 0]A
> [site  ] DEBUG 2024/05/10 14:57:56 pv power: 2586W
> [site  ] DEBUG 2024/05/10 14:57:56 battery soc: 80%
> [site  ] DEBUG 2024/05/10 14:57:56 battery power: -170W
> [site  ] DEBUG 2024/05/10 14:57:56 grid meter: -18W
> [site  ] DEBUG 2024/05/10 14:57:56 site power: -88W
> [lp-1  ] DEBUG 2024/05/10 14:57:56 detected active phases: 1p
> [lp-1  ] DEBUG 2024/05/10 14:57:56 charge total import: 371.221kWh
> [lp-1  ] DEBUG 2024/05/10 14:57:56 charger status: C
> [lp-1  ] DEBUG 2024/05/10 14:57:56 soc estimated: 79.04% (vehicle: 79.00%)
> [lp-1  ] DEBUG 2024/05/10 14:57:56 vehicle soc: 79%
> [lp-1  ] DEBUG 2024/05/10 14:57:56 vehicle range: 290km
> [lp-1  ] DEBUG 2024/05/10 14:57:56 plan: charge 2h31m46s between 2024-05-10 15:00:00 +0200 CEST until 2024-05-11 05:00:00 +0200 CEST (power: 7360W, avg cost: 110.465)
> [lp-1  ] DEBUG 2024/05/10 14:57:56 pv charge current: 8.75A = 8.37A + 0.383A (-88W @ 1p)
> [lp-1  ] DEBUG 2024/05/10 14:57:56 max charge current: 8.75A
> [site  ] DEBUG 2024/05/10 14:57:56 set my_car1 plan soc: 100 @ 2024-05-11 06:00:00 +0200 CEST
> [site  ] DEBUG 2024/05/10 14:58:06 ----
> [lp-1  ] DEBUG 2024/05/10 14:58:06 charge power: 1856W
> [lp-1  ] DEBUG 2024/05/10 14:58:06 charge currents: [8.13 0 0]A
> [site  ] DEBUG 2024/05/10 14:58:06 pv power: 2721W
> [site  ] DEBUG 2024/05/10 14:58:06 battery soc: 80%
> [site  ] DEBUG 2024/05/10 14:58:06 battery power: -51W
> [site  ] DEBUG 2024/05/10 14:58:06 grid meter: -222W
[... 377 lines cut ...]
> [site  ] DEBUG 2024/05/10 15:01:56 site power: 5589W
> [lp-1  ] DEBUG 2024/05/10 15:01:56 detected active phases: 1p
> [lp-1  ] DEBUG 2024/05/10 15:01:56 charge total import: 371.544kWh
> [lp-1  ] DEBUG 2024/05/10 15:01:56 charger status: C
> [lp-1  ] DEBUG 2024/05/10 15:01:56 soc estimated: 80.26% (vehicle: 80.00%)
> [lp-1  ] DEBUG 2024/05/10 15:01:56 vehicle soc: 80%
> [lp-1  ] DEBUG 2024/05/10 15:01:56 vehicle range: 290km
> [lp-1  ] DEBUG 2024/05/10 15:01:56 plan: charge 2h24m38s between 2024-05-10 15:01:57 +0200 CEST until 2024-05-11 16:00:00 +0200 CEST (power: 7360W, avg cost: 110.233)
> [site  ] DEBUG 2024/05/10 15:02:06 ----
> [lp-1  ] DEBUG 2024/05/10 15:02:06 charge power: 6880W
> [lp-1  ] DEBUG 2024/05/10 15:02:06 charge currents: [30 0.005 0.001]A
> [site  ] DEBUG 2024/05/10 15:02:06 pv power: 3836W
> [site  ] DEBUG 2024/05/10 15:02:06 battery soc: 80%
> [site  ] DEBUG 2024/05/10 15:02:06 battery power: 4291W
> [site  ] DEBUG 2024/05/10 15:02:06 grid meter: -138W
> [site  ] DEBUG 2024/05/10 15:02:06 site power: 4253W
> [lp-1  ] DEBUG 2024/05/10 15:02:06 detected active phases: 1p
[... cut ...] ge":89.13154112340055},"total":{"avgCo2":24.649183380704315,"avgPrice":0.10880141602298848,"chargedKWh":370.8493000000002,"solarPercentage":89.13154112340055}},"tariffCo2":113,"tariffCo2Home":0,"tariffCo2Loadpoints":0.19705911075768134,"tariffFeedIn":0.08,"tariffGrid":0.345,"tariffPriceHome":0.08,"tariffPriceLoadpoints":0.0804621297730158,"vehicles":{"my_car1":{"title":"BMW iX1","icon":"car","capacity":64.7,"limitSoc":80,"plans":[{"soc":100,"time":"2024-05-11T14:00:00Z"}]}},"version":"0.126.2"}}

naltatis, 2024-05-10:
Wie sieht der Einstellungsdialog am Ladepunkt aus? Welche CO2 Grenze zeigt er dir da an?
Wie ändert sich das Verhalten, wenn du die Grenze auf "keine", "kleiner Wert" oder "sehr großer Wert" setzt?

kvray, 2024-05-10:
Es stand auf <=0. Im Ladeplan NaN.
Änderung auf "keine". Im Ladeplan NaN.
Änderung auf "<= 350". Im Ladeplan 350g.
Wieder Änderung auf "keine". Anzeige Co2 im Ladeplan ist komplett weg -> Wahrscheinlich richtig.

Ab jetzt kann ich alles einstellen was ich will und es wird im Ladeplan richtig angezeigt.

Neustart des Dockers mit Grenze <= 5 führt dazu, dass es immer noch funktioniert.
Neustart des Dockers mit Grenze "keine" führt dazu, dass erstmal wieder NaN da steht, bis ich wieder einen hohen Wert einstelle (gleiches Problem wie oben).

Ergebnis: Nach hin und her stellen wird es richtig angezeigt, aber trotzdem lädt er weiterhin direkt los, solange ich den Grünstromindex in der yaml habe. Mache ich ihn raus, funktioniert die Ladeplanung.

naltatis, 2024-05-10:
Ok, wir scheinen hier ein Problem beim Publishen des Wertes zu haben.

Aber zum Laden: Wie sieht denn das CO2 Diagramm in den Einstellungen des Ladepunkts aus?

kvray, 2024-05-10:
Das sieht so aus:

[image]

kvray, 2024-05-10:
Zu Punkt 1.: Ehrlich gesagt nutze ich das gar nicht. Ich hatte es bei der Ersteinrichtung nur aus Neugierde aufgenommen. Es soll nur Überschuss und nach Zeitplan geladen werden, weshalb es für mich kein Problem ist, dass zu entfernen. Ich helfe aber dennoch gerne weiter.
2.: Ja, Überschussladen funktioniert wunderbar.
3.: Genau. Ohne Grünstromindex funktioniert alles einwandfrei.
4.: Genau. Das variiert aber, wenn ich z.B. einige Tage in der Zukunft plane. Dann lädt er mal sofort und mal erst am Folgetag. Es sind aber immer irgendwie mindestens 24 Stunden vorher. Vielleicht Zufall, wenn er die "sauberen Zeiten" berücksichtigt.

Okay. Co2 Grenze steht auf "keine" bzw. springt dann auf <=0 um, wenn ich die Einstellungen erneut öffne. Ladeplan siehe Screenshot. Nun verstehe ich was Du meinst, er will jetzt um 19Uhr starten. Ist das bei "keine Co2 Grenze" richtig? Hier habe ich gerade einen Knoten im Kopf.

[image]
 
[image]

naltatis, 2024-05-10:
> Ist das bei "keine Co2 Grenze" richtig?

Ja, die CO2 Grenze ist außer Kraft gesetzt sobald du eine Planung aktiv hast. Hast du keine, muss auch nicht außer Kraft gesetzt werden -> keine Warnmeldung

> Genau. Das variiert aber, wenn ich z.B. einige Tage in der Zukunft plane.

Das klingt alles aber erstmal erwarnungskonform. Der Planner sucht sich die Zeiten mit den günstigsten/saubersten Netzstunden raus und lädt dann in den den entsprechenden Stunden. Das kann durchaus auch einige Stunden vor der Abfahrt sein. Hast du keine CO2-Daten oder zeitabhängigen Preistarife wird die Ladung erst passend zur Abfahrt gestartet. Das Verhalten ist auch hier beschrieben https://docs.evcc.io/docs/features/plans

kvray, 2024-05-10:
Bedeutet konkret: Wenn ich Co2 nicht berücksichtigen möchte muss es aus der yaml raus? Ich kann nicht "keine" wählen, damit sauberes Netzladen ignoriert wird?

Das ist wäre für mich in Ordnung. Dann habe ich hier scheinbar etwas missverstanden.
Vielen Dank für Deine Bemühungen!

naltatis, 2024-05-10:
Ja, das sind zwei unabhängige Feature.

- Das eine ist ne feste Grenze unter der immer geladen wird. Auch nur verfügbar wenns ne CO2 Datenquelle gibt.
- Den Planer gibts immer und der optimiert nach den besten Daten, die er hat.

Aktuell gibts noch die Möglichkeit für den Planner einen separaten Tarif zu hinterlegen. Da könntest du dann einfach deinen fixen Netztarif einstellen. Das Feature ist aber undokumentiert und könnte sich später mal ändern bzw. rausfliegen. Sollte aber jetzt den von dir gewünschten Effekt haben ohne CO2 zu löschen.

```
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.294
  co2:
    ...
  planner:
    type: fixed
    price: 0.294
```