evcc-io/evcc #13829: UI does not allow to override strong password suggestions on MacOS safari

GrimmiMeloni, 2024-05-09:
### Describe the bug

The new admin password enforcement dialogue on startup is behaving "strange" in Safari on my Mac.
I have password suggestions of Safari enabled.

Upon first boot of my 0.126.x upgrading from 0.125 I was asked to enter an admin password.
The dialog was autofilled by Safari, looking like this.

[image]

However, I am used to having the option to override Safari's suggestion. 
Usually when clicking onto the suggested password, a (safari rendered) menu opens up with additional choices including the option to enter an individual password. Example screenshot of it:

[image]

For the evcc password dialogue this menu does not open, no matter how much I click into the fields. 
I am stuck with the safari generated password, and I cannot override it.
Since I have never seen this problem in any other site so far, I assume this has to do with the CSS/UI/Click Event handling of evcc.

I worked around the issue by using Chrome for the initial setup.

### Steps to reproduce

1. Upgrade evcc from version < 0.126
2. Open the UI in Safari
3. click in the password field
4. see it get autofilled
5. click the suggested password
6. nothing happens

### Configuration details

```yaml
Mac OS 14.4.1 (23E224)
Safari Version 17.4.1 (19618.1.15.11.14)
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

0.126.2

naltatis, 2024-05-09:
Ich hab mich hier ab den Passwortmanager Empfehlungen von Google und 1Password orientiert. Hatte gehofft, dass Safari da nicht noch was anderes macht 😄

Ich schau da mal rein.

naltatis, 2024-05-09:
Ich hab jetzt mal Safari Passwörter aktiviert (bin selbst 1Passwort Nutzer) um zu sehen, wie sich das auf anderen Seiten und bei uns verhält. Ich konnte keine Unterschiede feststellen.

Ist mit Safari 17.4.1 auf macOS Sonoma gegen den Nightly Stand getestet.
Kannst du das Verhalten reproduzieren?

spiegel

[spiegel.webm](https://github.com/evcc-io/evcc/assets/152287/d5dbf0f4-6995-437e-b1ca-771ad8156e2b)

about you

[aboutyou.webm](https://github.com/evcc-io/evcc/assets/152287/bfe52ddf-53ae-4eea-b1f6-1621b059c51b)

stackoverflow

[stackoverflow.webm](https://github.com/evcc-io/evcc/assets/152287/faaa4e46-fde2-4573-ba0f-0a152315aff3)

evcc

[evcc.webm](https://github.com/evcc-io/evcc/assets/152287/fc39e1aa-d6fb-472d-a0bb-d805be2ea4cf)

---

Safari Ändern Dialog

[image]

RTTTC, 2024-05-09:
Works fine for me... (0.126.2 release version, latest MacOS, latest safari)
(You need to click "other options")
[image]

GrimmiMeloni, 2024-05-10:
OK, launched a local build from yesterday's master, started with demo config. The dialogue looks correct.

[image]

I'll restart and see if it helps. I'll close here for now, as it seems the problem is locally on my end.

closed 2024-05-10 by GrimmiMeloni

GrimmiMeloni, 2024-05-10:
Just to follow up. Restarting did not help.
Other findings:
Visiting my evcc instance via the IP of my docker host makes things work.
Accessing this way also keeps Safari from claiming the site to be insecure.

Just a quick cross check @RTTTC & @naltatis - do you run evcc in a "all bells and whistles" setup in your network, e.g. individual hostnames behind a reverse proxy, potentially even TLS enabled?

naltatis, 2024-05-10:
The tests I've made were direct access via HTTP. I've tested it locally (same machine) and in my local network (Synology/docker) both work as expected.

That said, security related browser features may behave differently depending on the context you open the site in. Clipboard API e.g. works on localhost (unencrypted) but requires a https context when accessing sites over the network.

RTTTC, 2024-05-11:
Very basic installation here, Raspberry with an IP reservation.
HTTP access via IP address.